espressif/esp-idf #8592: 调用esp_mqtt_client_publish重启

cnlnlhb, 2022-03-16:
想实现一个按键发送MQTT消息功能，MQTT收发测试已经没有问题，按键按下后中断函数也可以正常打印，但是无法再按键按下后任务函数中发送MQTT消息，一直报错

IO中断部分代码
/**
* GPIO中断部分
*/
static xQueueHandle gpio_evt_queue = NULL;
static void IRAM_ATTR gpio_isr_handler(void* arg)
{
uint32_t gpio_num = (uint32_t) arg;
xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
}

/**
* 按键处理函数
*/
int last_time = 0;
static void key_task(void* arg)
{
uint32_t io_num;
for(;;) {
if(xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY)) {
// printf("GPIO[%d] intr, val: %d\n", io_num, gpio_get_level(io_num));

int now = xTaskGetTickCountFromISR();
if(gpio_get_level(io_num) == 1){ //高电平 判断短按还是长按
//获取系统当前运行的时钟节拍数,此函数用于在中断服务程序里面调用， 如果在任务里面调用的话，需要使用函数 xTaskGetTickCount，这两个函数切不可混用
if (now - last_time > MIN_DELAY) { //长按
printf("long %d %d\n", now, last_time);
gpio_set_level(3, 0);
//长按业务逻辑
}else if (now - last_time < 5){ //消抖 无操作

}else{ //短按
printf("short %d %d\n", now, last_time);
gpio_set_level(3, 1);
//短按业务逻辑 上报
// char topic_init[64];
// sprintf(topic_init, "/c/%s/send", mqtt_user);
if(client){
esp_mqtt_client_publish(client, '/test', "find", 0, 1, 0); //这行有问题,执行后重启
printf("send ok\n");
}else{
printf("no client");
}
// ESP_LOGI(TAG, "sent publish successful, msg_id=%d", msg_id);
}
}else{ //低电平 按下按键

}
last_time = now;
}
}
}

错误信息
assert failed: xQueueTakeMutexRecursive queue.c:673 (pxMutex)
Stack dump detected
Core 0 register dump:
MEPC : 0x4038061e RA : 0x403879ec SP : 0x3fcbe1d0 GP : 0x3fc904f8
0x4038061e: panic_abort at C:/Users/micro/Desktop/esp-idf/components/esp_system/panic.c:404

0x403879ec: rtos_int_enter at C:/Users/micro/Desktop/esp-idf/components/freertos/port/riscv/portasm.S:35

TP : 0x3fc7f014 T0 : 0x37363534 T1 : 0x7271706f T2 : 0x33323130
S0/FP : 0x0000008a S1 : 0x00000001 A0 : 0x3fcbe20c A1 : 0x3fc91ed1
A2 : 0x00000001 A3 : 0x00000029 A4 : 0x00000001 A5 : 0x3fc9c000
A6 : 0x7a797877 A7 : 0x76757473 S2 : 0x00000009 S3 : 0x3fcbe31d
S4 : 0x3fc91ed0 S5 : 0x74657374 S6 : 0x00000000 S7 : 0x00000000
S8 : 0x00000000 S9 : 0x00000000 S10 : 0x00000000 S11 : 0x00000000
T3 : 0x6e6d6c6b T4 : 0x6a696867 T5 : 0x66656463 T6 : 0x62613938
MSTATUS : 0x00001881 MTVEC : 0x40380001 MCAUSE : 0x00000007 MTVAL : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', 'c:\\code\\esp\\template-app\\build\\template-app.elf', '-ex', 'target remote | "C:\\Users\\micro\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe" "C:/Users/micro/Desktop/esp-idf/tools\\..\\tools\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\micro\\AppData\\Local\\Temp\\tmp8kn1j2ve"', '-ex', 'bt']' returned non-zero exit status 1.
b'C:\\Users\\micro\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r2-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe: warning: Couldn\'t determine a path for the index cache directory.\r\nerror starting child process \'"C:\\Users\\micro\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe" "C:/Users/micro/Desktop/esp-idf/tools\\..\\tools\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\micro\\AppData\\Local\\Temp\\tmp8kn1j2ve"\': CreateProcess: No such file or directory\r\nNo stack.\r\n'

Core 0 register dump:
MEPC : 0x4038061e RA : 0x403879ec SP : 0x3fcbe1d0 GP : 0x3fc904f8
TP : 0x3fc7f014 T0 : 0x37363534 T1 : 0x7271706f T2 : 0x33323130
S0/FP : 0x0000008a S1 : 0x00000001 A0 : 0x3fcbe20c A1 : 0x3fc91ed1
A2 : 0x00000001 A3 : 0x00000029 A4 : 0x00000001 A5 : 0x3fc9c000
A6 : 0x7a797877 A7 : 0x76757473 S2 : 0x00000009 S3 : 0x3fcbe31d
S4 : 0x3fc91ed0 S5 : 0x74657374 S6 : 0x00000000 S7 : 0x00000000
S8 : 0x00000000 S9 : 0x00000000 S10 : 0x00000000 S11 : 0x00000000
T3 : 0x6e6d6c6b T4 : 0x6a696867 T5 : 0x66656463 T6 : 0x62613938
MSTATUS : 0x00001881 MTVEC : 0x40380001 MCAUSE : 0x00000007 MTVAL : 0x00000000
MHARTID : 0x00000000

尝试了使用esp_mqtt_client_enqueue 代替esp_mqtt_client_publish，仍不能解决
报错：
assert failed: xQueueTakeMutexRecursive queue.c:673 (pxMutex)
有时候也有
Guru Meditation Error: Core 0 panic'ed (Load access fault). Exception was unhandled.
或者
assert failed: xQueueSemaphoreTake queue.c:1451 (pxQueue->uxItemSize == 0)

只有在static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event)中使用esp_mqtt_client_publish没有问题，其他地方如app_main函数中、任务函数中使用都会立即触发重启

ESP-IDF版本：4.3.2
芯片：ESP32-C3

euripedesrocha, 2022-03-18:
Hi @cnlnlhb, 
could you direct the core dump to a file and share with us? 

In your snippet, the only thing I see is that you are calling `xTaskGetTickCountFromISR` instead of `xTaskGetTickCount` and that you don't have any initialization for your MQTT client, but I'm assuming those are just issues in the snippet.

cnlnlhb, 2022-03-19:
> Hi @cnlnlhb, could you direct the core dump to a file and share with us?
>
> In your snippet, the only thing I see is that you are calling `xTaskGetTickCountFromISR` instead of `xTaskGetTickCount` and [...]

assert failed: xQueueTakeMutexRecursive queue.c:673 (pxMutex)
Stack dump detected
Core  0 register dump:
MEPC    : 0x40380626  RA      : 0x40387abe  SP      : 0x3fcc5450  GP      : 0x3fc908f8  
0x40380626: panic_abort at C:/Users/micro/Desktop/esp-idf/components/esp_system/panic.c:404

0x40387abe: rtos_int_enter at C:/Users/micro/Desktop/esp-idf/components/freertos/port/riscv/portasm.S:35

TP      : 0x3fc85bfc  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x0000008a  S1      : 0x00000001  A0      : 0x3fcc548c  A1      : 0x3fc93311
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc9e000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcc559d
S4      : 0x3fc93310  S5      : 0x74657374  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', 'c:\\code\\esp\\220301_mqtt_switch\\build\\template-app.elf', '-ex', 'target remote | "C:\\Users\\micro\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe" "C:/Users/micro/Desktop/esp-idf/tools\\..\\tools\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\micro\\AppData\\Local\\Temp\\tmp_xh6utg7"', '-ex', 'bt']' returned non-zero exit status 1.
b'C:\\Users\\micro\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r2-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe: warning: Couldn\'t determine a path for the index cache directory.\r\nerror starting child process \'"C:\\Users\\micro\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe" "C:/Users/micro/Desktop/esp-idf/tools\\..\\tools\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\micro\\AppData\\Local\\Temp\\tmp_xh6utg7"\': CreateProcess: No such file 
or directory\r\nNo stack.\r\n'

Core  0 register dump:
MEPC    : 0x40380626  RA      : 0x40387abe  SP      : 0x3fcc5450  GP      : 0x3fc908f8
TP      : 0x3fc85bfc  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x0000008a  S1      : 0x00000001  A0      : 0x3fcc548c  A1      : 0x3fc93311
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc9e000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcc559d
S4      : 0x3fc93310  S5      : 0x74657374  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fcc5450: 0xa5a5a5a5 0xa5a5a5a5 0x3c13485c 0x4038e0e0 0x3fc94400 0x3c13485c 0x3fc946ec 0x3c1349a4
3fcc5470: 0x3fc94410 0x3fcc5484 0x3fc94414 0x3c134d68 0x3fc93310 0x00333736 0xa5a5a5a5 0x65737361
3fcc5490: 0x66207472 0x656c6961 0x78203a64 0x75657551 0x6b615465 0x74754d65 0x65527865 0x73727563
3fcc54b0: 0x20657669 0x75657571 0x3a632e65 0x20333736 0x4d787028 0x78657475 0x3fc90029 0x40387fae
3fcc54d0: 0x00000000 0x00000000 0x3fc9ebd0 0x4038859a 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000
3fcc54f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001
3fcc5510: 0x00000000 0x3fcc558c 0x3c137064 0x00000001 0x00000005 0x00000001 0x3c135d7c 0x4212919a
3fcc5530: 0x4038448a 0x4038b102 0x00000001 0x00000000 0x00000000 0x74657374 0x00000001 0x00000000
3fcc5550: 0x3c13638c 0x00000000 0x00000000 0x40000000 0x00000000 0x00000000 0x00000001 0x00000000
3fcc5570: 0x3c13638c 0x00000000 0x3fcba800 0x40388d16 0x00000000 0x3c13637c 0x3fcb7438 0x00000000
3fcc5590: 0x3c13638c 0x00000004 0x3fcba800 0x4200ebee 0x00000000 0x00000000 0x00000310 0x42009bc2
3fcc55b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000310 0x42009bec
3fcc55d0: 0x00000000 0x00000000 0x00000000 0x00000009 0x00000000 0x00000000 0x00000000 0x4038affa
3fcc55f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcc5610: 0x27146414 0x01010107 0x00000001 0x62746c65 0x34366469 0x00000001 0x01366469 0x00200100
3fcc5630: 0x00000020 0x000000ff 0x7f7fffff 0x00000000 0x3fe00000 0x00000000 0x3ff00000 0x00000000
3fcc5650: 0x40000000 0x94a03595 0x3fdfffff 0x35afe535 0x3fe00000 0x94a03595 0x3fcfffff 0xffffffff
3fcc5670: 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3ff80000 0x636f4361
3fcc5690: 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000 0x40240000 0x00000000
3fcc56b0: 0x401c0000 0x00000000 0x40140000 0x00000000 0x43500000 0x00000000 0x7ff80000 0x7fc00000
3fcc56d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00019979 0x3fcc0014 0x3fcc0014 0x38658723
3fcc56f0: 0x990c9730 0x471eb831 0x2242a9a0 0xb4ec8085 0xdc73074d 0xdfece1da 0x7be6eadb 0x1934c3b9
3fcc5710: 0x6b890ac7 0xcb48e669 0xca10137f 0xfba0050f 0x10d5f428 0xd3abcd46 0xf2655fbf 0xbe8a3a7d
[... cut ...]
IAAAAP8AAAD//39/AAAAAAAA4D8AAAAAAADwPwAAAAAAAABAlTWglP//3z815a81
AADgP5U1oJT//88/////////738AAAAAAADAPwAAAAAAADBAAAAAAAAA+D9hQ29j
p4fSP7PIYIsoisY/+3mfUBNE0z8AAAAAAAAkQAAAAAAAABxAAAAAAAAAFEAAAAAA
AABQQwAAAAAAAPh/AADAfw==
FAAAAEgAAABKIAAA
RVNQX0NPUkVfRFVNUF9JTkZPAAA=
AAEFAGJhMzFjM2M1MzZjZWE3NjUAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
DAAAAAQAAAClAgAA
RVhUUkFfSU5GTwAA
3HPLPw==
46QXRg==
================= CORE DUMP END =================
Coredump checksum='4617a4e3'
I (2566) esp_core_dump_uart: Core dump has been written to uart.

euripedesrocha, 2022-04-01:
Hi @cnlnlhb, could you share the decoded core dump? Instructions [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump.html#save-core-dump-to-flash).

Alvin1Zhang, 2022-04-12:
@cnlnlhb Thanks for reporting, would you please help share the decoded core dump? Thanks.

closed 2022-05-06 by Alvin1Zhang