espressif/esp-idf #8593: ESP-IDF SPIFFS fails to save files even with alot more bytes left

Sjurinho, 2022-03-16:
## Environment

Development Kit: ESP32-DevKitC
Kit version: v1
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.2.2
Build System: idf.py
Compiler version (run xtensa-esp32-elf-gcc --version to find it): 8.4.0
Operating System: Linux
Using an IDE?: VSCode with the ESP-IDF extension
Power Supply: External 5v
## Problem Description

I am running a logger on my esp32 microcontroller. The logs are circular and thus consists of 6 files of size 16384 bytes which are deleted as the spiffs fill up. Additionally I have a small file (approx. 1000 bytes)  of config values. After a while I saw that the spiffs wasn't able to save new logs, and turning on the debug logs showed that the gc in the spiffs couldn't erase certain memory addresses. This is shown in the attached logs (spiffs_erros.txt). I have tried to search around on the forums, but I have not been able to find the cause of this problem. It does not appear to be the spiffs running out of memory, as the esp_spiffs_info() api shows that only 38905 bytes are used out of 233681. My spiffs partition (see below) is set to be 256k bytes. I think the problem has something to do with the free blocks being equal to 0, but not sure why or how to fix it. To generate the spiffs I use the standard CMake api spiffs_create_partition_image(spiffs spiffs_image FLASH_IN_PROJECT), as shown in [storage/examples](https://github.com/espressif/esp-idf/blob/5963de1caf284b14ddfed11e52730a55e3783a3d/examples/storage/spiffsgen/main/CMakeLists.txt). Correct me if I'm wrong, but I think this gives a page size of 256 bytes (as in the sdkconfig) and a 4 KiB erase block size.
Is anyone familiar with the issue and how to resolve/work around it?

### Expected Behavior
Spiffs continue to work as long as there are bytes left (according to esp_spiffs_info), and the circular logs work.

### Actual Behavior
Spiffs run out of blocks(?) and therefore stops being able to save new logs.

### Steps to reproduce

1. Mount Spiffs
2. Write logs (approximately 95bytes per entry)
3. Wait for spiffs to fail at erasing blocks

### Partition Table
\# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      0x9000,   20k,
otadata,  data, ota,      ,         8k,
factory,  app,  factory,  ,         1215k,
ota_0,    app,  ota_0,    ,         1215k,
ota_1,    app,  ota_1,    ,         1215k,
spiffs,   data, spiffs,   ,         256k,
coredump, data, coredump, 0x3F0000, 64k

### Initialization of spiffs
I initialize my spiffs accordingly
```cpp
if (!esp_spiffs_mounted(NULL)) {
        esp_vfs_spiffs_conf_t conf = {
            // Data partition name ("spiffs", "/spiffs") in file: custom_partitions_spiffs.csv
            .base_path = kBasePath.substr(0, kBasePath.length() - 1).c_str(),
            .partition_label = NULL,
            .max_files = 5,
            .format_if_mount_failed = false,
        };

        ESP_LOGI(TAG, ">> esp_vfs_spiffs_register(conf)");
        esp_err_t err_resp = esp_vfs_spiffs_register(&conf);
```

## LogFiles

- The file spiffs_error.txt shows the debug logs of the spiffs write failure.
- The file spiffs_format_errors.txt shows the debug logs and errors when trying to format the spiffs partition.
[spiffs_error.txt](https://github.com/espressif/esp-idf/files/8264953/spiffs_error.txt)
[spiffs_format_errors.txt](https://github.com/espressif/esp-idf/files/8264954/spiffs_format_errors.txt)

Sjurinho, 2022-03-16:
I state more clearly that a format of the mounted spiffs did NOT fix the problem, as I have seen other forums mention as a solution. The logs of that is attached in spiffs_format_errors.txt

igrr, 2022-03-16:
Thanks for the detailed issue description @Sjurinho!

It seems that first it would be nice to solve the issue with failed erase operations. I'm not sure if SPIFFS will continue working correctly after some erase operations fail.

If you run the `spiffs` example provided in ESP-IDF, do you get similar errors about failed erase operations? If not, do you think you could reduce your application to some bare minimum amount of code which still reproduces these erase errors?

Another question, could you please upload the `sdkconfig` file from your project and your partition table CSV file?

igrr, 2022-03-17:
I'm sorry, clicking the above spiffs.zip link gives me "Not found" error, could you please try again?

igrr, 2022-03-17:
The issue is that the the `spiffs` partition in your partition table is not aligned to 4kB boundary. Currently it is at 0x39fc00.
Since the flash erase size is 4kB, any data partition which has to support erase operations needs to be aligned to 4kB.
We have recently added a warning about this in commit https://github.com/espressif/esp-idf/commit/b56c9aafe46b00456c12b6979ee0c711575a5697 (part of v4.4 release), and starting from v5.0 unaligned data partitions won't be supported.
Please adjust your partition table to make all partitions aligned to 4kB, e.g. by changing the sizes of factory/ota partitions to 1216kB. Then all the partitions will be aligned:
```
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
factory,app,factory,0x10000,1216K,
ota_0,app,ota_0,0x140000,1216K,
ota_1,app,ota_1,0x270000,1216K,
spiffs,data,spiffs,0x3a0000,256K,
coredump,data,coredump,0x3f0000,64K,
```
You can check the result by running `idf.py partition_table` command.

Sjurinho, 2022-03-18:
@igrr, Indeed, changing the app partitions seems to have fixed my problem. That makes sense. Thank you very much for your help, you are a lifesaver! :) 

That being said, I do miss some warning in the docs about this though.

igrr, 2022-03-18:
Thanks for confirming the fix!

Indeed this is an easy mistake to make, and it's a simple thing to add a check/warning for.

We'll backport the commit I have linked to release/v4.3 and release/v4.4 and close this issue afterwards.

Sjurinho, 2022-03-18:
After further inspection, I see that the app partitions do not actually change in size with the modification. They are still 0x130000 in size, as before. However, it does move the offset of the spiffs partition. I think what happened may have been that the spiffs actually overlapped with the ota1 partition, and was therefore unable to delete the blocks related to those memory addresses.

closed 2022-04-07 by espressif-bot

allacmc, 2024-09-28:
I'm having the same problem. And I don't know how to solve it.

This concept of aligning with 4kb?

How do you do this?

`

I (2213) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (2233) INFO: Wifi-Conectado
I (2263) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3223) esp_netif_handlers: sta ip: 192.168.1.24, mask: 255.255.255.0, gw: 192.168.1.254
I (3233) IP:: got ip:192.168.1.24
I (3243) MQTT: (MQTT-IN Usuario) - Other event id:7 - MQTT-IN(Usuario)
I (3253) MQTT: Other event id:7  - MQTT-OUT(Licenca Enervision)
I (3343) NTP:: Sincronizando hora via NTP...
I (3413) CONFIG FILE: Dados Wifi
I (3413) NTP:: Aguardando sincronizacao com NTP...
W (3583) Datalogger: Datalogger Ativado
I (5033) MQTT: MQTT_EVENT_CONNECTED - MQTT-OUT(Licenca Enervision)
I (5033) MQTT: MQTT_EVENT_CONNECTED - MQTT-IN(Usuario)
I (5243) MQTT: Other event id:3  - MQTT-OUT(Licenca Enervision)
I (5443) MQTT: Other event id:3  - MQTT-OUT(Licenca Enervision)
I (5483) NTP:: Sincronização com NTP realizada...
I (13613) main_task: Returned from app_main()
I (52643) wifi:<ba-add>idx:1 (ifx:0, 38:6b:1c:01:ea:a8), tid:6, ssn:0, winSize:64
I (322483) NTP:: Sincronizando hora via NTP...
I (322503) NTP:: Aguardando sincronizacao com NTP...
I (322513) NTP:: Sincronização com NTP realizada...
E (339793) ERRO: Erro ao abrir arquivo na flash
W (339803) httpd_txrx: httpd_resp_send_err: 404 Not Found - Nothing matches the given URI
W (339813) httpd_uri: httpd_uri: uri handler execution failed
W (345113) HTML: TempoTotal:11197
I (391903) TAG: String de consulta da URL: name=2024-09-27.csv
I (391913) TAG: Solicitando arquivo: /2024-09-27.csv
I (404713) TAG: String de consulta da URL: name=setup.json
I (404723) TAG: Solicitando arquivo: /setup.json
I (5201493) NTP:: Sincronizando hora via NTP...
I (5201603) NTP:: Aguardando sincronizacao com NTP...
I (5201613) NTP:: Sincronização com NTP realizada...
E rmt: user buffer too small, received symbols truncated
E rmt: hw buffer too small, received symbols truncated
E rmt: user buffer too small, received symbols truncated
E rmt: hw buffer too small, received symbols truncated
E rmt: user buffer too small, received symbols truncated
E rmt: hw buffer too small, received symbols truncated
E rmt: user buffer too small, received symbols truncated
E rmt: hw buffer too small, received symbols truncated
I (9522203) MQTT: MQTT_EVENT_DISCONNECTED - MQTT-OUT(Licenca Enervision)
I (9532323) MQTT: Other event id:7  - MQTT-OUT(Licenca Enervision)
E (9546213) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (9547343) esp-tls-mbedtls: Failed to verify peer certificate!
E (9547603) esp-tls: Failed to open new connection
E (9547753) transport_base: Failed to open a new connection
E (9547863) mqtt_client: Error transport connect
I (9548343) MQTT: MQTT_EVENT_ERROR - MQTT-OUT(Licenca Enervision)
E (9548613) MQTT: Last error (MQTT-OUT Licenca Enervision) - reported from esp-tls: 0x801a
E (9548723) MQTT: Last error r(MQTT-OUT Licenca Enervision) - eported from tls stack: 0x2700
I (9549253) MQTT: (MQTT-OUT Licenca Enervision) - Last errno string (Success)
I (9549393) MQTT: MQTT_EVENT_DISCONNECTED - MQTT-OUT(Licenca Enervision)
I (9559523) MQTT: Other event id:7  - MQTT-OUT(Licenca Enervision)
E rmt: user buffer too small, received symbols truncated
E rmt: hw buffer too small, received symbols truncated
E (9571713) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (9572233) esp-tls-mbedtls: Failed to verify peer certificate!
E (9573303) esp-tls: Failed to open new connection
E (9573493) transport_base: Failed to open a new connection
E (9573933) mqtt_client: Error transport connect
I (9574093) MQTT: MQTT_EVENT_ERROR - MQTT-OUT(Licenca Enervision)
E (9574233) MQTT: Last error (MQTT-OUT Licenca Enervision) - reported from esp-tls: 0x801a
[... 239 lines cut ...]
I (912) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (922) LCD:: Install ILI9341 panel driver
I (922) ili9341: LCD panel create success, version: 1.2.0
I (1052) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1052) TouchPanel: Touch Inicializado
I (1052) TouchPanel: Calibration data loaded from NVS
I (1062) LCD:: Turn on LCD backlight
I (1062) LCD:: Initialize LVGL library
I (1062) LCD:: Register display driver to LVGL
I (1082) LCD:: Install LVGL tick timer
I (1082) LCD:: Create LVGL task
I (1082) Display: Starting LVGL task
I (1222) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1232) gpio: GPIO[22]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (1222) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1242) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1252) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1262) CONFIG FILE: Configuring SPIFFS
I (1342) CONFIG FILE: SPIFFS: total: 474641, used: 343368
I (1342) CONFIG FILE: SPIFFS mounted successfully
I (1342) LOG: Sistema de log configurado
W (1402) SPIFFS: Failed to update mtime (-10001)
I (1402) SensorTask: Dispositivo DS18B20 encontrado: 6400000011F5DC28

`