espressif/esp-idf #8598: Esp_timer get time in days, hours, minutes...

domko99, 2022-03-16:
Hi there. I am looking for measuring time from start of my program. I know there is a timer esp_timer which can measure time from start of program  in microseconds. Is it a good start to use this timer and convert it is values to days, hours, minutes... or are there any better options to be used.

I was using basic C functions to get time from start but the problem is I cant reset timer when I need to (it even doesnt reset after esp_restart()). So I am trying to find some timer which I can reset and esp_timer functions esp_timer_delete() and esp_timer_create() I can. But there is problem with only microseconds return value by esp_timer.

The best solution for me would be to set date and time (2022-03-17 14:15:00) by user (by me in code) and start timing from that date. I am measuring data (temperature, humidity...) and I want to mark every sample with time when the sample was taken and also be able to reset timer after the restart of chip, But I am not sure which solution would be the best. There is the soultion with SNTP server but I am not just there with my skills to make it so I am looking for easier solutions. 

Thanks for any help and suggestions.

0xjakob, 2022-03-17:
@domko99 Have you considered using an external real time module?

domko99, 2022-03-17:
I dont have a such a hardware....

igrr, 2022-03-17:
@domko99 You can call `settimeofday` function at the beginning of the program, then call `gettimeofday` later to get the current time. `settimeofday` and `gettimefoday` operate on UNIX timestamps. You can use C standard library functions to convert between UNIX timestamps and broken-down time.

Here's an example: https://gist.github.com/igrr/3f322b773eda81c713615a2337f37b94

domko99, 2022-03-18:
Thank you. This is the solution which I needed. I can close this issue now.

closed 2022-03-18 by domko99