espressif/esp-idf #8601: mdns_delegate_hostname_add crashes if hostname hasn't been set

tannewt, 2022-03-17:
I think @david-cermak has been working in mdns code most recently.

## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used: ESP-32-S3
- IDF version (run ``git describe --tags`` to find it):
 ESP-IDF v4.4-351-g121ddb87e5-dirty
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32s3-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

The IDF crashes with a LoadProhibited if mdns_delegate_hostname_add is called before mdns_hostname_set. This is caused by `_hostname_is_ours` not checking to see if `_mdns_server->hostname` is NULL. The top check should be:

```c
    if (_mdns_server->hostname != NULL && strcasecmp(hostname, _mdns_server->hostname) == 0) {
        return true;
    }
```

### Expected Behavior

It shouldn't crash.

### Actual Behavior

It crashes with a LoadProhibited.

### Steps to reproduce

Call `mdns_delegate_hostname_add` before `mdns_hostname_set`.

## Debug Logs

```
Getting IP
Adding delegate
hostname copy 0x3fcf6c10 tannewt
address list copy 0x3fcf6c44
mdns actioho
            tnrme
hhosnnmm d0ne
000d0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400553cb  PS      : 0x00060630  A0      : 0x82097870  A1      : 0x3fcf7b80  
A2      : 0x3fcf6c10  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3fcf7af0  
A6      : 0x00000004  A7      : 0x3fca333c  A8      : 0x00000074  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3c1584e1  A12     : 0x3ff191fd  A13     : 0x3fcf7ad1  
A14     : 0x00000008  A15     : 0xffffffff  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0x400553c8:0x3fcf7b80 |<-CORRUPTED
```

(Garbled prints are from printing from both cores at once. It isn't the issue.)

chegewara, 2022-03-18:
https://github.com/espressif/esp-idf/issues/8307
Probably not backported yet.

david-cermak, 2022-04-01:
Hi @tannewt 

Any particular reason why you've called `add_delegated_host()` before `mdns_hostname_set()` ? Are you trying to advertise or delegate services with *no internal hostname*? Or is it just the init part of your code?

tannewt, 2022-04-01:
It wasn’t deliberate. I’m using the delegate to advertise a second name for the same ip. (One name is unique and one is a catch all.)

david-cermak, 2022-04-06:
ok, thanks. I'll update the docs to make it clear that the own hostname is a prerequisite to advertizing services or delegating other names. it's just mentioned in the example:

https://github.com/espressif/esp-idf/blob/4350e6fef859b94f3efae6ceced751911c9d2cbc/examples/protocols/mdns/main/mdns_example_main.c#L41-L42

Also, as @chegewara (Thank you!) pointed out, some fixes from master haven't been backported to `v4.4` yet. If you see any other issue with delagating services, please try to backport:

* 51a5de2525 mdns: Fix potential read behind parsed packet
* 9cbdb8767b mdns: Fix memleak when adding delegated host
* a57be7b7d1 mdns: Fix null-service issue when parsing packets

will be added to `v4.4` soon.

tannewt, 2022-07-15:
I think there is also a race condition in this code between the task setting `_mdns_server->hostname` and adding a service. It is possible to add a service with a NULL hostname if the mdns task hasn't processed the set hostname action yet. It then later crashes in a couple places that don't check that hostname for a service is not NULL.

closed 2024-05-07 by david-cermak