espressif/esp-idf #8602: ipv6 udp multicast problem over w5500

achao1104, 2022-03-18:
sample: sockets\udp_multicast, idf4.3.2
connections: use esp32s2(with w5000) to connect other devices via pc2pc cable

Here are the test results:

1) esp32s2 connects to a host of python simulator running on Raspberry Pi
   a) if esp32s2 listens on "ff02::1:ffxx:xxxx" and sends multi cast udp to "ff02::1", esp32s2 successfully receives udp response from simulator.
   b) if esp32s2 listens on "ff02::1" and sends multi cast udp to "ff02::1", esp32s2 can not receive udp response from simulator.

2) a client of  python simulator running on Raspberry Pi connects to a host of python simulator running on Raspberry Pi
   a) if client listens on "ff02::1" and sends multi cast udp to "ff02::1", client successfully receives udp response from simulator.

3) a client of  python simulator running on Raspberry Pi connects to smart meter device of third party.
   a) if client listens on "ff02::1" and sends multi cast udp to "ff02::1", client successfully receives udp response from simulator.

4) esp32s2 connects to smart meter device of third party.
   a) if esp32s2 listens on "ff02::1:ffxx:xxxx" and sends multi cast udp to "ff02::1", esp32s2 can not receive udp response from smart meter device of third party.
   
Could you please check why scenario of 1.b, 4.a failed?
Thanks for your help in advanced, it is really urgent for us.

yuanjianmin, 2022-03-18:
For the 1.b, it maybe the simulator send udp response to address "ff02::1:ffxx:xxxx", and the esp32s2 not listen on it.
For the 4.a, it maybe the smart meter device of third party send udp response to address "ff02::1", and the esp32s2 not listen on it.
I think you can capture the packet, include the which address the esp32s2 listened on and the peer device response to.

achao1104, 2022-03-18:
thanks for your reply. for the scenario of esp32s2 connects to smart meter device of third party, it is not easy to capture the packet. I am going to let esp32s2 listen on both "ff02::1:ffxx:xxxx" and "ff02::1" for a new test (connects smart meter). Do you have any other inputs besides listening on both "ff02::1:ffxx:xxxx" and "ff02::1"?

ESP-YJM, 2022-03-29:
Hi @achao1104, could you please try add `NETIF_FLAG_MLD6` flag for ethernetif in ethernetif .c. You can follow the wlanif.

Alvin1Zhang, 2022-04-12:
@achao1104 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

closed 2022-05-29 by espressif-bot

AxelLin, 2022-08-03:
@ESP-YJM 
This issue was reported on v4.3.2, but v4.3 branch does include this fix so far.

ESP-YJM, 2022-08-08:
> 

Yes, we not backport this fix to v4.3 branch after internal discussion. This is not a serious bug fix.

AxelLin, 2022-08-08:
> @AxelLin Yes, we not backport this fix to v4.3 branch after internal discussion. This is not a serious bug fix.

https://github.com/espressif/esp-idf/commit/667a7c6cd05dcbe37f309080925ac13d12b0c815 is really very trivial fix.
It's reported on v4.3 and the fix is so trivial, so what's the reason prevent backporting the fix?

ESP-YJM, 2022-08-11:
@AxelLin Add this 'NETIF_FLAG_MLD6' flag will make more traffic for ethernet, which maybe could potentially cause some timing issues. And v4.3 branch approach its maintenance period. If user want to use this function, they can add it munally.

AxelLin, 2022-08-11:
> @AxelLin Add this 'NETIF_FLAG_MLD6' flag will make more traffic for ethernet, which maybe could potentially cause some timing issues. And v4.3 branch approach its maintenance period. If user want to use this function, they can add it munally.

@ESP-YJM 
I don't get it.
You are telling me the fix may cause other issue, but v4.4 8215bd6670e545e08bbfcf84a91702e16f3d19fd is fine.
Why the commit log does not mention such side-effect and now the side-effect becomes a concern?
It's important to clarify if the fix is a correct fix or not becasue the code is in v4.4+ now.
I don't think maintenance period is a good excuse especially this issue was reported against v4.3.
Currently there are many fixes (including several important fixes) after v4.3.3, I believe the v4.3.4 will be release someday.

ESP-YJM, 2022-08-17:
@AxelLin I will re-open merge request and re-discuss internally. Thanks for your comments.

AxelLin, 2022-08-17:
> @AxelLin I will re-open merge request and re-discuss internally. Thanks for your comments.

@ESP-YJM 
Could you share what's exactly the problem of this fix?

wlan inteface also has similar code:
https://github.com/espressif/esp-idf/blob/release/v4.3/components/lwip/port/esp32/netif/wlanif.c#L99-L103

ESP-YJM, 2022-08-17:
@AxelLin This fix makes ethernet interface support MLD6. And will increase packet traffic on ethernet(Receive MLD6 and send MLD6 packet).

AxelLin, 2022-08-17:
> @AxelLin This fix makes ethernet interface support MLD6. And will increase packet traffic on ethernet(Receive MLD6 and send MLD6 packet).

The point is why it's a concern? (The same setting is also in wlanif.c)

david-cermak, 2022-08-18:
@AxelLin we will not backport this commit to `v4.3`, because this is a minor, more or less a consistency bugfix and the release branches of 4.1, 4.2, 4.3 only accept high severity and security bugfixes.

> 667a7c6 is really very trivial fix.

This is a trivial fix, yes, but what we have to look at is the severity and impact of the change, which we cannot simply measure in LOC changed. And I agree that our process of fixing issues and propagating these fixes to different release branches is not ideal and we should try to communicate better about assessments of the reported bugs and which releases to be patched (plus optionally the rationale behind the decision and the expected time span?)

The process is simple, though: We backport all bugfixes to the releases in the "service period", and only major and security-related fixes to those in "maintenance period". Deciding ff a certain fix is assessed as *major* and qualified for backporting to the releases in maintenance is not that simple and we have to very carefully weigh what benefit it brings to users on those release branches. 

In this particular case, the benefit of adding 667a7c6 to `v4.3` is negligible, at-most; the change makes the ESP32's LL IPv6 address discoverable on the Ethernet interface, so applications using current `v4.3.3` and `v4.3.2` would not even work if they rely on the `MLD6` on Ethernet (they could enable it manually, or they could use IPv6 without this flag if addressed directly), but  again, backporting this would be of no-benefit to users on this release branch.

> Could you share what's exactly the problem of this fix?

Enabling the `MLD6` by default makes the interface to join a multicast group and in turn receive more packets (in some  environments). This is a very-very-very marginal concern and applicable to 0.00% of applications and networks, but adds-up to the other side of the equation about why to backport or not. It would be a shame if we'd get a report that *"the ethernet is slower after upgrading from `v4.3.3` to `v4.3.4`"* even if it comes from a very unusual use-case.