espressif/esp-idf #8603: wifi_prov_mgr (BLE provision) to add more endpoints does not work

AxelLin, 2022-03-18:
## Environment

- Module or chip used: ESP32-C3-DevKitM-1
- IDF version: v5.0-dev-2137-g2f9d47c708
- Build System: idf.py
- Compiler version: riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Test wifi_prov_mgr exmple code.

I try to add more enpoints with below changes, but it does not work.
I verify the BLE using nRFConnect app on android, I can find the Characteristic with descriptor "custom-data".
But I cannot find the Characteristic with descriptor "custom-data2" or "custom-data3".

(My sdkconfig is using nimble)

````
diff --git a/examples/provisioning/wifi_prov_mgr/main/app_main.c b/examples/provisioning/wifi_prov_mgr/main/app_main.c
index 81925e2e7b..fc88f402f2 100644
--- a/examples/provisioning/wifi_prov_mgr/main/app_main.c
+++ b/examples/provisioning/wifi_prov_mgr/main/app_main.c
@@ -254,7 +254,8 @@ void app_main(void)
          *          using X25519 key exchange and proof of possession (pop) and AES-CTR
          *          for encryption/decryption of messages.
          */
-        wifi_prov_security_t security = WIFI_PROV_SECURITY_1;
+        wifi_prov_security_t security = WIFI_PROV_SECURITY_0;
+        //wifi_prov_security_t security = WIFI_PROV_SECURITY_1;
 
         /* Do we want a proof-of-possession (ignored if Security 0 is selected):
          *      - this should be a string with length > 0
@@ -299,6 +300,8 @@ void app_main(void)
          * This call must be made before starting the provisioning.
          */
         wifi_prov_mgr_endpoint_create("custom-data");
+        wifi_prov_mgr_endpoint_create("custom-data2");
+        wifi_prov_mgr_endpoint_create("custom-data3");
         /* Start provisioning service */
         ESP_ERROR_CHECK(wifi_prov_mgr_start_provisioning(security, pop, service_name, service_key));
 
@@ -307,6 +310,8 @@ void app_main(void)
          * has already been created above.
          */
         wifi_prov_mgr_endpoint_register("custom-data", custom_prov_data_handler, NULL);
+        wifi_prov_mgr_endpoint_register("custom-data2", custom_prov_data_handler, NULL);
+        wifi_prov_mgr_endpoint_register("custom-data3", custom_prov_data_handler, NULL);
 
         /* Uncomment the following to wait for the provisioning to finish and then release
          * the resources of the manager. Since in this case de-initialization is triggered

````

laukik-hase, 2022-03-21:
Hello, @AxelLin!

Thank you for the issue report. I was investigating this issue but unfortunately could not reproduce it with the `wifi_prov_mgr` example with `ESP32-C3`. 

I was able to discover and send data to all the 3 endpoints (as given in your patch) with the `nRF Connect` app. I have attached the debug logs from my test run [here](https://github.com/espressif/esp-idf/files/8313555/wifi_prov_endpoint.zip) (Sent the character `U` to all the endpoints).

Could you share logs from your side with `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y` and `CONFIG_BT_NIMBLE_LOG_LEVEL_DEBUG=y`?

You will need to increase the `nimBLE` task stack size at `CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE` to 8192 when turning on debug logs.

AxelLin, 2022-03-23:
@laukik-hase 

I found the nRFConnect seems cache some information so it shows old characteristic list
e.g. After you build the application as my patch, you can find custom-data2 and custom-data3 characteristics.
If you then comment out the changes for custom-data2 and custom-data3 , build and run the app.
Then the nRFConnect still shows custom-data2 and custom-data3 characteristics.
I have to click the "Refresh services" on the nRFConnect app, then it shows correct characteristics.
(In my previous test, I try build with standard sample code first, then build with adding custom-data2 and custom-data3 characteristics, so the nRFConnect still shows old characteristics. i.e. does not show custom-data2 and custom-data3.)
(So this is probably not the esp32 issue)

BTW, I didn't set CONFIG_WIFI_PROV_BLE_BONDING in my build.
But nRFConnect shows the device in "BOUNED" tab, not sure if this is normal or not.

laukik-hase, 2022-03-28:
> BTW, I didn't set CONFIG_WIFI_PROV_BLE_BONDING in my build.
But nRFConnect shows the device in "BOUNED" tab, not sure if this is normal or not.

There seems to be a default tab for `BONDED` devices in the app. Two devices can have different bonding configurations (in your case, the target has set it to false while the provisioner device could have 

When bonding is true from both sides, the devices will first pair and then store each other's pairing info (LTK). The next time they go for setting up a connection, it will be established directly without the pairing step.

Closing the issue.

closed 2022-03-28 by laukik-hase