espressif/esp-idf #8604: How does the Bluetooth server determine whether the connected client is an Android device or an apple device？

fan183579, 2022-03-18:
When using "gatt_service" to connect the client, I have to judge whether the client is an Android device or an apple device because of some needs, but I can't find the relevant fields. How can I know?

moefear85, 2022-03-19:
this question belongs on the [esp forum](https://www.esp32.com/)

fan183579, 2022-03-21:
使用的ESP32-C3芯片，IDF是4.4的，参考的是gatt_service例程

fan183579, 2022-04-24:
在使用8080驱动显示屏的时候，过一段时间就会出现显示错位的情况出现，使用的ESP32-S3-DevKitC-1开发板，LCD驱动IC为hx8369a，使用LCD下LVGL例程，按照现有驱动自行改写的驱动代码，没有找到出现这个情况的原因

esp-cjh, 2022-08-10:
We have no way to get the device type of the master. But you can use custom GATT service to achieve this after connected.
Such as a characteristic which is writable and Client write the characteristic value to tell server it's device type.

fan183579, 2022-10-11:
ESP论坛并没有人回复

closed 2023-06-20 by Alvin1Zhang