espressif/esp-idf #8605: partition_ops example does not work (boot looping)

rush0815, 2022-03-18:
I am trying to run the partition_ops example as it is. I have just created a new project in VS code (platformIO), copy-pasted the content of main.c and created the file partitions_examples.csv with its content give in the example file.
Additionally I have configured the platformio.ini for my board. I am using the Lolin D32 Pro Dev board.

Compiling works, but the program ends up in a boot loop with that error:

`
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m

assert failed: app_main main.c:27 (partition != NULL)

Backtrace:0x400813da:0x3ffb5e10 0x40085ff1:0x3ffb5e30 0x4008aed1:0x3ffb5e50 0x400d0aaf:0x3ffb5f70 0x400e38dc:0x3ffb5f90 0x400889a1:0x3ffb5fb0

ELF file SHA256: 4a0e66022a85148b

Rebooting...
`

These lines are causing the failure but I don't understand why *partition ist NULL 

`
const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "storage");
assert(partition != NULL);
`

Any Idea what I am doing wrong?

Kind regards... Rush

igrr, 2022-03-18:
Hi @rush0815,
When creating the issue, you have been presented with the issue template. Information requested there, such as IDF version is useful for us to provide you with support in solving this issue. Would you mind filling in the requested info?

In this case it seems that the missing piece is to actually make the build system use the partition table file you have copied. Just putting the file into the project directory is not sufficient. I can't tell you how to do this with PlatformIO, but in ESP-IDF you would set it in menuconfig. This is described here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#overview

rush0815, 2022-03-18:
Yes sorry, the platformIO esp32 platform version is this one (v 3.5.0) which includes esp-idf version 4.3.2:

https://github.com/platformio/platform-espressif32/releases

I also ran menueconfig and configured the partition table. That configuration has been wirtten into sdkconfig.lolin_d32_pro. The partitioning part looks like the one stated in the examples 

CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_example.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions_example.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x8000

I just try to rename the partition_example.csv to something other so that the configuration should not find the file and should give an error. But it compiles without any issue. Thats strange...  It seems that the partitioning options set by menueconfing are simply ignored.

igrr, 2022-03-18:
> Thats strange... It seems that the partitioning options set by menueconfing are simply ignored.

Indeed. Please check this section of Platformio docs: https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables

rush0815, 2022-03-18:
Well, I did not set

board_build.partitions in platformio.ini

So now it works, thanks for that hint! :-)

But for what is the partitioning config section in the sdkconfig if it has no functionality ?

igrr, 2022-03-18:
Not sure about that, sorry. I'll probably have to ask about that somewhere in platformio community forums or the issue tracker.

My guess is that:
1. for arduino-esp32 framework, partition tables can be adjusted, but sdkconfig is fixed. Hence it was implemented in the ini file.
2. later, sdkconfig modification support support for esp-idf in platformio was added, but the partition table settings were already implemented in the ini file.

closed 2022-03-18 by igrr