espressif/esp-idf #8608: Problem configuring UART Hal for Bootloader hook

mauriciofarina, 2022-03-18:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (DevKitC): v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.4
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Linux|macOS
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Hello Espressif team,

We are developing a bootloader hook for product to receive serial messages, but we are having some trouble using the UART Hal library to make use of it. Right now, we are trying to make this work on the [bootloader_hooks](https://github.com/espressif/esp-idf/tree/v4.4/examples/custom_bootloader/bootloader_hooks) example. However, we having some trouble understanding the library and with CMake. We already included `PRIV_REQUIRES hal` to the `my_boot_hooks` CMake file, but some include functions are still not been found. Would be possible for you to provide us a simple example of how to initialize and communicate using the UART Hal for this hook?

negativekelvin, 2022-03-19:
Have you tried using rom functions?

o-marshmallow, 2022-03-21:
Hi @mauriciofarina,

As @negativekelvin said, it is possible to use ROM functions within the bootloader, that's the simplest solution:
```
#include "esp_log.h"
#include "esp_rom_uart.h"

void bootloader_after_init(void) {
    uint8_t str[16];
    ESP_LOGI("HOOK", "This hook is called AFTER bootloader initialization");
    /* Receive 16 bytes from the UART */
    esp_rom_uart_rx_string(str, sizeof(str));
    esp_rom_printf("String received: %s\n", str);
}
```
If you add such behavior to the `bootloader_before_init` hook, keep in mind that the watchdog has not been disabled/configured yet, so you would get such error:
```
entry 0x400806a8
I (66) HOOK: This hook is called BEFORE bootloader initialization
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15148
ho 0 tail 12 room 4
load:0x40080400,len:3900
0x40080400: _init at ??:?
```

If for some reasons you don't want to use ROM functions, you can still use `ll` functions:
```
#include "esp_log.h"
#include "hal/uart_ll.h"
#include "esp_rom_uart.h"

#ifndef MIN
#define MIN(a,b) ((a) > (b) ? (b) : (a));
#endif

void bootloader_after_init(void) {
    ESP_LOGI("HOOK", "This hook is called AFTER bootloader initialization");
    uint32_t total = 0;
    uint8_t buf[8] = { 0 };
    /* Keep one free byte to keep the ending \0 byte */
    const uint32_t torcv = sizeof(buf) - 1;

    while (total < torcv) {
        /* Check if we've received bytes on UART0 */
        uint32_t len = uart_ll_get_rxfifo_len(UART_LL_GET_HW(0));
        if (len == 0)
            continue;
        len = MIN(len, torcv - total);
        uart_ll_read_rxfifo(UART_LL_GET_HW(0), buf + total, len);
        total += len;
    }
    esp_rom_printf("Received %d chars, %s\n", total, buf);
```
I haven't initialized the UART here as it has already been initalized (by ROM and/or bootloader), but if you want to reconfigure it or initialize it yourself, you can call the following functions:
```
void bootloader_after_init(void) {
    ESP_LOGI("HOOK", "This hook is called AFTER bootloader initialization");
    uart_ll_set_sclk(UART_LL_GET_HW(0), UART_SCLK_APB);
    // Set default baud: 115200, use APB clock.
    const uint32_t baud_def = 115200;
    uart_ll_set_baudrate(UART_LL_GET_HW(0), baud_def);
    // Set UART mode.
    uart_ll_set_mode(UART_LL_GET_HW(0), UART_MODE_UART);
    // Disable UART parity
    uart_ll_set_parity(UART_LL_GET_HW(0), UART_PARITY_DISABLE);
    // 8-bit word
    uart_ll_set_data_bit_num(UART_LL_GET_HW(0), UART_DATA_8_BITS);
    // 1-bit stop bit
    uart_ll_set_stop_bits(UART_LL_GET_HW(0), UART_STOP_BITS_1);
    // Set tx idle
    uart_ll_set_tx_idle_num(UART_LL_GET_HW(0), 0);
    // Disable hw-flow control
    uart_ll_set_hw_flow_ctrl(UART_LL_GET_HW(0), UART_HW_FLOWCTRL_DISABLE, 100);
}
```

mauriciofarina, 2022-03-21:
Hello Everyone!

Thank you for your help @negativekelvin and @o-marshmallow!

I got able to access the serial ports using both methods. However, to access uart 1 it was also needed to configure the gpio pins for it:

```c

#define UART    1
#define UART_TX 17
#define UART_RX 16

static void init_boot_serial(void) {

    esp_rom_gpio_connect_out_signal(UART_TX, UART_PERIPH_SIGNAL(UART, SOC_UART_TX_PIN_IDX), 0, 0);
    esp_rom_gpio_connect_in_signal(UART_RX, UART_PERIPH_SIGNAL(UART, SOC_UART_RX_PIN_IDX), 0);

    uart_ll_set_sclk(UART_LL_GET_HW(UART), UART_SCLK_APB);
    uart_ll_set_baudrate(UART_LL_GET_HW(UART), 115200);
    uart_ll_set_mode(UART_LL_GET_HW(UART), UART_MODE_UART);
    uart_ll_set_parity(UART_LL_GET_HW(UART), UART_PARITY_DISABLE);
    uart_ll_set_data_bit_num(UART_LL_GET_HW(UART), UART_DATA_8_BITS);
    uart_ll_set_stop_bits(UART_LL_GET_HW(UART), UART_STOP_BITS_1);
    uart_ll_set_tx_idle_num(UART_LL_GET_HW(UART), 0);
    uart_ll_set_hw_flow_ctrl(UART_LL_GET_HW(UART), UART_HW_FLOWCTRL_DISABLE, 100);
}
```

Is this implementation correct or there is a better/proper way of doing this?

o-marshmallow, 2022-03-22:
Hi @mauriciofarina ,

I am glad to hear it is working for you! Initializing the UART1 doesn't surprise me, the bootloader and the ROM code don't use it by default, so they shouldn't initialize it.

That's the proper way to do, this is exactly how the UART initialization is done in the HAL layer.

closed 2022-03-22 by mauriciofarina