espressif/esp-idf #8609: region `iram0_0_seg' overflowed by x bytes with lates git pull

listout, 2022-03-19:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-2137-g2f9d47c708
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Getting region `iram0_0_seg' overflowed by x bytes with lates git pull but not on tag v4.2.3. I've a code that I'm building on latest v5 of idf and after yesterday's commit I'm getting this issue. I don't know what or where the issue is cause my code was building fine up until the day before yesterday.
 
### Expected Behavior

Project builds without any error

### Actual Behavior

Project fails to build with "section `.iram0.text' will not fit in region `iram0_0_seg`"

### Steps to reproduce

### Code to reproduce this issue

## Debug Logs

```
/home/listout/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: walk.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/listout/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/listout/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 3748 bytes
```

listout, 2022-03-19:
Output of `idf.py size`

```
Total sizes:
Used static DRAM:   73064 bytes (  51516 remain, 58.6% used)
      .data size:   23568 bytes
      .bss  size:   49496 bytes
Used static IRAM:  134818 bytes (  -3746 remain, 102.9% used) Overflow detected!
      .text size:  133791 bytes
   .vectors size:    1027 bytes
Used Flash size : 1133926 bytes
      .text     :  890023 bytes
      .rodata   :  243647 bytes
Total image size: 1292312 bytes (.bin may be padded larger
```

listout, 2022-03-19:
I was able to fix this by changing optimization level to `-Os`. It was on `-Og` by default. Did something change in the latest merge?

listout, 2022-03-19:
Okay I was able to figure out what was taking so much IRAM. I was using nimble SPP and it alone takes 74% of the IRAM. Wasn't nimBLE supposed to be small?

moefear85, 2022-03-19:
if it is resolved, please close this issue

closed 2022-03-20 by listout

ESP-Marius, 2023-04-11:
On latest master we've added a kconfig option for using parts of SRAM1 as IRAM in https://github.com/espressif/esp-idf/commit/5cbd311ecf0b381a0711ef8a96980a67b30a135f which will help with this issue.

We've also focused on reducing IRAM usage/making placement optional for several components. For an full overview of all such option see [Optimizing-iram-usage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage). All new options for saving IRAM we add will be listed in this chapter.