espressif/esp-idf #8610: OTA - Update fail with checksum error

mihammouda, 2022-03-19:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version:V4.1
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.3.2
- Build System: idf.py
- Operating System: Windows
- vscode ide
- Power Supply: USB

## Problem Description
Ota update fail with checksum error:
E (99212) esp_image: Checksum failed. Calculated 0xd2 read 0x7f

 C:\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\esp_4_3_2\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 image_info build/main.bin
esptool.py v3.2-dev
Image version: 1
Entry point: 40081408
6 segments

Segment 1: len 0x26348 load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x03544 load 0x3ffbdb60 file_offs 0x00026368 [BYTE_ACCESSIBLE,DRAM]
Segment 3: len 0x0675c load 0x40080000 file_offs 0x000298b4 [IRAM]
Segment 4: len 0x8d9b0 load 0x400d0020 file_offs 0x00030018 [IROM]
Segment 5: len 0x112f0 load 0x4008675c file_offs 0x000bd9d0 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x000cecc8 [RTC_DATA]
Checksum: 7f (valid)
Validation Hash: 8c6282e4b48c7e4eb890c7753918affde045c8e35349e3f07611f4b552738a7e (valid)

 C:\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\esp_4_3_2\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 image_info ota.bin
esptool.py v3.2-dev
Image version: 1
Entry point: 40081408
6 segments

Segment 1: len 0x26348 load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x03544 load 0x3ffbdb60 file_offs 0x00026368 [BYTE_ACCESSIBLE,DRAM]
Segment 3: len 0x0675c load 0x40080000 file_offs 0x000298b4 [IRAM]
Segment 4: len 0x8d9b0 load 0x400d0020 file_offs 0x00030018 [IROM]
Segment 5: len 0x112f0 load 0x4008675c file_offs 0x000bd9d0 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x000cecc8 [RTC_DATA]
Checksum: 7f (invalid - calculated d2)
Validation Hash: 7327b06b8e93718fa44a68fe11e079b5b7e1f9e983258eb433ba6fe6224cba8d (invalid)

moefear85, 2022-03-19:
this is not a bug with esp-idf. you have some issues with the chip or its flash, most likely electrical (insufficient power), or you've worn out the flash so it has bad sectors. try a different chip first, and verify the voltage is sufficient. moreover, such troubleshooting issues go on the [esp forum](https://www.esp32.com/), not here.

dobairoland, 2022-03-21:
Are you able to verify this on another ESP32?

dobairoland, 2022-04-04:
Closed for inactivity.

closed 2022-04-04 by dobairoland