espressif/esp-idf #8611: ability to peek data from a streambuffer or queue

moefear85, 2022-03-19:
**Is your feature request related to a problem? Please describe.**

I need the ability to repeatedly re-read from the "recieve" end of a freertos streambuffer, and then only selectively pop off that data at an arbitrarily chosen moment. To my knowledge, this is a very frequently needed feature with any queue/buffer. I don't know why freertos didn't incorporate this ability to begin with.

**Describe the solution you'd like**

Is it feasible for espressif to patch its freertos fork and include the required peeking capability for stream buffers and every type built on top of it (or atleast message queues)?

**Describe alternatives you've considered**

Any alternative I can think of is unnecessarily complicated. I would need to either create a second buffer and move the data around back and forth, which in turn would require alot of complicated mutexes & if-statements for all senders to the buffer, or I would need to use a ringbuffer, which to my knowledge is always static in size? This means I am using freertos streambuffers under the assumption that when data is recieved from it, its size (and the amount of memory occupied by and dedicated to it) shrinks. Is this assumption correct?

**Additional context**

As I need this feature immediately, I will in parallel look into the freertos implementation, and see if I can write a quick hack/patch. If I have something usable, I'll share it here.

igrr, 2022-03-21:
Hi @moefear85,

Stream buffers are an upstream FreeRTOS feature, I would recommend you opening a feature request at https://github.com/FreeRTOS/FreeRTOS-Kernel/issues instead. Currently we are in the process of reducing the number of differences between the upstream FreeRTOS and our version, so it is unlikely that we will add any significant functionality which isn't present upstream.

(You have probably already found it, but queues do support the "peek" operation: [xQueuePeek](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv410xQueuePeek13QueueHandle_tPCv10TickType_t))

> Update: Reading the docs, I realize there is a xRingbufferCreateStatic(). Does this mean that xRingbufferCreate() creates a dynamically sized ringbuffer?

The difference between xRingbufferCreateStatic and xRingbufferCreate is whether the memory for the buffer is provided by the caller (`xRingbufferCreateStatic` version) or allocated from heap (`xRingbufferCreate`). In both cases the size has to be specified at the time when the ringbuffer is created.

moefear85, 2022-03-21:
ok i'm dumbfounded here. i literally searched the sourcebfile for the word "peek" but found nothing. either way thx for the tip, as i would prefer using unpatched sources