espressif/esp-idf #8612: HTTP Restful API Server Example - error in recv : 104

riegaz, 2022-03-19:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes]: Visual Studio Code
- Power Supply:         [USB]

## Problem Description

I took the HTTP Restful API Server example from here: https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/restful_server
I changed the following lines to trigger GET requests every 5 seconds:

rest_server.c
I added the following lines at line 152:
```
cJSON_AddStringToObject(root, "version", IDF_VER);
cJSON_AddNumberToObject(root, "cores", chip_info.cores);
cJSON_AddStringToObject(root, "version1", IDF_VER);
cJSON_AddNumberToObject(root, "cores1", chip_info.cores);
cJSON_AddNumberToObject(root, "version2", a);
cJSON_AddNumberToObject(root, "cores2", b);
```
Home.vue
I changed solely the last lines of this file to trigger get requests every 5 seconds:
```
  methods: {
  pollFunc(fn, timeout, interval) {
        var canPoll = true;
          var startTime = (new Date()).getTime();
          interval = interval || 1000,

          (function p() {
              //canPoll = ((new Date).getTime() - startTime ) <= timeout;
              if (!fn() && canPoll)  { // ensures the function exucutes
                  setTimeout(p, interval);
              }
          })();
      },
retriveDataAmp () {
      this.$ajax.get("/api/v1/system/info")
      .then(data => {
        console.log(data.data)
        this.version = data.data.version;
        this.cores = data.data.cores;
        this.version1 = data.data.version1;
        this.cores1 = data.data.cores1;
        this.version2 = data.data.version2;
        this.cores2 = data.data.cores2;
      })
      .catch(error => {
        console.log(error);
      });
      },
  },
  mounted() {
    this.pollFunc(this.retriveDataAmp, 60000, 5000);
  }
};
</script>
```
### Expected Behavior
Smoothly running rest/http server

### Actual Behavior
Partially smooth running until this error occurs:
```
W (1277626) httpd_txrx: httpd_sock_err: error in recv : 104
```
### Steps to reproduce

1. Get this example: https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/restful_server
2. Set your wifi ssid and password and set the the target for the HTML content to "Deploy website to SPI Nor Flash" in menuconfig
3. Flash the bin file as is to your esp32.
4. Open the website http://esp-home.local on two browsers.
5. Wait for this error to occur, refresh the pages from time to time.
6. Once the browsers start to have issues during refreshing the page (should occur after 30 minutes of waiting) the "error in recv: 104" is near.

### Code to reproduce this issue

[code.zip](https://github.com/espressif/esp-idf/files/8310283/code.zip)

riegaz, 2022-03-19:
UPDATE:
This also happens on the plain example. Just reload the page from time to time.
Once the website isn't really loading completely anymore, a restart fixes this issue.

hmalpani, 2022-04-04:
Hello @riegaz,
Thanks for reporting the issue. As of now, I am unable to reproduce the issue. Can you please share the Debug logs

Alvin1Zhang, 2022-04-12:
@riegaz Thanks for reporting, would you please help share the Debug logs? Thanks.

closed 2022-05-06 by Alvin1Zhang