espressif/esp-idf #8614: Error in task.c in xTaskCreatePinnedToCore

GermanSheepDog, 2022-03-20:
## Environment

- Development Kit:    M5Core2
- Module or chip used:  ESP32-D0WDQ6-V3
- IDF version:  ESP-IDF v4.4 
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     macOS
- (Windows only) environment type: -
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
I used the esp-idf example */esp-idf/examples/protocols/http_server/file_serving standalone
and it worked fine. Then I integrated the Example in a whole program and I get this error if
I started FileServer in line httpd_start(&server, &config). The error code is ESP_ERR_HTTPD_TASK
or "Failed to launch server task/thread").

//Detailed problem description goes here.
The critical part is in task.c in function 'xTaskCreatePinnedToCore' behind 
'#if ( configSUPPORT_DYNAMIC_ALLOCATION == 1 )'. I build in some prints:
```
             /* Allocate space for the stack used by the task being created. */
            //I changed! pxStack = pvPortMallocStackMem( ( ( ( size_t ) usStackDepth ) * sizeof( StackType_t ) ) ); /*lint !e9079 All values returned by pvPortMalloc() have at least the alignment required by the MCU's stack and this allocation is the stack. */
            int32_t new_size =  (size_t) usStackDepth * sizeof(StackType_t);
            //This worked!  pxStack = malloc(new_size); /*lint !e9079 All values returned by pvPortMalloc() have at least the alignment required by the MCU's stack and this allocation is the stack. */
            pxStack = pvPortMallocStackMem(new_size);
printf("Requested size: %d\n", new_size );
printf("heap_caps_get_free_size:          %d\n", heap_caps_get_free_size(MALLOC_CAP_8BIT));
printf("heap_caps_get_minimum_free_size:  %d\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT));
printf("heap_caps_get_largest_free_block: %d\n", heap_caps_get_largest_free_block(MALLOC_CAP_8BIT));
                if( pxStack != NULL )
                {
                    /* Allocate space for the TCB. */
                pxNewTCB = ( TCB_t * ) pvPortMallocTcbMem( sizeof( TCB_t ) ); /*lint !e9087 !e9079 All values returned by pvPortMalloc() have at least the alignment required by the MCU's stack, and the first member of TCB_t is always a pointer to the task's stack. */
printf("Ok - get size\n"  );
                    if( pxNewTCB != NULL )
                    {
                        /* Store the stack location in the TCB. */
                        pxNewTCB->pxStack = pxStack;
                    }
                    else
                    {
                        /* The stack cannot be used as the TCB was not created.  Free
                         * it again. */
                        vPortFree( pxStack );
                    }
                }
                else
                {
                    pxNewTCB = NULL;
printf("ERROR in get Size\n"  );
                }
            }
        #endif /* portSTACK_GROWTH */
```

### Expected Behavior
I expected this output of my program. I used the line  'pxStack = malloc(new_size);'.
```
...
Start 'xTaskCreatePinnedToCore'
Requested size: 4096                                      // <- This is for file server!!!
heap_caps_get_free_size:          4191910
heap_caps_get_minimum_free_size:  4191318
heap_caps_get_largest_free_block: 4128768
Ok - get size
```

### Actual Behavior
[... 133 lines cut ...]
heap_caps_get_minimum_free_size:  4195226
heap_caps_get_largest_free_block: 4128768
ERROR in get Size
error: COULD_NOT_ALLOCATE_REQUIRED_MEMORY
E (6304) pg_file_server: File Server: Failed to launch server task/thread

```

### Steps to reproduce

The error occurred, if I used 'xTaskCreatePinnedToCore' 12 times with the parameters
showed above.

### Code to reproduce this issue
Please see above

### Debug Logs
Please see above

## Other items if possible

- [x ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8311020/sdkconfig.txt)

negativekelvin, 2022-03-20:
You need to check MALLOC_CAP_INTERNAL for task stack memory

GermanSheepDog, 2022-03-20:
Hello @negativekelvin.
thank you for the quick answer. In sdkconfig and in idf.py menuconfig I found no  MALLOC_CAP_INTERNAL.
Only in esp-idf directory I found at esp_heap_caps.h:
```
#define MALLOC_CAP_INTERNAL         (1<<11) ///< Memory must be internal; specifically it should not disappear when 
```
What can I check here? It is a constant value.

igrr, 2022-03-20:
I think @negativekelvin meant that when checking the available heap size, you need to check just the size of internal RAM (DRAM) rather than all the available heap. Task stacks can only be placed into internal RAM. You might be running out of internal RAM, even though there is still plenty of external RAM available. If that is the case, task stack allocation will fail.

To check the amount of internal RAM, change your debugging code as follows:
```c++
printf("heap_caps_get_free_size:          %d\n", heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL));
printf("heap_caps_get_minimum_free_size:  %d\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL));
printf("heap_caps_get_largest_free_block: %d\n", heap_caps_get_largest_free_block(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL));
```

GermanSheepDog, 2022-03-21:
Hello @igrr ,

thank you for your informations. I added the three lines to my debugging code and the result is:
´´´
Start 'xTaskCreatePinnedToCore'
Requested size: 4096
heap_caps_get_free_size:          4160218
heap_caps_get_minimum_free_size:  4159434
heap_caps_get_largest_free_block: 4128768
heap_caps_get_free_size (internal):          2195
heap_caps_get_minimum_free_size(internal):  1411
heap_caps_get_largest_free_block(internal): 496
Ok - get size
´´´ 
I used this version of code in task.c:
```
int32_t new_size =  (size_t) usStackDepth * sizeof(StackType_t);
pxStack = malloc(new_size);
//pxStack = pvPortMallocStackMem(new_size);
```
There is not enough size, but the pxNewTCB is not NULL! Why?
How can I get (enough) space?

negativekelvin, 2022-03-21:
You changed it to malloc which bypassed the restriction to internal memory which is not recommended because it may cause a crash later.  See https://github.com/espressif/esp-idf/blob/2f9d47c708f39772b0e8f92d147b9e85aa3a0b19/components/esp32/Kconfig#L330-L341

GermanSheepDog, 2022-03-21:
Ok, I changed back my code to the regular statement and I changed SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY :
```
pxStack = pvPortMallocStackMem(new_size);

CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y
```
I got my error:
```
Start 'xTaskCreatePinnedToCore'
Requested size: 4096
heap_caps_get_free_size:          4164266
heap_caps_get_minimum_free_size:  4163710
heap_caps_get_largest_free_block: 4128768
heap_caps_get_free_size (internal):          1991
heap_caps_get_minimum_free_size(internal):  1435
heap_caps_get_largest_free_block(internal): 312
ERROR in get Size
error: COULD_NOT_ALLOCATE_REQUIRED_MEMORY
E (9848) pg_file_server: File Server: Failed to launch server task/thread
```

GermanSheepDog, 2022-03-21:
Hello @igrr and @negativekelvin,
I changed two wifi-parameters in sdkconfig by halve the values:
```
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=5
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=16
```
Not it worked fine. Thank you very much.

closed 2022-03-21 by GermanSheepDog

igrr, 2022-03-21:
For future reference, this documentation page lists a number of things which can be done to reduce heap usage. It might be useful in case anyone else runs into this issue: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#reducing-heap-usage