espressif/esp-idf #8617: esp32 is not booting up any more

holznan, 2022-03-21:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    // v4.4
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Yes (please give details)]
- Power Supply:         [USB]

## Problem Description

when the esp is powered the application on the chip doesn´t start any more. 
when I have a look at the serial terminal there is already something wrong at the start of the controller. 
The load is not working correct any more.

This problem i only have with on chip! An other chip with the exact same application is working correct (see expected behavior).

### Expected Behavior
[image]

### Actual Behavior
[image]

franz-ms-muc, 2022-03-21:
espefuse.py -p COM7 set\_flash\_voltage 3.3V

when there is a SD-Card connected we execute the above command on every Board. 

@holznan  please check. 

also check: https://fliegl.atlassian.net/wiki/spaces/GE/pages/86507521/Pr+fung+TECU4+on+ESP32

o-marshmallow, 2022-03-22:
Hi @holznan ,

Sounds like an image corruption or defective flash: `0x07800000` is not a valid address and the length is way too big.
I guess you have already tried to reflash your binary, right? Did you also reflash the bootloader? The problem still happens?

You said you have multiple chips. Let's try something to check your SPI flash.
What I would suggest you to do is to flash the exact same image/application to your two devices:
```
idf.py flash monitor
```
Then dump their respective SPI flash (4MB):
```
# Plug the working device
esptool.py read_flash 0 4194304 working.img
# Plug the non-working device
esptool.py read_flash 0 4194304 nonworking.img
``` 
Check if the files are similar, the easiest way is by using `md5sum`:
```
md5sum working.img nonworking.img
``` 
If the resulted hashes are different, then the SPI flash of the non-working device is defective. Else, it is a problem with the bootloader which cannot manage to communicate with the SPI flash on boot.

closed 2022-05-06 by Alvin1Zhang