espressif/esp-idf #8619: Nimble spp_server and spp_client Apps (Esp32c3)

iamcalledusman, 2022-03-21:
- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:  [ESP32-C3-DevKitM-1 v1.0]
- Kit version              (DevKitM): [v1]
- Module or chip used:  [ESP32-C3-MINI-1]
- IDF version (ESP-IDF v4.4-dev-3224-g23a03a3bd)
- Build System:         [idf.py]
- Compiler version (xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0)
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (VSCODE Studio)]
- Power Supply:         [USB]

**## Problem Description**

I am using esp-idf (vscode studio) NIMBLE , SPP_Client and SPP_Server application . It works perfectly for only 1 device and if increase the number of SPP_Client device then still connects only 1 device. I am using three esp32c3 for this testing and one nrf connect android app as well.As per the application configuration we can connect 10 esp32c3 devices and NIMBLE documentation we can connect 32 concurrent connections.But practically when i tested with esp-idf command line and as well as the VScode studio as an IDE in both the cases facing the same issue of connecting maximum 1 Device (SPP_Client) with (SPP_Server).

### Expected Behavior

It should be connected to at least 10 devices (esp32c3) and 32 devices if we go follow the nimble documentation 

https://github.com/espressif/esp-nimble#readme

### Actual Behavior

Its connects only 1 device (esp32c3 ) .For client app i am using (SPP_Client ) and for server app(SPP_Server).

### Steps to reproduce

1. Go to examples then bluetooth then nimble then ble_spp then  SPP_Server
2. Open the SPP_Server example in Vscode studio OR by using esp-idf command prompt
3. Compile and Load into Device # 1 (esp32c3)
4. Open Monitor window to see the output.
5. Repeat the steps from 1 to 4 as mentioned above for the SPP_Client
6. Load the SPP_client Code into Device # 2 (Esp32c3)
7. I have attached the two pictures , one for server and other for client.
8. You can see the actual behaviour as i mentioned above.
9. I have pasted the screen shots to show my testing and issue i am facing. Below image is the two esp32c3 , one is SPP_server and second os SPP_Client.

[image]

10.This is the image when only one device connected!

[image]

You can see the peer addr is 7c: df: a1:43:0a:76.Altough I am able to connect second device as well but after disconnecting the first one.

11.Below image showing the second device connected when I disconnected the first one as mentioned above. Device address is also highlighted in the below image.

[image]

You can see the peer addr is 5d:42:6d:10:29:a4

12.In this image you can see the menuconfig setting as per mentioned by the documents.

[image]

13.This image is the BT_CTRL_BLE_MAX_ACT setting for esp32c3 which is 10.

[image]

14.Here is the image of nrf connect App. You can see that it goes to connecting/busy state when one device already connected to the spp_server.

[image]

15.I also tried the code with ESP-IDF command terminal but same behaviour as i mentioned in above points.

[image]

16.**Important note and observation**. After the setting I saved compiled and load the program but same results. I also tried **Arduino_NimBLE library with arduino** it works perfect and I am able to connect 4 devices without any issue. But in this **esp-idf Firmware** stack I am unable to connect more than 1 device. 

Arduino Nimble server 
[image]

[image]

[image]

Docklight image showing the output of arduino_nimble /client and server 

[image]

### Code to reproduce this issue
To reproduce the code , i have mentioned above the detail to reproduce / test the code. Code is same as provided by example no changes made in actual code.

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8318253/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

- [ ] coredump (This provides stacks of tasks.)

rahult-github, 2022-03-23:
Hi @iamcalledusman ,

 Currently, a custom BLE service to show SPP ( a BR-EDR profile ) using NimBLE stack  has been added.  We will take this up as a feature request to have support for multiple profile connections and enhance the existing application and plan for it. 

Thanks,
 Rahul

iamcalledusman, 2022-03-24:
> Hi @iamcalledusman ,
> 
> Currently, a custom BLE service to show SPP ( a BR-EDR profile ) using NimBLE stack has been added. We will take this up as a feature request to have support for multiple profile connections and enhance the existing application and plan for it.
> 
> Thanks, Rahul

Hi @rahult-github 

Thank you for the response! we will wait for it !

IshaESP, 2022-03-31:
Hi @iamcalledusman ,

We have added support for multiple profile connection in NimBLE SPP. For ESP32C3, it supports maximum number of connections between 9 devices( 1 client and 8 servers ) as mentioned in the SPP example's menuconfig. 

Can you please try this patch out?
[feature_spp_multiple_profile.zip](https://github.com/espressif/esp-idf/files/8388293/feature_spp_multiple_profile.zip)

iamcalledusman, 2022-03-31:
Hi @IshaESP 

Thank you so much i will try and update on the forum as well.

Regards.,
Muhammad Usman

rahult-github, 2022-05-23:
Hi @iamcalledusman ,
 
Did you get chance to test the above ?. any comments ?

Thanks,
 Rahul

iamcalledusman, 2022-05-23:
Hi Rahul!

Sorry, I am unable to test it further. because my client changed the board from esp32c3 to nrf52820.
So that is the reason that I have to change the direction!.

Thank you!
Muhammad Usman

rahult-github, 2022-06-01:
Feature has been added as part of IDF and is available on github:  6a27983fc9f8db88fb76ea2c5c88b4dec536ee51

closed 2022-06-01 by rahult-github