espressif/esp-idf #8620: Feature Request: Merge LCD support improvements into v4.4

kmatch98, 2022-03-21:
**Is your feature request related to a problem? Please describe.**

I am working to add support for ESP32-S3 LCD RGB displays into [CircuitPython](https://circuitpython.org).  The recent updates to LCD support on the master branch correct some major issues [in the existing v4.4 `esp_lcd` library](https://github.com/espressif/esp-idf/tree/release/v4.4/components/esp_lcd).  These recent updates are necessary to complete that support into CircuitPython.  CircuitPython is currently running only on v4.4 of the ESP-IDF.

**Describe the solution you'd like**

- Please merge the LCD updates from @suda-morris into the version 4.4 of the ESP-IDF.

**Describe alternatives you've considered**

- One option is to run CircuitPython on a fork of ESP-IDF v4.4.  It has been done before, but we would prefer to stay on the release v4.4.
- One option is to use the existing `esp-lcd` library in v4.4, but there are bugs that cause poor performance.

**Additional context**

Here are some of the commits that are relevant to improving the use of LCD display using the ESP32-S3 LCD peripheral.

[image]

suda-morris, 2022-03-22:
@kmatch98 Thank you for reminding, yes we will backport these fixes to 4.4

itavero, 2022-04-11:
@suda-morris Any indication on when why might expect these changes to be backported?

I finally had the example working on `v4.4.0`, but the performance/refresh rate was poor at had some glitches.
I tried upgrading to `master` (also removed the `on_frame_trans_done` callback from the config and added `lv_disp_flush_ready` to the `example_lvgl_flush_cb`, as is done in the example on the `master` branch).
Unfortunately, I no longer get any image on my display.

As you can imagine, because of this, we'd really like to know when we can expect a "more stable" version to be released.

suda-morris, 2022-04-21:
@kmatch98 lcd improvement has been backported. See https://github.com/espressif/esp-idf/commit/213da04fcd51c876767f675b9e0ce03c0e4facaa

@itavero Hi, do you mean the master RGB LCD example can't work on your side? If so, could you open another Github issue (as this one is a request for backporting) and post your init code there?

closed 2022-04-21 by suda-morris