espressif/esp-idf #8621: I2C module crash due to writing to NULL pointer

luis0visa, 2022-03-22:
## Environment

- Development Kit:     Custom
- Kit version: Custom
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    // v4.4-348-gc45dee0a7a
- Build System:        CMake|idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes, eclipse with IDF plugin
- Power Supply:         Regulated 3.3V

## Problem Description

Tracked down the issue to the i2c.c file, where the code is trying to write a structure using a pointer without verifying if it's NULL.
Found a fix that was done for the same isse in the same function at https://github.com/espressif/esp-idf/commit/0cfa7ca20f40ac481495d4312953329ec5b00960

```c
static void IRAM_ATTR i2c_master_cmd_begin_static(i2c_port_t i2c_num)
{
    i2c_obj_t *p_i2c = p_i2c_obj[i2c_num];
    portBASE_TYPE HPTaskAwoken = pdFALSE;
    i2c_cmd_evt_t evt = { 0 };
    if (p_i2c->cmd_link.head != NULL && p_i2c->status == I2C_STATUS_READ) {
        i2c_cmd_t *cmd = &p_i2c->cmd_link.head->cmd;
        i2c_hal_read_rxfifo(&(i2c_context[i2c_num].hal), cmd->data + cmd->bytes_used, p_i2c->rx_cnt);
        /* rx_cnt bytes have just been read, increment the number of bytes used from the buffer */
        cmd->bytes_used += p_i2c->rx_cnt;

        /* Test if there are still some remaining bytes to send. */
        if (cmd->bytes_used != cmd->total_bytes) {
            p_i2c->cmd_idx = 0;
        } else {
            p_i2c->cmd_link.head = p_i2c->cmd_link.head->next;
            p_i2c->cmd_link.head->cmd.bytes_used = 0;
        }
```

The line where is crashing is the next one, that is located in the last else (line 1289 in my I2C.c file)
p_i2c->cmd_link.head->cmd.bytes_used = 0;

### Expected Behavior
Execute the code only when the pointer is not NULL

### Actual Behavior
Trying the write the bytes_used field when head is NULL

### Steps to reproduce

Not pretty sure, the issue only appeared when an I2C slave was connected to the I2C bus. And the code being executed would be this one
I fixed it by copying the fix in the mentioned commit in that line
```cpp
p_i2c->cmd_link.head = p_i2c->cmd_link.head->next;
if(p_i2c->cmd_link.head) {
    p_i2c->cmd_link.head->cmd.bytes_used = 0;
}
```

### Code to reproduce this issue

Part of the code, can't copy the whole code due to company's NDA

```cpp

    /* Create cmd link to store the streaming bits */
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();

    /* Set the start bit */
    i2c_master_start(cmd);

    /* Send the slave address with the read bit set and ack verification enabled */
    i2c_master_write_byte(cmd, (address << 1) | READ_BIT, ACK_CHECK_EN);

    /* Read the first byte containing the packet size */
    i2c_master_read_byte(cmd, data_rd, ACK_VAL);

    /* Execute the I2C communication */
    ret = i2c_master_cmd_begin(I2C_MASTER_NUM, cmd, I2C_TIMEOUT_MS/portTICK_PERIOD_MS);

```
## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
```

AxelLin, 2022-03-22:
https://github.com/espressif/esp-idf/issues/8548

o-marshmallow, 2022-03-22:
Hi @luis0visa ,

As @AxelLin stated, this issue is known and currently being fixed.

In your example though, you forgot the STOP signal at the end (was it intended?), this is why is error is happening. Moreover, if you are only willing to perform simple I2C transactions, you can use the following functions, which are tested and working:
```
esp_err_t i2c_master_write_to_device(i2c_port_t i2c_num, uint8_t device_address,
                                     const uint8_t* write_buffer, size_t write_size,
                                     TickType_t ticks_to_wait);
esp_err_t i2c_master_read_from_device(i2c_port_t i2c_num, uint8_t device_address,
                                      uint8_t* read_buffer, size_t read_size,
                                      TickType_t ticks_to_wait);
esp_err_t i2c_master_write_read_device(i2c_port_t i2c_num, uint8_t device_address,
                                       const uint8_t* write_buffer, size_t write_size,
                                       uint8_t* read_buffer, size_t read_size,
                                       TickType_t ticks_to_wait);
```
[More info in the documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv426i2c_master_write_to_device10i2c_port_t7uint8_tPK7uint8_t6size_t10TickType_t)

luis0visa, 2022-03-22:
Hi there,

You guys are right, my issue is the same that @AxelLin pointed out, I didn't find it when looking for it. 

And @o-marshmallow, yes it was intentional at that point, I updated the code to reproduce the issue to include the rest of the code where the stop is appended, what is being done is to read the slave using a custom "protocol", where the first step is to read one byte where the slave will indicate the number of bytes that it will reply, then the master compares it to the size indicated by the user (to verify that the provided buffer is big enough to get the whole transmission), and then read the slave finishing the communication.

This would be the read function, the equivalent custom write function is kind of the same, but the ESP32 (master) starts sending one byte with the size of the data, finishing the communication, and initiating a new one after waiting some little time (to allow the slave to set up and be ready for the reception. This is done like this due to some issues, even though it's supposed to be like the read (one stream).

This code was working in previous IDF versions though, thank you guys, this issue could be marked as duplicaded or closed as you consider it better.

o-marshmallow, 2022-03-23:
@luis0visa I understand, thanks for the details :+1: 

Closing this issue then

closed 2022-03-23 by o-marshmallow