espressif/esp-idf #8624: Setting compiler flag

domko99, 2022-03-22:
Hi. I am trying to implement sensor MLX90632 and I have a problem with compilation. I was told that I need to add compiler flag (-D) when I am compiling source files of sensor component. I should define `BITS_PER_LONG = 32` as a compiler flag (-D). My component structure is:

````
- components/  - mlx90632/ - CMakeLists.txt
                           - mlx90632_extended_meas.c
                           - mlx90632.c
                           - include/ - mlx90632.h
                                      - mlx90632_extended_meas.h
                                      - mlx90632_depends.h
 ````
I assume that I need to add into  `CMakeLists.txt` compiler flag on source files  `mlx90632_extended_meas.c` and  `mlx90632.c`. 
The command should be  `target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-unused-variable)` to set compiler flag for source files. But I dont know what is the exact command to set `BITS_PER_LONG = 32` as a compiler flag (-D), there is also no info about it in ESP-IDF documentation and I have never added compiler flags in ESP-IDF framework before. I was also defining BITS_PER_LONG value in header file `mlx90632.h`:

````
/* BIT, GENMASK and ARRAY_SIZE macros are imported from kernel */
#ifndef BIT
#define BIT(x)(1UL << (x))
#endif
#ifndef GENMASK
#ifndef BITS_PER_LONG
//#warning "Using default BITS_PER_LONG value"
#define BITS_PER_LONG 32 /**< Define how many bits per long your CPU has */
#endif
#define GENMASK(h, l) \
    (((~0UL) << (l)) & (~0UL >> (BITS_PER_LONG - 1 - (h))))
#endif
#ifndef ARRAY_SIZE
#define ARRAY_SIZE(arr) (sizeof(arr) / sizeof(arr[0])) /**< Return number of elements in array */
#endif
 ````
I dont know if it is needed to set this compiler flag for ESP32 because the sensor is working but I cant write to his EEPROM memory via I2C and not set compiler flag can be one of the problems according to sensor library maker. Link to library is here: https://github.com/melexis/mlx90632-library If there is someone who can help how to solve this issue I would be very glad. Thanks.

igrr, 2022-03-22:
Hi @domko99,
For this you can use [target_compile_definitions CMake function](https://cmake.org/cmake/help/latest/command/target_compile_definitions.html):
```cmake
target_compile_definitions(${COMPONENT_LIB} PUBLIC BITS_PER_LONG=32)
```

Alvin1Zhang, 2022-04-12:
@domko99 Thanks for reporting, would you please help share if any updates? Thanks.

domko99, 2022-04-12:
I solve the issue. Thanks for help.