espressif/esp-idf #8625: idf_tools.py info suggest

zwidny, 2022-03-22:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

tools/idf_tools.py
```python
def download(url, destination):  # type: (str, str) -> None
    info('Downloading {} to {}'.format(os.path.basename(url), destination))
```
`os.path.basename(url)` 这句推荐改为url, 国内下载特别慢， 一般都用下载器下载， 
如果这里是url, 我可以直接复制链接下载了。
但是是basename的话， 我需要等待失败的时候才会有url, 特别烦

dobairoland, 2022-03-22:
Hi @zwidny. If you are looking for the links then you can find them in https://github.com/espressif/esp-idf/blob/master/tools/tools.json.

BTW, by setting `IDF_GITHUB_ASSETS` to `dl.espressif.com/github_assets` you can avoid downloading from Github. More details can be found at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#github-assets-mirror.

zwidny, 2022-03-23:
> Hi @zwidny. If you are looking for the links then you can find them in https://github.com/espressif/esp-idf/blob/master/tools/tools.json.
>
> BTW, by setting `IDF_GITHUB_ASSETS` to [...]

Thx @dobairoland.  Your info are userful, and I can get the url from that. 
But for a newbie, lot of info are from  terminal hint and this is a straightforward way.
If the newbie doest not the info about tools.json and IDF_GITHUB_ASSETS, they can still download the tools too

closed 2022-04-07 by espressif-bot