espressif/esp-idf #8630: v4.4 can't initialize mqtt client

silardg, 2022-03-23:
## Environment

- Development Kit:    ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF Version v4.4
- Build System:         idf.py

On v4.3.1 mqtt connected without a problem, but in the new v4.4 it won't connect.
It can't initialize the mqtt client `client = esp_mqtt_client_init(&config);`.

I haven't changed anything in the configuration (`&config`)

silardg, 2022-03-23:
```
config.event_handle           = eventHandler;                        // add event handler function
    config.host                   = MQTTDriver::host.c_str();            // add the host
    config.uri                    = MQTTDriver::uri.c_str();             // add the URL
    config.port                   = port;                                // add the port
    config.client_id              = MQTTDriver::clientID.c_str();        // add the generated clientID
    config.username               = "";                                  // username blank, added later
    config.password               = "";                                  // password blank, added later
    config.lwt_topic              = MQTTDriver::lastWillTopic.c_str();   // adding lastwill topic
    config.lwt_msg                = LWT_PAYLOAD;                         // LWT payload
    config.lwt_qos                = ONLINE_REPORT_QOS;                   // QOS
    config.lwt_retain             = ONLINE_REPORT_RETAIN;                // retain
    config.lwt_msg_len            = 0;                                   // message length is 0
    config.disable_clean_session  = false;                               // we are not disabling clean sessions
    config.keepalive              = 30;                                  // keeping it alive for 30s
    config.disable_auto_reconnect = false;                               // enable auto reconnect
    config.user_context           = 0;                                   // no user context
    config.task_prio              = 5;                                   // 5 task priority
    config.task_stack             = 20480;                               // stack size
    config.buffer_size            = 8192;                                // buffer size
    config.cert_pem               = NULL;                                // no certification
    config.client_cert_pem        = NULL;                                // no certification for client
    config.client_key_pem         = NULL;                                // no key for client

    // depending on mqtt or mqtts set the transport
    if (strcmp("mqtt", transport.c_str()) == 0) {
        config.transport = MQTT_TRANSPORT_OVER_TCP;
    } else if (strcmp("mqtts", transport.c_str()) == 0) {
        config.transport = MQTT_TRANSPORT_OVER_SSL;
    }

    config.refresh_connection_after_ms = 600000;   // 10 min refresh time
```

silardg, 2022-03-23:
I'm using wolfssl, but I have tried it with the default one too.

chegewara, 2022-03-23:
Yesterday i had similar problem and i am still trying to solve it.
- 2 different devices, on one device exactly the same code is working on another it stuck trying to init mqtt
- when i change mqtt user/pass then code works on both devices
- mqtt transport layer
- arduino 2.0.0, which is some old v4.4 version

Working user/pass is: `test/test`
Not working user/pass:  `zz7c9ebdf8xxxx/zz7c9ebdf8xxxx`

I cant even to diagnose it, because there is no usable backtrace. App just stuck on `esp_mqtt_client_init` for few seconds and then crash.

Thanks

silardg, 2022-03-23:
@chegewara I had the same with two different devices, one working other doesn't. But both had random generated password, but with 123456 password and test username it doesn't work. 

`esp_mqtt_client_init ` just returns NULL

silardg, 2022-03-23:
I will try to add debug messages in mqtt_client.c to see if I can get a rough position of the return NULL.

chegewara, 2022-03-23:
Yeah, it would be great if it return NULL, then i could do something with it, but like i said, it stuck and then crash with no backtrace that can be decoded.

chegewara, 2022-03-24:
Ok, in my case it was probably brownout which was disabled, but it still is very strange that it worked with the same code and only changed mqtt user/pass.

Alvin1Zhang, 2022-04-12:
@silardg Thanks for reporting, would you please help share logs with log level set to debug? Thanks.

Alvin1Zhang, 2022-06-29:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details/updates, thanks.

closed 2022-06-29 by Alvin1Zhang

srk517, 2023-05-26:
Hello I am having same issue with esp_mqtt_client_init(&config) it returns "NULL" everytime. I am not adding password or user name cause i do not have one. Please respond.