espressif/esp-idf #8631: Example of websocket not working on ESP32C3

dreamcmi, 2022-03-23:
## Environment

- Development Kit:      [ESP32-C3-DevKitm-1]
- IDF version: ESP-IDF v4.4-12-gfb1d00a9d1
- Build System: [idf.py]
- Compiler version : riscv32-esp-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System: [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [VSCode]
- Power Supply:         [USB]

## Problem Description

I try to use example/protocols/websocket, just modify the wifi SSID and password, and I get an error when I run it.
```shell
I (4884) esp_netif_handlers: example_connect: sta ip: 192.168.30.147, mask: 255.255.255.0, gw: 192.168.30.1
I (4884) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.30.147
I (5384) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:7edf:a1ff:feb4:ec34, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5384) example_connect: Connected to example_connect: sta
I (5394) example_connect: - IPv4 address: 192.168.30.147
I (5394) example_connect: - IPv6 address: fe80:0000:0000:0000:7edf:a1ff:feb4:ec34, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5404) WEBSOCKET: Connecting to ws://echo.websocket.org...
W (5484) wifi:<ba-add>idx:1 (ifx:0, 28:d1:27:5a:59:c0), tid:0, ssn:0, winSize:64
E (6344) TRANSPORT_WS: Sec-WebSocket-Accept not found
E (6344) WEBSOCKET_CLIENT: Error transport connect
I (6344) WEBSOCKET_CLIENT: Reconnect after 10000 ms
I (6344) WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
I (15414) WEBSOCKET: No data received for 10 seconds, signaling shutdown
E (17094) TRANSPORT_WS: Sec-WebSocket-Accept not found
E (17094) WEBSOCKET_CLIENT: Error transport connect
I (17094) WEBSOCKET_CLIENT: Reconnect after 10000 ms
I (17104) WEBSOCKET: WEBSOCKET_EVENT_DISCONNECTED
```

gabsuren, 2022-03-25:
@dreamcmi thank you for reporting. 
The problem cause is that "ws://echo.websocket.org" service is no longer available(See https://www.lob.com/blog/websocket-org-is-down-here-is-an-alternative), example should be modified to use
`ws://echo.websocket.org -> ws://echo.websocket.events`

It has been already fixed in latest master branch [esp-protocols](https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client/examples)

We will back port the fix for release/v4.4 and release/v4.3.
Thanks again!

dreamcmi, 2022-03-25:
> @dreamcmi thank you for reporting. The problem cause is that "ws://echo.websocket.org" service is no longer available(See https://www.lob.com/blog/websocket-org-is-down-here-is-an-alternative), [...]

@[gabsuren](https://github.com/gabsuren) thanks
But I still have a question, if a websocket server really needs the parameter `Sec-WebSocket-Accept`, is there any way to add it?

gabsuren, 2022-06-20:
> @[gabsuren](https://github.com/gabsuren) thanks But I still have a question, if a websocket server really needs the parameter `Sec-WebSocket-Accept`, is there any way to add it?

`Sec-WebSocket-Accept` key is sent by **WebSocket Server** during handshake response, the `Sec-WebSocket-Accept` header is used in the WebSocket opening handshake. It would appear in the response headers. That is, this is header is sent from Server to Client to inform that Server is willing to initiate a WebSocket connection, so there is no need to add this parameter from **WebSocket Client**.

closed 2022-07-13 by Alvin1Zhang