espressif/esp-idf #8632: at this moment, the max connections of esp32-c3 BLE controller have been fixed to 9? it is impossible to increase the number of connections? even if the host stack can support more than 9 or more, the final connections are still 9? correct?

iamcalledusman, 2022-03-23:
Hi 
Hope you are doing good!
I was reading that link it was very informative. I am facing an issue if you could help me. I am using Nimble spp_client and server example using esp-idf (esp32c3 ). Problem is that I am unable to connect more than 1 connection. As per your discussion esp32c3 can handle 9 concurrent connections when using nimble.

Here is the link to which I tried to explain the testing and results.

https://github.com/espressif/esp-idf/issues/8619

If you could help me in this regard.I will be very thankful for you.

rahult-github, 2022-03-24:
Hi @iamcalledusman ,

It is true that ESP32C3 has support for 9 connections in controller currently. However, for your use case, you will also need profile / application support to initiate 9 connections. In current SPP client / server demo example, support is added for one connection. The example was developed to showcase addition of a custom service in NimBLE stack and SPP service was picked for this demonstration.  

We can track this request in the #8619 that you have already raised and has this [comment.](https://github.com/espressif/esp-idf/issues/8619#issuecomment-1076273015) 

Thanks,
 Rahul

iamcalledusman, 2022-03-24:
Hi @rahult-github!

Thank You so much for the help and response! Can you please guide me for the few queries I have, if you have time? 
1)Can **spp_client** and **spp_server** applications be used for coded phy? if yes then any guidelines.

2)I used **multi_adv** for coded PHY , is there any option that we can take the ACK from the receiver that packet arrived? Or if **multi_adv** has no provision in that can we use **ble50_security_server** application? If it has can you please guide me about that . 

Thank you for your time and support! 

Regards,
Muhammad Usman

rahult-github, 2022-03-28:
Hi @iamcalledusman ,

> 1)Can **spp_client** and **spp_server** applications be used for coded phy? if yes then any guidelines.
> 

Yes , possible. The existing SPP demo application doesn't use extended related functions. But, Nimble has support for it. The [blecent](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/blecent/main/main.c#L274) application shows usage of extended API usage . You can refer to same and extend the existing application to suit your use case. 

> 2)I used **multi_adv** for coded PHY , is there any option that we can take the ACK from the receiver that packet arrived? 

So, i am sorry, but i didn't understand this query . Can you please elaborate ? you are referring to which ACK ?  ( an event from controller to host for a report ? ) 

>Or if **multi_adv** has no provision in that can we use **ble50_security_server** application? If it has can you please guide me about that .
> 

Ble50_security_server is application developed for bluedroid stack. It is not present for NimBLE stack. 

Thanks, 
Rahul

iamcalledusman, 2022-03-28:
Hi @rahult-github!

Thank you so much for the response!

I have few questions, i tried [blecent and bleprph ](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/blecent/main/main.c) and https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph

1)when I set the default Phy's

[image]

Then I am able to receive the legacy + extended advertisement packets. 

2)But when I set this setting, with coded PHY as shown in the image below 

[image]

Then still the same behavior as in the previous case. Unable to receive the coded PHY packets. For coded PHY packets reception, I am using nrf52833. 

3)In last , i set  params.legacy_pdu =0;

then following stack dump error message as mentioned in the image 

[image]

Can you please help me in this regard?

Thanks in advance.

rahult-github, 2022-03-29:
Hi @iamcalledusman ,

 The original issue was for number of connections attempt via existing BLE SPP Nimble demo app. The above query is now more for coded phy functionality. May i request you to  please open a separate ticket so as to track it properly, instead of clubbing all issues on one ticket ? 

Thanks,
 Rahul

iamcalledusman, 2022-03-29:
Hi @rahult-github,

Thank you for the response! Sure I will open a separate issue. You are right!

Thanks & regards

rahult-github, 2022-04-05:
Hi @iamcalledusman ,

 Closing this issue as have provided information [above](https://github.com/espressif/esp-idf/issues/8632#issuecomment-1077074039). Please feel to reopen if any more information needed. 

Thanks,
 Rahul

closed 2022-04-05 by rahult-github