espressif/esp-idf #8633: NimBLE stack is broken again, NimBLE SPP server is causing reboot

listout, 2022-03-23:
I'm on IDF version v5.0-dev-2203-g95b3dad771 with a version 1 devkit

Just ran the example NimBLE SPP server example and got this 

```
I (469) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40091b90
0x40091b90: _esp_error_check_failed at /home/gogo/esp/esp-idf/components/esp_system/esp_err.c:42

file: "/IDF/components/bt/controller/esp32/bt.c" line 1480
func: esp_bt_controller_mem_release
expression: try_heap_caps_add_region(mem_start, mem_end)

abort() was called at PC 0x40091b93 on core 0
0x40091b93: _esp_error_check_failed at /home/gogo/esp/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x40081e26:0x3ffbbe700x40091b9d:0x3ffbbe90 0x40096da6:0x3ffbbeb0 0x40091b93:0x3ffbbf20 0x400e0a99:0x3ffbbf40 0x400dfa76:0x3ffbbf60 0x400d68fc:0x3ffbbfb0 0x40123081:0x3ffbbfd0 0x40094839:0x3ffbbff0 
0x40081e26: panic_abort at /home/gogo/esp/esp-idf/components/esp_system/panic.c:403

0x40091b9d: esp_system_abort at /home/gogo/esp/esp-idf/components/esp_system/esp_system.c:135

0x40096da6: abort at /home/gogo/esp/esp-idf/components/newlib/abort.c:46

0x40091b93: _esp_error_check_failed at /home/gogo/esp/esp-idf/components/esp_system/esp_err.c:43

0x400e0a99: esp_bt_controller_mem_release at /home/gogo/esp/esp-idf/components/bt/controller/esp32/bt.c:1480 (discriminator 1)

0x400dfa76: esp_nimble_hci_and_controller_init at /home/gogo/esp/esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:465

0x400d68fc: app_main at /tmp/spp_server/main/main.c:397 (discriminator 2)

0x40123081: main_task at /home/gogo/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127 (discriminator 2)

0x40094839: vPortTaskWrapper at /home/gogo/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141
```

Is there a way I can use nimble spp on stable release branch?

listout, 2022-03-23:
Just moved to 4.4 and it's fixed. So I'm guessing it's a v5 specific issue.

rahult-github, 2022-03-24:
Hi @listout ,

 Please find fix for issue mentioned attached herewith. Can you please confirm it works for you.

[0001-heap-Fix-regression-in-heap_caps_add_region-API-rela.patch.txt](https://github.com/espressif/esp-idf/files/8339290/0001-heap-Fix-regression-in-heap_caps_add_region-API-rela.patch.txt)

Thanks
 Rahul

listout, 2022-03-24:
Yes @rahult-github patch fixes issue.

closed 2022-03-24 by listout