espressif/esp-idf #8634: DAC audio playback broken in ESP-IDF 4.4 (works in 4.3)

phoddie, 2022-03-23:
## Environment

- Development Kit: M5Stack
- Module or chip used:  ESP32
- IDF version: v4.4
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Using an IDE?: No

## Problem Description

Audio playback using `i2s_` API to DAC fails:

- The same code works under ESP-IDF 4.3
- No errors are reported by the `i2s_*` calls
- The `I2s_write`  calls return nearly immediately. For example, about 1000 ms of  16-bit 11 kHz mono data is accepted by `is2_write` in about 50 ms. This suggests that the audio data is not being correctly queued by the driver. When it works correctly, the `i2s_write` calls are roughly evenly spaced across the one second interval.

The [implementation](https://github.com/Moddable-OpenSource/moddable/blob/d7895e48268b2a347cac01348c920695d05ab9bf/modules/pins/i2s/audioout.c#L1224-L1353) used is in the Moddable SDK audio output module. 
  
## Code to reproduce this issue

Summary of implementation:

```c
	i2s_config_t i2s_config = {
		.mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN,
		.sample_rate = 11025,
		.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
		.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
		.communication_format = I2S_COMM_FORMAT_STAND_I2S,
		.intr_alloc_flags = 0,
		.dma_buf_count = 2,
		.dma_buf_len = 1024,
		.use_apll = false
	};
	i2s_driver_install(0, &i2s_config, 0, NULL);
	i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);

	while (moreAudioToPlay)
		i2s_write(0, buffer, 4096, &bytes_written, portMAX_DELAY);
```

## Other information

We first became aware of this issue when reported against the Moddable SDK ([#881](https://github.com/Moddable-OpenSource/moddable/issues/881)). There is a report (#6470) of ADC output being broken in ESP-IDF v4.1, but this is not the issue we have (our code works with ESP-IDF v4.3). That report has a [comment](https://github.com/espressif/esp-idf/issues/6470#issuecomment-1046837827) which mentions ADC audio is broken on ESP-IDF v4.4 also.

L-KAYA, 2022-04-12:
As mentioned in [No AudioOut sounds with ESP-IDF v4.4](https://github.com/Moddable-OpenSource/moddable/issues/881)
1. Checkout to the latest `release/v4.4` branch;
2. Set `communication_format` to `I2S_COMM_FORMAT_STAND_MSB`;
3. For stereo case, set `channel_format` to `I2S_CHANNEL_FMT_RIGHT_LEFT` and for mono case, set `channel_format` to `I2S_CHANNEL_FMT_ONLY_RIGHT` or `I2S_CHANNEL_FMT_ONLY_LEFT`

phoddie, 2022-04-12:
It looks like the ESP-IDF v4.4 changes for analog audio did break the Moddable SDK. Following the advice above, we have it working again. Thank you!

Alvin1Zhang, 2022-04-13:
Thanks for reporting and sharing the updates, good to hear it working again, feel free to reopen.

closed 2022-04-13 by Alvin1Zhang