espressif/esp-idf #8635: UART send data from buffer twice when other side reply too soon.

MiyukiDark, 2022-03-23:
## Environment

- Development Kit:      [ESP32-DEVKITC-32UE]
- Module or chip used:  [ESP32-WROOM-32UE]
- PatformIO 3.5.0 
- framework-arduinoespressif32 3.10006.210326 (1.0.6)
- tool-esptoolpy 1.30100.210531 (3.1.0)
- toolchain-xtensa32 2.50200.97 (5.2.0)
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

Data from UART2 are sent twice when another side replies in a few hundred us.
I use Arduino based  Serial2.print("#APP:cw:data?\n"); 
Port speed is set to 115200
So message is send in 1.2ms 
Other side start to reply in 200us 
But after about 570us message is sent again with no clear reason

Everything is checked on a logic analyzer. 
Without a reply from another side, it works just fine and the message is sent only once. 

On UART2 is this message "#APP:cw:data?\n" twice with 570us spacing 

### Code to reproduce this issue

```cpp

#include <WiFi.h>
#include <WiFiUdp.h>

#include <Preferences.h>

static Preferences configuration_storage;
static String ssid_used;
static String pswd_used;

static WiFiUDP udp_to_SW;
static WiFiUDP udp_from_Sensor;
static WiFiUDP udp_from_SensorPing;

void MainSenderTask(void * parameter)
{
  TickType_t  xLastWakeTime = xTaskGetTickCount();
  Serial.println("Enter Main sender Task");
  for( ;; )
  {
    //run this task every x ms 
    vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));

    //Send a packet
    udp_to_SW.beginPacket(udpBCast, udpForSW);
    if (Serial2.available() > 0)
    {
      String str = Serial2.readString(); 
      int str_len = str.length() + 1;
      char char_array[str_len];
      str.toCharArray(char_array, str_len);
      udp_to_SW.printf(char_array, str.length()); 
    }
    else
    {
      udp_to_SW.printf("empty");
    }
    udp_to_SW.endPacket();

    //ping sensors
    udp_from_SensorPing.beginPacket(udpBCast, udpForSensorPing);
    udp_from_SensorPing.printf("ping");
    udp_from_SensorPing.endPacket();

    //request data from main app
    Serial2.print("#APP:cw:data?\n");

    // If packet received...
    char packet[255];
    int packetSize = udp_to_SW.parsePacket();
    if (packetSize) 
    {
      int len = udp_to_SW.read(packet, 255);
      if (len > 0)
      {
        packet[len] = '\0';
      }
      //send received packet to main app
      Serial2.println(packet);
      //copy received packet to debug output
      Serial.print("Packet from APP received: ");
      Serial.println(packet);
    }
  }
}

void mode_WithPC_setup()
{
  configuration_storage.begin(NVM_Name, false);
  ssid_used = configuration_storage.getString("ssid", default_ssid);
  pswd_used = configuration_storage.getString("pswd", default_pswd);
  configuration_storage.end();

  Serial.printf("Starting WiFi: %s@%s \n", ssid_used.c_str(), pswd_used.c_str());

  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid_used.c_str());
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

  Serial.printf("TX power: %i \n", WiFi.getTxPower());

  udp_to_SW.begin(udpForSW);
  udp_from_Sensor.begin(udpForSensors);
  udp_from_SensorPing.begin(udpForSensorPing);

  xTaskCreate(
    MainSenderTask,    // Function that should be called
    "PC communication Task",   // Name of the task (for debugging)
    2000,            // Stack size (bytes)
    NULL,            // Parameter to pass
    1,               // Task priority
    NULL             // Task handle
  );
}

void setup() {
  //debug meessage uart
  Serial.begin(115200);
  //communicate with main APP
  Serial2.begin(115200);

  mode_WithPC_setup();
}

void loop() {

}

```

MiyukiDark, 2022-03-23:
I was trying pieces of code again and
Dual sending is caused by this piece of code 
So it is not a reply from the other side but reading from receive buffer what cause dual send 
```cpp
if (Serial2.available() > 0)
    {
      String str = Serial2.readString(); 
      int str_len = str.length() + 1;
      char char_array[str_len];
      str.toCharArray(char_array, str_len);
      udp_to_SW.printf(char_array, str.length()); 
    }
```
This causes message to be sent twice and timing was just a coincidence

atanisoft, 2022-03-23:
This should likely be over on https://github.com/espressif/arduino-esp32 since it looks possibly like a bug in the HardwareSerial wrapper code.

MiyukiDark, 2022-03-23:
> This should likely be over on https://github.com/espressif/arduino-esp32 since it looks possibly like a bug in the HardwareSerial wrapper code.

I expected it just pass data to layer down.

atanisoft, 2022-03-23:
That would depend on the version of arduino-esp32.

MiyukiDark, 2022-03-24:
It is way stranger than I expected. 
It might be some other issue because the code actually **executes twice** the whole task 
This is way too strange

```cpp
void MainSenderTask(void * parameter)
{
  TickType_t  xLastWakeTime = xTaskGetTickCount();
  Serial.println("Enter Main sender Task");
  for( ;; )
  {
    //run this task every x ms 
    vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));

    uint8_t run_countert;

    //Send a packet
    udp_to_SW.beginPacket(udpBCast, udpForSW);
    if (Serial2.available() > 0)
    {
      String str = Serial2.readString(); 
      int str_len = str.length() + 1;
      char char_array[str_len];
      str.toCharArray(char_array, str_len);
      udp_to_SW.printf(char_array, str.length()); 
      Serial.println(str.c_str());
    }
    else
    {
      udp_to_SW.printf("empty");
      Serial.printf("empty\n");
    }
    udp_to_SW.endPacket();

    //ping sensors
    udp_from_SensorPing.beginPacket(udpBCast, udpForSensorPing);
    udp_from_SensorPing.printf("ping");
    udp_from_SensorPing.endPacket();

    run_countert++;
    Serial.println(run_countert);

    //request data from main app
    Serial2.print("#APP:cw:data?\n");

    // If packet received...
    char packet[255];
    int packetSize = udp_to_SW.parsePacket();
    if (packetSize) 
    {
      int len = udp_to_SW.read(packet, 255);
      if (len > 0)
      {
        packet[len] = '\0';
      }
      //send received packet to main app
      Serial2.println(packet);
      //copy received packet to debug output
      Serial.print("Packet from APP received: ");
      Serial.println(packet);
    }
  }
}
```

And this is the result every second

```
197
2:0:1:0:1:0:1:0:1:0:0:0:0:1
198
empty
```

VojtechBartoska, 2022-03-24:
Hi @MiyukiDark, thanks for your contribution. Let's close this issue and continue investigating this in Arduino ESP32 core repository as it is related to Arduino.

closed 2022-03-24 by VojtechBartoska