espressif/esp-idf #8637: Ble notification not working, but works with Arduino ESP

Dariu5, 2022-03-23:
## Environment

- Development Kit:      ESP32-POE-ISO
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.4 
- Build System:         idf.py
- Compiler version xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     Windows
- (Windows only) Plain Command Prompt
- Using an IDE? : Visual code studio
- Power Supply:         USB

## Problem Description

I have a problem with ESP-IDF ble notification problem. I have subscribed notification, then send a command to get band activity data, and then get nothing. ESP-IDF ble client example works fine with other devices, I have a problem with this particular device. I'm using ESP-IDF v4.4.

Also, I have tried Arduino ESP ble client example and to my surprise, it's working fine. I have found out, that Arduino uses the same Bluedroid stack for BLE connectivity, but uses precompiled BT library.

### Expected Behavior

Ble device sends notifications

### Actual Behavior

Ble doesn't send notifications

### Steps to reproduce

Problem is reproduced by using GATT client example

## Debug Logs

ESP-IDF log:

https://pastebin.com/czf4Kq7s

## Logs captured with BLE sniffer
To compare what is difference between Arduino and IDF communication with the device I have captured ble logs with Adafruit sniffer.

Logs show that programs subscribe for notification for the same characteristic, and write commands to get data to the same characteristic. The only noticeable difference for me is that in Arduino master responds to slave MTU requests just after the start of the connection and ESP-IDF doesn't.

[esp_idf.zip](https://github.com/espressif/esp-idf/files/8335865/esp_idf.zip)

Logs show that programs subscribe for notification for the same characteristic, and write commands to get data to the same characteristic. The only noticeable difference for me is that in Arduino master responds to slave MTU requests just after the start of the connection and ESP-IDF doesn't.
[image]

ESP-IDF later(after getting characteristics info) sends MTU request and gets a response from the slave.
[image]

So I have two questions:

1) Can missing MTU response prevent the device from sending notifications? Or I'm looking at the wrong place?
2) If 1) is true, how could I enforce the same behavior(response to client MTU request) in ESP-IDF as in Arduino ESP.

Thanks for the help.

Dariu5, 2022-03-27:
I find the cause of the problem in gatt_main.c file.

`                /* message from client */
                if ((op_code % 2) == 0) {
#if (GATTS_INCLUDED == TRUE)
                    gatt_server_handle_client_req (p_tcb, op_code, msg_len, p);
#endif  ///GATTS_INCLUDED == TRUE
                } else {
#if (GATTC_INCLUDED == TRUE)
                    gatt_client_handle_server_rsp (p_tcb, op_code, msg_len, p);
#endif  ///GATTC_INCLUDED == TRUE
                }
`
Activity band send GATT MTU request with opcode 0x02, but it is ignored because function "gatt_server_handle_client_req" which should form response is undefined.  My project has only the GATTC module enabled.

So I have enabled GATTS module and now ESP responds to MTU request and BLE notifications are working.

zhp0406, 2023-09-11:
@Dariu5 
Have you resolved the issue?

Dariu5, 2023-09-11:
> @Dariu5 Have you resolved the issue?

Yes, i have enabled GATTS module in esp configuration menu,

closed 2023-09-12 by Alvin1Zhang