espressif/esp-idf #8639: some noise at every start output of the i2s_write

Kingwulin, 2022-03-24:
Hi,I am using the demo [here](https://github.com/espressif/esp-idf/blob/56659cdf36790e182451cd892ea349b29a27d333/examples/peripherals/i2s/i2s_adc_dac/main/app_main.c#L247)
I modify this demo to make the .wav file output to a speaker over the i2s driver
mostly,every time at the starting of the .wav file ,the sound always some noise like "de" 
How to deal with it ,Thanks

L-KAYA, 2022-03-24:
This example uses `i2s_write` to play raw music that store in flash or header file. Raw music means only audio data in it, but a `.wav` format music has a protocol header at the beginning, which indicates the `sample_rate`, `data_bit_width` and so on, these information are used to decode the following music and shouldn't be played. You can try to parse the header of `.wav` file and play the music without the header.

Hope it can help~

L-KAYA, 2022-03-24:
FYI, you can refer to how a `.wav` header is generated in [i2s_audio_recorder_sdcard](https://github.com/espressif/esp-idf/blob/56659cdf36790e182451cd892ea349b29a27d333/examples/peripherals/i2s/i2s_audio_recorder_sdcard/main/i2s_recorder_main.c)

Kingwulin, 2022-03-24:
> This example uses `i2s_write` to play raw music that store in flash or header file. Raw music means only audio data in it, but a `.wav` format music has a protocol header at the beginning, which [...]

think you for your reply
below it my code to play a .wav file stora at  spiffs,  I had read the header of the file ,but the noise still exit

```
static esp_err_t  play_wav(const char *filepath)
{
    FILE *fd = NULL;
    fd = fopen(filepath, "r");

    if (NULL == fd) {
        ESP_LOGE(TAG, "Failed to read existing file : %s", filepath);
        return ESP_FAIL;
    }
    const size_t chunk_size = 2048;
    uint8_t *buffer = malloc(chunk_size);

    if (NULL == buffer) {
        ESP_LOGE(TAG, "audio data buffer malloc failed");
        fclose(fd);
        return ESP_FAIL;
    }
    int len; 
    /**
     * read head of WAV file
     */
    wav_header_t wav_head;
    len = fread(&wav_head, 1, sizeof(wav_header_t), fd);
    if (len <= 0) {
        ESP_LOGE(TAG, "Read wav header failed");
        fclose(fd);
        return ESP_FAIL;
    }
    if (NULL == strstr((char *)wav_head.Subchunk1ID, "fmt") &&
            NULL == strstr((char *)wav_head.Subchunk2ID, "data")) {
        ESP_LOGE(TAG, "Header of wav format error");
        fclose(fd);
        return ESP_FAIL;
    }

    // ESP_LOGI(TAG, "frame_rate=%d, ch=%d, width=%d", wav_head.SampleRate, wav_head.NumChannels, wav_head.BitsPerSample);

    /**
     * read wave data of WAV file
     */
    size_t write_num = 0;
    size_t cnt;
    //i2s_start(I2S_NUM_0);
	do {
        /* Read file in chunks into the scratch buffer */
        len = fread(buffer, 1, chunk_size, fd);
        if (len <= 0) {
            ESP_LOGI(TAG, "break out");
            //i2s_stop(I2S_NUM_0);
            break;
        }
        
        //pwm_audio_write(buffer, len, &cnt, 1000 / portTICK_PERIOD_MS);
		i2s_write(I2S_NUM_0, buffer, len, &cnt, 100);
        ESP_LOGI(TAG, "%d,%d",cnt,len);
        write_num += len;
    } while (1);

    free(buffer);
	fclose(fd);
	return ESP_OK;
}
```

``
afasdfasfasf
``

Kingwulin, 2022-03-24:
Hello，there is [video](https://user-images.githubusercontent.com/16952072/159858450-fcb4e220-a1a6-4855-a3e5-ba9dbaaf0782.mp4)，There is a "de" sound at the beginning of most tones，and below is my main code

```
void app_play_num(int index)
{
    char file_buf[16];

    sprintf(file_buf,"/audio/%d.wav",index);
    
    play_wav(file_buf);
    ESP_LOGI(TAG,"going to player %s",file_buf);
}

void speaker_setTemp2_say(void)
{
    char file_buf[32];

    sprintf(file_buf,"/audio/setTime.wav");
    ESP_LOGI(TAG,"going to player %s",file_buf);
    play_wav(file_buf);   
}
void speaker_du_say(void)
{
    char file_buf[16];

    sprintf(file_buf,"/audio/min.wav");
    ESP_LOGI(TAG,"going to player %s",file_buf);
    play_wav(file_buf);   
}
void app_voice_play(int num)
{
    speaker_setTemp2_say();
 
    if(num/10)
    {
        app_play_num(num/10); //noise exit at the begin of this code 
        app_play_num(0); //noise exit at the begin of this code 
    }

    app_play_num(num%10); //noise exit at the begin of this code 
 
    speaker_du_say(); //noise exit at the begin of this code 
   
}
void app_main(void)
{
    spiffs_driver_init();
    
    hal_i2s_driver_init();
    //play_wav(MOUNT_POINT"/sample.wav");
    app_voice_play(19); 
    vTaskDelay(100);
    app_voice_play(13);
    vTaskDelay(100);
    app_voice_play(31);
    vTaskDelay(100);
    app_voice_play(39);
    vTaskDelay(100);
    app_voice_play(91);
    vTaskDelay(100);
    app_voice_play(93);
}
```

L-KAYA, 2022-03-24:
Firstly, for the current I2S driver, due to the driver will start automatically after installation, the uninitialized data may be transported, you can try to stop the i2s after installation and calling `i2s_zero_dma_buffer` to clear the DMA buffer before start and write data.
Secondly, just in case the audio itself contains the noise, try to print the beginning data of the sending buffer.

L-KAYA, 2022-03-24:
Another notice, the DAC only support 8 bit-width, but most audios are at least 16 bit-width, the audio data need to be right shifted 8 bits

L-KAYA, 2022-03-24:
The clipped sound mainly caused by the rapid change of data. I guess the polling time between two `i2s_write` may be too long which lead to the audio not consecutive, try to remove other codes in the `i2s_write` loop. and also clear the dma buffer every time after finished sending the data.

Kingwulin, 2022-03-24:
> Firstly, for the current I2S driver, due to the driver will start automatically after installation, the uninitialized data may be transported, you can try to stop the i2s after installation and [...]

here is the wav file
[spiffs_image.zip](https://github.com/espressif/esp-idf/files/8339203/spiffs_image.zip)

Kingwulin, 2022-03-24:
> 
how to clear the dma buffer

L-KAYA, 2022-03-24:
Call `i2s_zero_dma_buffer`

Kingwulin, 2022-03-24:
> Call `i2s_zero_dma_buffer`

Thanks for your reply,would you tell me which file the "i2s_zero_dma_buffer" function is. I can't find the call in the git project

[image]

L-KAYA, 2022-03-24:
It's not a file, it's a function declared in `components/driver/include/driver/i2s.h` line 379.

Kingwulin, 2022-03-24:
> It's not a file, it's a function declared in `components/driver/include/driver/i2s.h` line 379.

Thank you!
I found event I write the code like this:
to make the num output data consecutly,but the 'de' noise is still exit in num

Is it a problem of wav format, is it better for me to change to pcm format?

```
#include <stdio.h>
#include "driver/i2s.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "sdkconfig.h"
#include "esp_spiffs.h"
#include <string.h>
static const char *TAG = "speaker";

#define I2S_SAMPLE_RATE     (16000)
#define SAMPLE_BITS     (16)

#define MASTER_BCK_IO 4
#define MASTER_WS_IO 5
#define DATA_OUT_IO 18
#define DATA_IN_IO 21

#define MOUNT_POINT  "/audio"

static const char *partition_label = "audio";

typedef struct {
    // The "RIFF" chunk descriptor
    uint8_t ChunkID[4];
    int32_t ChunkSize;
    uint8_t Format[4];
    // The "fmt" sub-chunk
    uint8_t Subchunk1ID[4];
    int32_t Subchunk1Size;
    int16_t AudioFormat;
    int16_t NumChannels;
    int32_t SampleRate;
    int32_t ByteRate;
    int16_t BlockAlign;
    int16_t BitsPerSample;
    // The "data" sub-chunk
    uint8_t Subchunk2ID[4];
    int32_t Subchunk2Size;
} wav_header_t;

#define MAX_OP_FILE 10
typedef struct
{
	FILE * fds[MAX_OP_FILE];
	int num_use;
}file_op_t;
file_op_t file_op;
char file_name_buf[MAX_OP_FILE][16];

static esp_err_t  play_wav(const char *filepath)
{
    FILE *fd = NULL;
    //struct stat file_stat;

    // if (stat(filepath, &file_stat) == -1) {
    //     ESP_LOGE(TAG, "Failed to stat file : %s", filepath);
    //     return ESP_FAIL;
    // }

    // ESP_LOGI(TAG, "file stat info: %s (%ld bytes)...", filepath, file_stat.st_size);
[... 251 lines cut ...]
	wav_num_player();
 
    speaker_du_say(); //noise exit at the begin of this code 
   
}
void app_main(void)
{
    spiffs_driver_init();
    
    hal_i2s_driver_init();
    //play_wav(MOUNT_POINT"/sample.wav");
    app_voice_play(19); 
    vTaskDelay(100);
    app_voice_play(13);
    vTaskDelay(100);
    app_voice_play(31);
    vTaskDelay(100);
    app_voice_play(39);
    vTaskDelay(100);
    app_voice_play(91);
    vTaskDelay(100);
    app_voice_play(93);

}
```

L-KAYA, 2022-03-24:
You can put your `wav` file into a software named `audacity`, it can provide the wave shape and also able to play the audio.

closed 2024-07-08 by Alvin1Zhang