espressif/esp-idf #8640: IDF 4.4: GDB on command line is broken

readmodifywrite, 2022-03-24:
IDF 4.4

Following: https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-command-line

My OpenOCD seems to be working: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)

Using the WROVER Kit v4.1

When I run gdb, I get the following error:

gdbinit:1: Error in sourced command file:
Remote 'g' packet reply is too long: 4a85004000e0f63f0300000079000000fbffffff06000000000000001c0000000000000023990880d061fc3ff00000000038000005000000c60000000009fb3f2302060000800080b061fc3ff000000009adce0505000000c6000000010000000000000047850080a061fc3ff30000000400000000050000c600000003000000230f06006605c60110e00060000000020000000067480300f0490200e000000024000000dae511806062fc3fe00000008c0a0000860000000000000028f3fb3f03000000d45408804062fc3f40e309000000000034aece05005600000000000023000600478500803062fc3f130a02000100000000010000ffffbfffff01ffff64350100000000000000000000000000160000000c000000001400004a85004096fec51c2306060028abfa3f00000000ffff3fb300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

I've compiled with -Og and set the task wrapper in FreeRTOS.

The Eclipse debugger is also broken, it just goes haywire after hitting the first breakpoint and dies.

This all worked fine on 3.3.5....

igrr, 2022-03-24:
It seems like your OpenOCD version doesn't match the expected version for v4.4. It should be 20211120: https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-guides/tools/idf-tools.html#openocd-esp32

If you follow the installation instructions in the getting started guide, the correct version of OpenOCD will be installed when you run install.sh and will be added to your path when you source export.sh. The OPENOCD_SCRIPTS variable will also be set to point to the correct path in OpenOCD installation.

Please run `which -a openocd` after sourcing the export script. This could help figure out why the incorrect version of OpenOCD is launched.

readmodifywrite, 2022-03-24:
In the game of "do I have the correct OpenOCD version", I always lose.  Same story with the STM32 tbh.

Anyway, I tried v0.11.0-esp32-20211220.  Breakpoints don't seem to be working at all.  Actually, it's worse, on the old version it would try to break (but in the wrong place).  This one just throws up its hands and gives up.

Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, PC=0x4008379F, debug_reason=00000000
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x400803C0, debug_reason=00000001
Info : Flash mapping 0: 0x1 -> 0x4, 459920 KB
Info : esp32.cpu0: Target halted, PC=0x400803C0, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 320 KB
Info : Using flash bank 'esp32.cpu0.flash' size 320 KB
Info : esp32.cpu0: Target halted, PC=0x400803C0, debug_reason=00000001
Error: Failed to get flash maps (327732)!
Warn : Failed to get flash mappings (-4)!
Info : esp32.cpu0: Target halted, PC=0x400803C0, debug_reason=00000001
Info : Using flash bank 'esp32.cpu0.irom' size 0 KB
Info : esp32.cpu0: Target halted, PC=0x400803C0, debug_reason=00000001
Error: Failed to get flash maps (327732)!
Warn : Failed to get flash mappings (-4)!
Error: esp32.cpu0: DSR (FFFFFFFF) indicates target still busy!
Error: esp32.cpu0: DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: esp32.cpu0: DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : esp32.cpu0: Failed reading 984 bytes at address 0x3ffc0000
Error: no working area available, can't alloc space for stub data!
Error: esp32.cpu0: DSR (FFFFFFFF) indicates target still busy!
Error: esp32.cpu0: DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: esp32.cpu0: DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : esp32.cpu0: Failed writing 10968 bytes at address 0x40090000
Error: failed to restore 10968 bytes of working area at address 0x40090000
Error: Failed to get flash size!
Info : Using flash bank 'esp32.cpu0.drom' size 0 KB
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
Warn : xtensa_read_memory: esp32.cpu0: target not halted
Warn : xtensa_read_memory: esp32.cpu0: target not halted
Warn : xtensa_read_memory: esp32.cpu0: target not halted
Warn : xtensa_read_memory: esp32.cpu0: target not halted
Warn : xtensa_read_memory: esp32.cpu0: target not halted
Warn : xtensa_read_memory: esp32.cpu0: target not halted
Error: Failed to find HALTED core!
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : esp32.cpu0: Core was reset.
Error: timed out while waiting for target halted / 1 - 2
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Warn : address 0x400e794c not writable
Error: esp32.cpu0: Failed to write breakpoint instruction (-4)!
Error: esp32.cpu0: Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Warn : address 0x400e7ac5 not writable
Error: esp32.cpu0: Failed to write breakpoint instruction (-4)!
Error: esp32.cpu0: Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Info : The target is not running when halt was requested, stopping GDB.
Info : The target is not running when halt was requested, stopping GDB.
Info : The target is not running when halt was requested, stopping GDB.
Info : dropped 'gdb' connection

readmodifywrite, 2022-05-02:
@igrr I have not, unfortunately.  However, I did have automatic light sleep turned on, which probably was killing the GDB connection.  I'll close this one out, and re-open if I'm still having trouble.  Thanks!

closed 2022-05-02 by readmodifywrite