espressif/esp-idf #8641: Disabling gratuitous ARP in Lwip yields a complaint on the trace log

readmodifywrite, 2022-03-24:
IDF 4.4

I disabled gratuitous ARP, but get an warning on the trace log:
W (14064) esp_netif_lwip: CONFIG_LWIP_ESP_GRATUITOUS_ARP not enabled, but esp-netif configured with ESP_NETIF_FLAG_GARP

This was not a feature of 3.3.5.

It looks like there is a laborious and undocumented way to disable this behavior (I don't know if it actually breaks anything or not yet, so far it seems ok), but if the menuconfig has the option to disable GARP, was the intent for the IDF to be able to automatically configure itself to match?

david-cermak, 2022-03-24:
Hi @readmodifywrite 

The menuconfig option `CONFIG_LWIP_ESP_GRATUITOUS_ARP` is a compile time switch, so if disabled the code is removed and couldn't be enabled runtime. The second option `ESP_NETIF_FLAG_GARP` is a runtime configuration that could be enabled per interface basis (e.g. for users who want the gratuitous ARPs on WiFi, but not on Ethernet) -- this option is enabled by default.

Could you please share the way how you initialize the network and which network interface are you using? I would than propose the best way to disable it. Agree that it's not well documented: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html?highlight=esp_netif_flags_t#a-user-code-boiler-plate

readmodifywrite, 2022-03-24:
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());
esp_netif_create_default_wifi_sta();

My init code is still in flux from the move from 3.3.x to 4.4, so I'm open to changing it.  I can also live with the warning as long as it isn't breaking anything.

I also use the AP mode , but I haven't tested it on 4.4 yet.

david-cermak, 2022-03-24:
Just replace the `esp_netif_create_default_wifi_sta()` with these three lines:

```patch
-esp_netif_create_default_wifi_sta();
+esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_WIFI_STA();
+esp_netif_config.flags &= ~ESP_NETIF_FLAG_GARP; 
+esp_netif_create_wifi(WIFI_IF_STA, &esp_netif_config);
```

> as long as it isn't breaking anything.

the warning doesn't break anything, GARP is compile-time disabled (code is removed) and the periodic ARPs won't be sent.

closed 2022-12-22 by espressif-bot