espressif/esp-idf #8642: spi_bus_free crashes if interface was uninitialized

readmodifywrite, 2022-03-24:
IDF 4.4

This is another regression from 3.3.x.

Previously, you could call spi_bus_free if the bus hadn't been initialized, and it would just silently do nothing.

I use this ability frequently - my device needs to deinit and reinit the SPI bus often.  So I would just do a spi_bus_free and then spi_bus_initialize.

You have to do the free first, since spi_bus_initialize will crash if it has already been initialized.  However, now spi_bus_free crashes if it *wasn't* initialized (IE, the first time the routine is called).

I'm not seeing a way to check if the bus was initialized or not, so I will need to add my own state to track this.

Sorry to be filing so many bugs so quickly, but I'm about 20 hours in so far and I'm seriously reconsidering this upgrade.  I've barely touched 1/10 of the functionality I need to verify and it's just a mess all across the board.

ginkgm, 2022-03-24:
Hi @readmodifywrite ,

Thanks for reporting this issue. 
Though we don't encourage you rely on its returning, crash is obviously bad behavior... Will fix it.

Michael

closed 2022-03-30 by espressif-bot

MassimilianoMontagni, 2022-10-19:
+1
https://github.com/platformio/platform-espressif32/issues/944