espressif/esp-idf #8643: Firmware start address changed from 0x400D0018 to 0x400D0020?

readmodifywrite, 2022-03-24:
On IDF 4.4, the firmware start address is set to 0x400D0020.  On 3.3.5, it is set to 0x400D0018.  What was the reason for the move of 2 bytes?

The issue here is that existing bootloaders in the field can be broken by this.  My old bootloaders are looking for an image at 0x400D0018, not 0x400D0020.

Is it going to cause problems (either now, or eventually going forward into 5.x) to manually change the linker scripts to place the image back at 0x400D0018?

mahavirj, 2022-03-24:
@readmodifywrite 

As per app image format (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html) actual firmware data starts at offset of  `sizeof(esp_image_header_t) + sizeof(esp_segment_header_t)` which comes out as 0x20.

Hence the offset of 0x20 for flash DROM and IROM segments which allows to meet flash cache mapping criteria, also explained at:

https://github.com/espressif/esp-idf/blob/56659cdf36790e182451cd892ea349b29a27d333/components/esp_system/ld/esp32/memory.ld.in#L63-L69

> My old bootloaders are looking for an image at 0x400D0018, not 0x400D0020.

This may not be true. Bootloader parses firmware as per image format defined in above mentioned document and then prepares flash cache mappings based on load addresses.

You may find load addresses from following command:

```
esptool.py --chip esp32 image_info build/app.bin
```

1. In binary image layout, DROM is very first segment with load address as `load 0x3f400020`. This is where offset plays important role.
2. If you run this command of 3.x firmware then also you shall observe same load address in-spite the fact that offset was defined as 0x18 in ld script. Reason being, ESP32 linker used to round this offset to next 16-byte align boundary for new section (This was not true for ESP32-S2 case, and in the same context we found the issue and fixed LD script).

> Is it going to cause problems (either now, or eventually going forward into 5.x) to manually change the linker scripts to place the image back at 0x400D0018?

This is not required. And also you should not find any problem in staging OTA updates from v3.x to v4.x.

Please share logs if you are still running into any issues.

Thanks.

igrr, 2022-03-24:
One note for migration between v3.3.5 to v4.4, it applies since you were using CMake in v3.3.5. Quoting the release notes of v4.4:

Bootloader: The default app configuration is only compatible with bootloader and partition table generated from ESP-IDF V3.1 or newer. Apps that need to support OTA update on bootloaders and/or partition tables from older ESP-IDF must enable the new configuration option "App compatible with bootloader and partition table before ESP-IDF v3.1". Additionally, in ESP-IDF v3.3, CMake build system failed to append MD5 checksum to the partition table even when the option was enabled. This bug was fixed before v4.0 release. If the bootloader and partition table were initially built with CMake and IDF v3.3.x, then the compatibility option should also be enabled.

readmodifywrite, 2022-03-24:
So, for background context, most of my project is legacy from other microcontrollers.  The ESP32 port carries a lot of that.  I have my own bootloader that does it's own verification of the app image.  I can tell you for certain that the 3.3.x app images absolutely start at 0x400D0018 as is defined in the linker script and that is where my bootloader looks for them.

I'm not sure how the 3.3.x linker is somehow moving the app start from 0x400D0018 to 0x400D0020 when the linker script is putting iram0_2_seg at 0x400D0018.  FWIW, I'm not actually using CMake, I have my own build system.  I used the Espressif linker script and since it defined 0x400D0018 as the start address, that is what I used.  And it works just fine.  Until 4.x.

I don't use the MD5 on the partition table, so that's a non-issue for me.

I can live with/workaround most of the other bugs in 4.x, but messing around with the app image start address is a serious problem.  Generally you only get to define that one time and then once the product leaves the door, you can't change it again.

So my question stands: if I manually change the linker script back to the (for my system, correct) address, is that going to be a problem going forward?  Is the start address going to change again at some point?

readmodifywrite, 2022-03-24:
I can confirm that changing the iram0_2_seg address in the linker back to 0x400D0018 works with the bootloader.

KonstantinKondrashov, 2022-03-25:
@readmodifywrite 
just FYI, app image structure. [image_map.pdf](https://github.com/espressif/esp-idf/files/8347856/image_map.pdf).
Actually, there are no data movement (see the header: esp_image_header_t and esp_image_segment_header_t), all on the same place for the 3.3 and newer versions.
drom0_0_seg is placed to offset 0x20 instead of 0x18 (despite that in the scipt it is 0x18). The linker of esp32 places the app_desc (explicitly assigned to be the first segment in drom0_0_seg) to the 0x20 (alignment to 16-byte boundary),

esp32's ld rounds up to the next 0x10-byte aligned address when starting a new section - even though nothing in linker script seems to request this. SO the section started at 0x...20 even though the linker script said 0x...18. 

I think you can change it back to 0x18. Only one thing I am not sure - `Setting this offset makes it simple to meet the flash cache MMU's constraint that (paddr % 64KB == vaddr % 64KB)`. But I do not see that we faced with that issue, atleast I do not see it.

> On IDF 4.4, the firmware start address is set to 0x400D0020. On 3.3.5, it is set to 0x400D0018. What was the reason for the move of 2 bytes?

Actually, it is 8 bytes.

readmodifywrite, 2022-03-29:
> Actually, it is 8 bytes.

Yeah, I was pretty tired when I wrote that and forgot how to do hex math ;-)

> esp32's ld rounds up to the next 0x10-byte aligned address when starting a new section

That is an odd thing to have the linker do behind the scenes, but ok.

> I think you can change it back to 0x18

Yup, this works fine.

closed 2022-03-30 by Alvin1Zhang