espressif/esp-idf #8645: esp32s3 usb host example cdc_acm_bg96 receive some message repeatly

zhuyeaini9, 2022-03-24:
Hi all:
I used the esp32s3 board to develop one usb host fuction.
the device was the cdc-like device which connect to esp32s3's usb[GPIO 19 20] via one usb to uart chip(ft231xq).
I can find this device by the example of usb host lib,which show below info:
//-----------------------------------------------------------------------
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 8
idVendor 0x403
idProduct 0x6015
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
I (5417) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 32
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xa0
bMaxPower 90mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0xff
        iInterface 2
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
I (5477) CLASS: Getting Manufacturer string descriptor
FTDI
I (5487) CLASS: Getting Product string descriptor
FT231X USB UART
I (5487) CLASS: Getting Serial Number string descriptor
DQ008964
//-----------------------------------------------------------------------
but when I try the example of cdc_acm_bg96,
of course I change the code of vid,pid to my device.
comment some code of bg96 special,like send AT cmd etc.
so the code is simple:
1.open the device with rx handle function.
2.print data of rx handle function.
the device seems open ok from log:
//--------------
D (2430) USBH: Processing actions 0x100
D (2430) USBH: New device 1
D (2440) cdc_acm: New device connected
D (2440) cdc_acm: Checking list of connected USB devices
D (2450) CDC_ACM: Submitting poll for BULK IN transfer
D (2450) CDC_ACM: in xfer cb
//-------------
but I received repeated messages which has len 2,
first was 0x11,two was 0x60.
it received again and again.
the device is ok,I connect it to pc's usb port.
it works,it not send the 0x11 0x60 message.
it just return message after I send some cmd to it.
I check the ascii,the 0x11 ofter used for XON flag,I don't know what should I do.
from the cdc_acm_bg96's readme,the device should be support:
//--------------------------------------
## Supported Devices

The CDC-ACM Host driver supports the following types of CDC devices:

1. CDC-ACM devices
2. CDC-like vendor specific devices (usually found on USB to UART bridge devices)

### CDC-Like Vendor Specific Devices

The CDC-ACM Class driver supports CDC-like devices that meet the following requirements:
- The device class code must be set to the vendor specific class code `0xFF`
- The device needs to provide and interface containing the following endpoints:
    - (Mandatory) Two Bulk endpoints (IN and OUT) for data
    - (Optional) An interrupt endpoint (IN) for the notification element

For CDC-like devices, users are responsible for ensuring that they only call APIs (e.g., `cdc_acm_host_send_break()`) that are supported by the target device.
//-----------------------------------------------

any advice is welcome!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

tore-espressif, 2022-03-24:
Hello @zhuyeaini9 , thank you for reaching out to us.

I can confirm that I reproduced the problem with FTDI 232RQ (PID: 0x6001), that is almost the same as your chip (comparison [here](http://madeintheusb.blogspot.com/2016/02/ft232-versus-ft231.html))

It is something FTDI specific, but I'll prepare an example how to handle this

zhuyeaini9, 2022-03-25:
thanks@tore-espressif,really helpful.
for the line coding setting.
I remember I also try the method:cdc_acm_host_line_coding_set
after open the device,but the log show error of it:
//---------------------------------
E (9640) cdc_acm: cdc_acm_host_set_control_line_state(1113):
//---------------------------------

tore-espressif, 2022-03-25:
That's correct, FTDI chips don't use commands as defined in USB CDC specification but their own vendor specific commands.

The FTDI implementation is currently under internal review. I'll update this issue when it gets public, thanks!

zhuyeaini9, 2022-03-25:
> That's correct, FTDI chips don't use commands as defined in USB CDC specification but their own vendor specific commands.
> 
> The FTDI implementation is currently under internal review. I'll update this issue when it gets public, thanks!

got it.

tore-espressif, 2022-04-11:
@zhuyeaini9 sorry for the delay.

If you are willing to test my work-in-progress code, you can try this example: https://github.com/tore-espressif/esp-idf/tree/feature/usb_host/cdc_custom_commands/examples/peripherals/usb/host/cdc/cdc_acm_vcp

The API can change in the official release, though.

There are couple of points we need to internally agree upon:
* Currently the examples are in C++ only, as we need to override certain functions (FTDI specific) which is easily done in C++
* Creating and destroying the driver object is not clean 

Your feedback would be greatly appreciated, thanks!

S0UL4, 2024-03-26:
Hello, 
Thank you for your work @tore-espressif 
Does this example works also with the FT2232D ?

tore-espressif, 2024-03-26:
> Hello,
> Thank you for your work @tore-espressif
> Does this example works also with the FT2232D ?

Hello, I have not tested with FT2232D.

You can easily try it out by updating the PID of the device in the driver and invoke this constructor https://github.com/espressif/esp-usb/blob/master/host/class/cdc/usb_host_ftdi_vcp/include/usb/vcp_ftdi.hpp#L38

S0UL4, 2024-03-26:
Thank you so much for your quick reponse, the codes are compatible also with the SIP ESP32-S3-PICO as it has the USB 1.1 OTG support right ?

tore-espressif, 2024-03-26:
yep, all modules based on S3 have USB 1.1 host support.

S0UL4, 2024-03-26:
You are the best guys, thank you !

tore-espressif, 2024-03-26:
Beaware of hardware of your development board. Not all boards have USB (pin 19 and 20) wired out to USB-A connector

S0UL4, 2024-03-31:
Excuse me, please. Can this library be independent of ESP32? I believe it can be used on other boards as well, since it's written entirely in C++ code. Am I right? Also, does this library support bidirectional communication?