espressif/esp-idf #8646: ESP32 AP mode failed to connect to the ESP32

krupis, 2022-03-24:
I have been using my device in AP_STA mode for a few weeks and did not have any issues. Today, I have turned ON the device and noticed that my phone can no longer connect to the AP. Nothing has changed in the code. I get the following logs on the serial monitor:

```
I (2801) wifi:wifi driver task: 3ffbc914, prio:23, stack:6656, core=0
I (2821) wifi:wifi firmware version: 7679c42
I (2821) wifi:wifi certification version: v7.0
I (2821) wifi:config NVS flash: enabled
I (2821) wifi:config nano formating: disabled
I (2821) wifi:Init data frame dynamic rx buffer num: 32
I (2831) wifi:Init management frame dynamic rx buffer num: 32
I (2831) wifi:Init management short buffer num: 32
I (2841) wifi:Init dynamic tx buffer num: 32
I (2841) wifi:Init static rx buffer size: 1600
I (2851) wifi:Init static rx buffer num: 10
I (2851) wifi:Init dynamic rx buffer num: 32
I (2851) wifi_init: rx ba win: 6
I (2861) wifi_init: tcpip mbox: 32
I (2861) wifi_init: udp mbox: 6
I (2861) wifi_init: tcp mbox: 6
I (2871) wifi_init: tcp tx win: 5744
I (2871) wifi_init: tcp rx win: 5744
I (2881) wifi_init: tcp mss: 1440
I (2881) wifi_init: WiFi IRAM OP enabled
I (2891) wifi_init: WiFi RX IRAM OP enabled
I (2901) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (2991) wifi:mode : sta (8c:4b:14:0f:2a:a0) + softAP (8c:4b:14:0f:2a:a1)
I (2991) wifi:enable tsf
I (3001) wifi:Total power save buffer number: 16
I (3001) wifi:Init max length of beacon: 752/752
I (3001) wifi:Init max length of beacon: 752/752
I (3011) WIFI AP: wifi_init_softap finished. SSID:default_ssid password:default_pass channel:1
I (10841) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (10841) wifi:station: 44:91:60:40:91:ee join, AID=1, bgn, 20
I (14851) wifi:station: 44:91:60:40:91:ee leave, AID = 1, bss_flags is 658530, bss:0x3ffb22d4
I (14851) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (24971) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (24971) wifi:station: 44:91:60:40:91:ee join, AID=1, bgn, 20
I (28971) wifi:station: 44:91:60:40:91:ee leave, AID = 1, bss_flags is 658530, bss:0x3ffb22d4
I (28971) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (72431) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (72431) wifi:station: 44:91:60:40:91:ee join, AID=1, bgn, 20
I (76441) wifi:station: 44:91:60:40:91:ee leave, AID = 1, bss_flags is 658530, bss:0x3ffb22d4
I (76441) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1

```

I have tried multiple phones and computers and none of them could connect to the AP.  

The function that I use to initialise the wifi in AP_STA mode:

```

void WIFI_INIT_AP_STA(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    //ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL)); // since we always do this after scanning, dont need to add event loop anymore

    esp_netif_t *ap_netif = esp_netif_create_default_wifi_ap();
    assert(ap_netif);

    esp_netif_t *sta_netif = esp_netif_create_default_wifi_sta();
    assert(sta_netif);

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    //printf("device info elstat id = %s \n",device_info.elstat_id);
    // default_ssid 12 withou NULL
    // custom id is 21 without NULL
    wifi_config_t wifi_config = {
        .ap = {
            //.ssid = EXAMPLE_ESP_WIFI_SSID_AP,
            //.ssid_len = strlen(EXAMPLE_ESP_WIFI_SSID_AP),
            .ssid_len = strlen(device_info.elstat_id),
            .channel = EXAMPLE_ESP_WIFI_CHANNEL,
            .password = EXAMPLE_ESP_WIFI_PASS_AP,
            .max_connection = EXAMPLE_MAX_STA_CONN,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };
    strcpy((char*)(wifi_config.ap.ssid),device_info.elstat_id);

    if (strlen(EXAMPLE_ESP_WIFI_PASS_AP) == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }
    

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI("WIFI AP", "wifi_init_softap finished. SSID:%s password:%s channel:%d",
             EXAMPLE_ESP_WIFI_SSID_AP, EXAMPLE_ESP_WIFI_PASS_AP, EXAMPLE_ESP_WIFI_CHANNEL);
}
```

I have decided to change the wifi password 
```
#define EXAMPLE_ESP_WIFI_PASS_AP     "test12345"
```

After chaning the password I was able to connect to the AP without any issues. 

 What is the possible cause of this error? Is it possible that wifi messed up if the device had lost power during the wifi initialization once? What are other possible reasons?

nishanth-radja, 2022-03-31:
@krupis has the DUT lost power during the weeks it was up.Next time this issue occurs can we have a sniffer capture to see where the issue is.

nishanth-radja, 2022-04-01:
@krupis by sniffer capture,I meant the wireless sniffer,We will be able to see the packets that are going in the air.
If you have a linux laptop,we can take the capture.
You will have to install wireshark on your laptop and run the following script.
killall wpa_supplicant
ifconfig <wifi-interface> down
iwconfig <wifi-interface> mode monitor
ifconfig <wifi-interface> up
iwconfig <wifi-interface> channel <chan>
wireshark

jack0c, 2022-05-20:
@nishanth-radja I think it is possible that the Wi-Fi NVS broken when power down suddenly. @nishanth-radja I think we can do some power down test, and dump the SSID, password, PMK every time.

nishanth-radja, 2022-05-21:
@jack0c sure, will try to repro this issue.

nishanth-radja, 2022-05-31:
@krupis can you also pls mention the IDF version used?

xegod, 2022-07-15:
this is def related to NVS, I had the same issue and then I erased NVS  using the   nvs_flash_erase() function on startup and it seems to work fine