espressif/esp-idf #8647: python-env folder not installed in tool chain

Zeni241, 2022-03-24:
I downloaded the zip file for IDF4.4 in a folder and run intall.ps1 form this folder to install tool chain in another folder. "python-env" folder is not installed in tool chain folder.
Then I downloaded IDF4.4 using "git clone -b v4.4 --recursive https://github.com/espressif/esp-idf.git" from command prompt. 
This time when I ran "intall.ps1" form this folder to install tool chain,. "python-env" folder IS installed in tool chain folder and every thing worked fine.
Please check zip file of IDF4.4 as it may waste lot of time of some one as I have wasted.
Thanks

igrr, 2022-03-24:
@Zeni241 could you please give a link, which .zip file of IDF 4.4 you have downloaded?

Zeni241, 2022-03-24:
I think this is the link I used 
https://github.com/espressif/esp-idf/releases#:~:text=https%3A//dl.espressif.com/github_assets/espressif/esp%2Didf/releases/download/v4.4/esp%2Didf%2Dv4.4.zip

georgik, 2022-04-01:
@Zeni241  Thank you for the report. Solution is already in progress.

Meanwhile: It might help to use Windows Installer which contains multiple checks, provides integrated Python environment and it also works with shallow clones of ESP-IDF which are smaller: https://github.com/espressif/idf-installer#table-with-downloads