espressif/esp-idf #8648: How to enable esp_pppos component to connect internet with modbus tcp/ip via ethernet or wifi at the same time ?

NguyenMinhTri, 2022-03-24:
Currently, I'm using esp_pppos component to connect internet via 4G SIM, and now I want to enable modbus tcp/ip via wifi or ethernet but these do not work at the same time? 

My sample code
`    
    
    // init ethernet
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t * eth_netif = esp_netif_new( & cfg);
    ESP_ERROR_CHECK(esp_netif_dhcpc_stop(eth_netif));
    esp_netif_ip_info_t ip_info;
    IP4_ADDR( & ip_info.ip, 192, 168, 5, 11);
    IP4_ADDR( & ip_info.gw, 192, 168, 5, 1);
    IP4_ADDR( & ip_info.netmask, 255, 255, 255, 0);
    esp_netif_set_ip_info(eth_netif, & ip_info);
    assert(eth_netif);
    ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));

    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, & got_ip_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, & eth_event_handler, NULL));

    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = 1;
    phy_config.reset_gpio_num = 2;
    mac_config.smi_mdc_gpio_num = 23;
    mac_config.smi_mdio_gpio_num = 18;
    esp_eth_mac_t * mac = esp_eth_mac_new_esp32( & mac_config);
    esp_eth_phy_t * phy = esp_eth_phy_new_lan8720( & phy_config);
    esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
    esp_eth_handle_t eth_handle = NULL;
    ESP_ERROR_CHECK(esp_eth_driver_install( & config, & eth_handle));
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));

     ..............
    //init 4G SIM
    vTaskDelay(10000 / portTICK_PERIOD_MS);
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, & on_ip_event, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, & on_ppp_changed, NULL));
    sim_event_group = xEventGroupCreate();
    // Init netif object
    esp_netif_config_t cfgSim = ESP_NETIF_DEFAULT_PPP();
    esp_netifSim = esp_netif_new( & cfgSim);
    assert(esp_netifSim);

    /* create dte object */
    esp_modem_dte_config_t config2 = ESP_MODEM_DTE_DEFAULT_CONFIG();
    config2.tx_io_num = SIM_POWER_TX_PIN;
    config2.rx_io_num = SIM_POWER_RX_PIN;
    ESP_LOGI(TAG, "TX: %d", SIM_POWER_TX_PIN);
    config2.event_task_stack_size = 3024;
    dte = esp_modem_dte_init( & config2);

    /* Register event handler */
    ESP_ERROR_CHECK(esp_modem_set_event_handler(dte, modem_event_handler, ESP_EVENT_ANY_ID, NULL));
    modem_dce_t * dce = sim7600_init(dte);
    ESP_ERROR_CHECK(dce -> set_flow_ctrl(dce, MODEM_FLOW_CONTROL_NONE));
    ESP_ERROR_CHECK(dce -> store_profile(dce));
    /* Print Module ID, Operator, IMEI, IMSI */
    ESP_LOGI(TAG, "Module: %s", dce -> name);
    ESP_LOGI(TAG, "Operator: %s", dce -> oper);
    ESP_LOGI(TAG, "IMEI: %s", dce -> imei);
    ESP_LOGI(TAG, "IMSI: %s", dce -> imsi);
    /* Get signal quality */
    uint32_t rssi = 0, ber = 0;
    ESP_ERROR_CHECK(dce -> get_signal_quality(dce, & rssi, & ber));
    ESP_LOGI(TAG, "rssi: %d, ber: %d", rssi, ber);
    /* Get battery voltage */
    uint32_t voltage = 0, bcs = 0, bcl = 0;
    ESP_ERROR_CHECK(dce -> get_battery_status(dce, & bcs, & bcl, & voltage));
    ESP_LOGI(TAG, "Battery voltage: %d mV", voltage);
    /* setup PPPoS network parameters */
    esp_netif_ppp_set_auth(esp_netifSim, auth_type, CONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME, CONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD);

    void * modem_netif_adapter = esp_modem_netif_setup(dte);
    esp_modem_netif_set_default_handlers(modem_netif_adapter, esp_netifSim);
    /* attach the modem to the network interface */
    esp_netif_attach(esp_netifSim, modem_netif_adapter);
    
    
    
    .......................
    
    
    // modbus tcp/ip master

    void * master_handler = NULL; // Pointer to allocate interface structure

    // Initialization of Modbus slave for TCP
    esp_err_t error_holder = mbc_master_init_tcp( & master_handler);

    if (master_handler == NULL || error_holder != ESP_OK) {
      ESP_LOGE(TAG, "mb controller initialization fail.");
    }

    char * slave_ip_address_table[2] = {
      slave_ip, // Address corresponds to UID1 and set to predefined value by user    // corresponds to UID2 in the segment
      NULL // end of table
    };

    mb_communication_info_t comm_info = {
      .ip_port = tcp_port, // Modbus TCP port number (default = 502)
      .ip_addr_type = MB_IPV4, // version of IP protocol
      .ip_mode = MB_MODE_TCP, // Port communication mode
      .ip_addr = (void * ) slave_ip_address_table, // assign table of IP addresses
      .ip_netif_ptr = (void * ) eth_netif // esp_netif_ptr pointer to the corresponding network interface
    };

    ESP_ERROR_CHECK(mbc_master_setup((void * ) & comm_info));
    esp_err_t err = mbc_master_start();`

alisitsyn, 2022-03-30:
@NguyenMinhTri,

Thank you for this issue. I would like to get some clarification (confirmation) of this issue to make sure I correcttly understand your environment.

So, your Modbus TCP slave device is `SIMATIC S7-1200` connected to the Ethernet (as per code) network with the static IP, let us say `192.168.5.2`. You are configuring the Modbus TCP master connected to the same network with its static IP to read the slave. This works just fine but once you are initializing the internet connection over `esp_pppos` the connection is lost? Is this correct? If not please clarify. 

Thanks.

NguyenMinhTri, 2022-03-31:
@alisitsyn Thanks for your support. I have added detail information as below. Please help me to check it. 
## Environment
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version: v4.3.2
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:  Windows
- Environment type: Plain Command Prompt

## Debug Logs

```
I (27) boot: ESP-IDF v4.3.2 2nd stage bootloader
I (27) boot: compile time 19:09:51
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1ed20h (126240) map
I (155) esp_image: segment 1: paddr=0002ed48 vaddr=3ffb0000 size=012d0h (  4816) load
I (157) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=709bch (461244) map
I (328) esp_image: segment 3: paddr=000a09e4 vaddr=3ffb12d0 size=01834h (  6196) load
I (330) esp_image: segment 4: paddr=000a2220 vaddr=40080000 size=0e1a0h ( 57760) load
I (357) esp_image: segment 5: paddr=000b03c8 vaddr=50000000 size=00010h (    16) load
I (364) boot: Loaded app from partition at offset 0x10000
I (365) boot: Disabling RNG early entropy source...
I (377) cpu_start: Pro cpu up.
I (377) cpu_start: Starting app cpu, entry point is 0x40081344
0x40081344: call_start_cpu1 at G:/idf/esp-idf-v4.3.2/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (392) cpu_start: Pro cpu start user code
I (392) cpu_start: cpu freq: 160000000
I (392) cpu_start: Application information:
I (396) cpu_start: Project name:     lte_poc
I (401) cpu_start: App version:      1
I (406) cpu_start: Compile time:     Mar 31 2022 19:09:43
I (412) cpu_start: ELF file SHA256:  848a26ad308dfee9...
I (418) cpu_start: ESP-IDF:          v4.3.2
I (423) heap_init: Initializing. RAM available for dynamic allocation:
I (430) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (436) heap_init: At 3FFB48D0 len 0002B730 (173 KiB): DRAM
I (442) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (448) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (455) heap_init: At 4008E1A0 len 00011E60 (71 KiB): IRAM
I (462) spi_flash: detected chip: generic
I (466) spi_flash: flash io: dio
W (470) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (484) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (555) TEST: wait for 10 seconds as a work-around for crashes in LTE mode
I (20555) TEST: TX: 13
I (20555) uart: queue free spaces: 30
E (20555) esp-modem: esp_dte_handle_line(113): handle line failed
W (20555) TEST: Unknow line received: AT

E (20565) esp-modem: esp_dte_handle_line(113): handle line failed
W (20565) TEST: Unknow line received: ATE0

I (20665) TEST: Module: A7670C-LAAS
I (20665) TEST: Operator: "45204"
I (20665) TEST: IMEI: 861881056228027
I (20665) TEST: IMSI: 452048835449509
I (20685) TEST: rssi: 28, ber: 99
I (21035) TEST: Battery voltage: 3779 mV
I (21165) TEST: Modem PPP Started
I (21165) TEST: PPP state changed event 259
I (21165) TEST: PPP state changed event 262
I (21175) TEST: PPP state changed event 263
I (21185) TEST: PPP state changed event 265
I (21205) esp-netif_lwip-ppp: Connected
I (21205) esp-netif_lwip-ppp: Name Server1: 203.113.131.6
I (21205) esp-netif_lwip-ppp: Name Server2: 203.113.131.5
I (21215) TEST: Modem Connect to PPP Server
I (21215) TEST: ~~~~~~~~~~~~~~
I (21225) TEST: IP          : 9.233.203.143
I (21225) TEST: Netmask     : 255.255.255.255
I (21235) TEST: Gateway     : 10.64.64.64
I (21235) TEST: Name Server1: 203.113.131.6
I (21245) TEST: Name Server2: 203.113.131.5
I (21245) TEST: ~~~~~~~~~~~~~~
I (21245) TEST: GOT ip event!!!
I (21255) Utils: Initializing SNTP
I (21255) TEST: PPP state changed event 266
I (23265) Utils: Current Time : Thu Mar 31 12:12:47 2022
I (23915) HTTP_CLIENT: HTTP GET Status = 200, content_length = 271
I (23915) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (23925) system_api: Base MAC address is not set
I (23925) system_api: read default base MAC address from EFUSE
I (23955) esp_eth.netif.glue: 34:ab:95:88:d2:4b
I (23955) esp_eth.netif.glue: ethernet attached to netif
I (25455) TEST: Ethernet Started
I (25455) TEST: Ethernet Link Up
I (25455) TEST: Ethernet HW Addr 34:ab:95:88:d2:4b
I (25455) esp_netif_handlers: eth ip: 192.168.5.11, mask: 255.255.255.0, gw: 192.168.5.1
I (25465) TEST: Ethernet Got IP Address
I (25465) TEST: ~~~~~~~~~~~
I (25475) TEST: ETHIP:192.168.5.11
I (25475) TEST: ETHMASK:255.255.255.0
I (25475) TEST: ETHGW:192.168.5.1
I (25485) TEST: ~~~~~~~~~~~
I (25485) MB_TCP_MASTER_PORT: TCP master stack initialized.
I (25495) MB_TCP_MASTER_PORT: Host[IP]: "192.168.5.55"[192.168.5.55]
I (25495) MB_TCP_MASTER_PORT: Add slave IP: 192.168.5.55
I (25515) MB_TCP_MASTER_PORT: Connecting to slaves...
-I (25515) MB_TCP_MASTER_PORT: Connected 1 slaves, start polling...
I (26525) TEST: Modbus master stack initialized...
E (31525) TRANS_TCP: [sock=55] select() timeout
E (31525) HTTP_CLIENT: Connection failed, sock < 0
E (31525) HTTP_CLIENT: HTTP GET request failed: ESP_ERR_HTTP_CONNECT
I (31525) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
```

## Other items if possible

- Please check the attach file that is my code.
[4G-LTE-MODBUS-TCP-IP.zip](https://github.com/espressif/esp-idf/files/8389248/4G-LTE-MODBUS-TCP-IP.zip)

alisitsyn, 2022-04-04:
@NguyenMinhTri, 

Thank you for your information. This will take some time to investigate and fix. The issue started to go through the formal process.

david-cermak, 2022-04-06:
@NguyenMinhTri My guess is that the DNS gets updated after getting the DHCP lease on the Ethernet interface and rewrites the DNS information acquired from the PPP server.
Could you please try to manually set the server(s) before running the http client connection?

You can use this API: 
https://github.com/espressif/esp-idf/blob/d75674579e676eb4533a6529da27a733976606fb/components/esp_netif/include/esp_netif.h#L586

(please try to set the DNS servers manually or get the information using `esp_netif_get_dns_info()` after connecting to the cellular network)

Alvin1Zhang, 2022-05-06:
@NguyenMinhTri Thanks for reporting, would you please help share if any further updates? Thanks.

sengulhamza, 2022-05-31:
> @NguyenMinhTri My guess is that the DNS gets updated after getting the DHCP lease on the Ethernet interface and rewrites the DNS information acquired from the PPP server. Could you please try to [...]

Hello David,
I think we can not set dns using cellular network over ppp.
DNS Set API not supported if netif type is P2P. https://github.com/espressif/esp-idf/blob/d75674579e676eb4533a6529da27a733976606fb/components/esp_netif/lwip/esp_netif_lwip.c#L1503-L1513
Am i right ?
Thanks.

NguyenMinhTri, 2022-12-21:
> @NguyenMinhTri My guess is that the DNS gets updated after getting the DHCP lease on the Ethernet interface and rewrites the DNS information acquired from the PPP server. Could you please try to [...]

Thank you, I have used this API but my issue still not fixed.
[image]

david-cermak, 2023-01-02:
@NguyenMinhTri The code is not very readable from the screenshot, but it looks like both interfaces are created with their default configs, correct? If so, the default route-prio for Ethernet is higher, so that the HTTP request is routed via this interface. (no DNS issue at this point, as the DNS was resolved in the previous step (first call of `run_http_test()`).
To fix this you can simply adjust the `route_prio` numbers when creating network interfaces.
(modbus should still work if you connect to nodes using IP addresses or via mDNS names)

> DNS Set API not supported if netif type is P2P.

@sengulhamza This is correct, thanks for mentioning. will fix!

david-cermak, 2023-01-02:
To update the routing priority for Ethernet interface, you can replace your
```cpp
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&cfg);
```
with
```cpp
    esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_ETH();
    esp_netif_config.route_prio = 10;
    esp_netif_config_t config = {
        .base = &esp_netif_config,
        .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH
    };
    esp_netif_new(&config);
```

sengulhamza, 2023-01-03:
@david-cermak thany you for your interest, fix is very helpful to me.

alisitsyn, 2023-01-06:
> Hi @alisitsyn alisitsyn, Can you help me to that how to use SIM for internet connection and ethernet for modbus connection in esp32 ?

Hi @NguyenMinhTri, Could you try the suggestion from @cermak posted above with your hardware? I can not check it quickly with modem HW right now. The modbus uses LWIP transport with its socket API and needs the configured netif pointer. The above solution looks reasonable to me. If you still have issues with the modbus and client disconnection I will try to reproduce the issue on my side in order to find the solution.

closed 2023-01-16 by espressif-bot

m1a2x32, 2024-11-20:
> @NguyenMinhTri My guess is that the DNS gets updated after getting the DHCP lease on the Ethernet interface and rewrites the DNS information acquired from the PPP server. Could you please try to [...]

It might be two years late, but you, sir, are a lifesaver.