espressif/esp-idf #8650: RMT Driver 多通道输出支持

Pickaxe-Hit, 2022-03-24:
我有一个项目，用来产生一个串行数据，它对io翻转速度要求非常高，所以我打算用rmt外设来实现。但此串行数据需要数据线和时钟线两个连接，但rmt外设好像一次只能操作一个通道，rmt能否配置为多通道同时发送不同数据内容？
我查找了技术参考手册，rmt寄存器中有一项是多通道同时发送，但idf似乎没有配置此寄存器的封装好的函数。

suda-morris, 2022-03-25:
Hi @Pickaxe-Hit  we do have provided such API: rmt_add_channel_to_group
And the test case can be found here: https://github.com/espressif/esp-idf/blob/master/components/driver/test/test_rmt.c#L493
But sorry, this API name is not that straightforward to understand.

Pickaxe-Hit, 2022-03-25:
Hi @suda-morris 
I overviewed this API but I still have a question: How do I send different data to different channel(s)? This API seems that it can only send the same data to different channel(s). In order to drive the serial device I need to send CLK signal and data signal, it needs me to send different data.
Thanks for your reply.

suda-morris, 2022-03-25:
`rmt_write_items` is used to send the data. You can transfer different data by different channels. `rmt_add_channel_to_group` is just to set what channels should transmit simultaneously. The transmission will not stated until all the channels has called the `rmt_write_items`

Pickaxe-Hit, 2022-03-25:
> `rmt_write_items`用于发送数据。您可以通过不同的渠道传输不同的数据。`rmt_add_channel_to_group`只是设置应该同时传输哪些通道。直到所有通道都调用了`rmt_write_items`
> 
> 对于问题，您可能还可以在 esp32 论坛上提问，在那里您将获得更多人回答您的问题。 https://esp32.com/

Thanks for your help. I will close this issue.

closed 2022-03-25 by Pickaxe-Hit