espressif/esp-idf #8651: IDF monitor crashes when trying to process coredumps when coredumps are disabled

msn444, 2022-03-24:
This is probably a result of the solution to #8099. When running `idf_monitor.py --decode-coredumps disable` I get 
```
I (171) esp_image: segment 0: paddr=0x002e0020 vaddr=0x3f400020 size=0x21740 (137024) map
I (232) esp_image: segment 1: paddr=0x00301768 vaddr=0x3ffbdb60 size=0x03948 ( 14664) load
I (239) esp_image: segment 2: paddr=0x003050b8 vaddr=0x40080000 size=0x00404 (  1028) load
I (239) esp_image: segment 3: paddr=0x003054c4 vaddr=0x40080404 size=0x0ab54 ( 43860) load
I (267) esp_image: segment 4: paddr=0x00310020 vaddr=0x400d0020 size=0xad7a4 (710564) map

Traceback (most recent call last):
  File "/home/.../esp/esp-idf/tools/idf_monitor.py", line 372, in <module>
    main()
  File "/home/.../esp/esp-idf/tools/idf_monitor.py", line 363, in main
    monitor.main_loop()
  File "/home/.../esp/esp-idf/tools/idf_monitor.py", line 161, in main_loop
    self._main_loop()
  File "/home/../esp/esp-idf/tools/idf_monitor.py", line 266, in _main_loop
    super()._main_loop()
  File "/home/.../esp/esp-idf/tools/idf_monitor.py", line 226, in _main_loop
    finalize_line=not self.coredump.in_progress)
AttributeError: 'NoneType' object has no attribute 'in_progress'
```

msn444, 2022-03-24:
More specifically, it happens when idf_monitor.py is run without providing the elf file.

dobairoland, 2022-04-04:
Thanks for the report. The fix is coming soon.

closed 2022-04-07 by espressif-bot