espressif/esp-idf #8654: Static constructors don't work in components

BrucePerens, 2022-03-24:
Hi,

If I initialize a static instance of a C++ class outside of any function, or if I decorate a C function with `__attribute__ ((constructor))`, these are constructors that should be called before main. One works in a file under main/ but none work in my components. Bugs have been filed on this for a 5-year-old build system, but they don't make much sense for today's platform.

    Thanks

    Bruce

igrr, 2022-03-25:
Hi @BrucePerens, could you check if the object file where the constructor function is defined included into the output elf file? The easiest way to check this is by examining the .map file, it is located in the build directory.

The most common reason why an object file is not included into the output by the linker is that the object file does not resolve any indefinitely references from the rest of the application. In other words, if no function or variable defined in an object file is explicitly referenced somewhere else in the app, the linker won't consider the object file for linking. This rule applies to object files located inside archives (static libraries).
The common ways to solve this issue are: 1) link the object file into the application directly, not via a static library; 2) surround the static library on the linker command line with -Wl,--whole-archive and --no-whole-archive; 3) define some 'hook' function in the source file (`include_foo`) and add a linker argument to consider this function undefined (`-Wl,--undefined=include_foo`), forcing the linker to include the object file into the output file.

BrucePerens, 2022-03-25:
Hi Ivan,

Thank you for the very clear explanation, indeed there was nothing pulling those object files out of the archive. For now I have added this to the component CMakeLists.txt, but I suspect it can be improved, because I am not familiar with the build system:
```cmake
idf_build_set_property(LINK_OPTIONS
"-Wl,--whole-archive ${CMAKE_CURRENT_BINARY_DIR}/lib${COMPONENT_NAME}.a -Wl,--no-whole-archive"
APPEND)
```
This could, I guess, be made into a CMake macro in `esp-idf/tools/cmake/component.cmake`, to save others from the learning curve.

igrr, 2022-03-25:
Thanks for confirming that this solved the issue! Your code looks okay, that's probably the easiest way to do this.

There is a very recent CMake feature which could help implement this in a nicer way: https://gitlab.kitware.com/cmake/cmake/-/merge_requests/6769. However it's only going to be released in CMake 3.24. 

We can probably add a some IDF-specific way of linking the whole component, for example via 
```
idf_component_set_property(${COMPONENT_NAME} WHOLE_ARCHIVE ON)
```
The build system will check the `WHOLE_ARCHIVE` property and add `-Wl,--whole-archive` and `-Wl,--no-whole-archive` when calling `target_link_library`.

BrucePerens, 2022-03-26:
Thank you. Closing this and opening a feature request.

closed 2022-03-26 by BrucePerens