espressif/esp-idf #8655: esp_wifi_deinit doesn't return ESP_ERR_WIFI_NOT_INIT if the wifi driver is no longer initialized

cmoser96, 2022-03-25:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version: v4.1
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: Linux

## Problem Description

If the wifi driver was initialized and then deinitialized, subsequent calls of `esp_wifi_deinit()` return ESP_OK, instead of ESP_ERR_WIFI_NOT_INIT.

### Steps to reproduce

1. Call `esp_wifi_init()`
2. Call `esp_wifi_deinit()`
3. Any future calls to `esp_wifi_deinit()` will return ESP_OK, even if `esp_wifi_init()` is never called again

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_err.h"
#include "nvs_flash.h"

const char* TAG = "example";

void app_main() {
	esp_err_t ret = nvs_flash_init();
	ESP_ERROR_CHECK(ret);

	ret = esp_wifi_deinit(); // Returns 0x3001 ESP_ERR_WIFI_NOT_INIT, as expected
	ESP_LOGI(TAG, "esp_wifi_deinit() returned: %s", esp_err_to_name(ret));

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));

	ret = esp_wifi_deinit(); // Returns 0 ESP_OK, as expected
	ESP_LOGI(TAG, "esp_wifi_deinit() returned: %s", esp_err_to_name(ret));

	// Wifi driver is no longer initialized, esp_wifi_start() would return ESP_ERR_WIFI_NOT_INIT if called
	for (;;) {
		ret = esp_wifi_deinit(); // Returns 0 ESP_OK, expected 0x3001 ESP_ERR_WIFI_NOT_INIT
		ESP_LOGI(TAG, "esp_wifi_deinit() returned: %s", esp_err_to_name(ret));
		vTaskDelay(pdMS_TO_TICKS(1000));
	}
}
```

## Debug Logs

```
W (532) wifi_init: Failed to unregister Rx callbacks
E (542) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
I (542) example: esp_wifi_deinit() returned: ESP_ERR_WIFI_NOT_INIT
I (552) system_api: Base MAC address is not set
I (552) system_api: read default base MAC address from EFUSE
I (582) wifi_init: rx ba win: 6
I (582) wifi_init: tcpip mbox: 32
I (582) wifi_init: udp mbox: 6
I (582) wifi_init: tcp mbox: 6
I (582) wifi_init: tcp tx win: 5744
I (582) wifi_init: tcp rx win: 5744
I (592) wifi_init: tcp mss: 1440
I (592) wifi_init: WiFi IRAM OP enabled
I (602) wifi_init: WiFi RX IRAM OP enabled
I (612) example: esp_wifi_deinit() returned: ESP_OK
W (1612) wifi_init: Failed to unregister Rx callbacks
I (1612) example: esp_wifi_deinit() returned: ESP_OK
W (2612) wifi_init: Failed to unregister Rx callbacks
I (2612) example: esp_wifi_deinit() returned: ESP_OK
W (3612) wifi_init: Failed to unregister Rx callbacks
I (3612) example: esp_wifi_deinit() returned: ESP_OK
W (4612) wifi_init: Failed to unregister Rx callbacks
I (4612) example: esp_wifi_deinit() returned: ESP_OK
```

jack0c, 2022-05-20:
@cmoser96 It looks like a bug. Will fix it soon.

Alvin1Zhang, 2022-08-12:
Thanks for reporting, fix on master branch is available at https://github.com/espressif/esp-idf/commit/804bdc1721eb8cbb24be6ab7bd585b1a83c38b97, we are also backporting the fix to other release branches.

Will update further if any progress, thanks.

Alvin1Zhang, 2023-04-25:
Thanks for reporting, feel free to reopen.

closed 2023-04-25 by Alvin1Zhang

KOEGlike, 2024-08-17:
i have the same problem