espressif/esp-idf #8656: Openssl_Client ,Idling issues when Internet restoration in a network

supradeeplohith, 2022-03-25:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    //ESP-IDF v4.3-dirty
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB]

## Problem Description
In firebase_server.h file that i have attached, for firebase get request i am using openssl_client (sockets) example template provided by espressif , i achieved the functionality using the function but I am facing the issue of idle in ESP when my local Wi-Fi router power restores or internet restores ESP is not trying to reconnection and establish the get functionality in the code (not resumes the functionality).

### Expected Behavior
ESP is becoming idle when internet or power restores in the house. unable to resume the functionality when esp connected internet/router's power restores  
### Actual Behavior
It should not become idle when internet or Wi-Fi  power restores , and reconnect to internet and resume the get functionality.
### Steps to reproduce

1. step1 : Attach all files given in the gist 
2. step2 : Go through the firebaseserver.h file, and refer the firebase_server_event_ssl(void *p) similar function  as in openssl_client() 
3. step3 : attach the main file and all .h files 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

complete code link :  [https://gist.github.com/supradeeplohith/65ec7a3368ee02ac5743ef58e3d42563](url)

In fireabaseserver.h file snip of the function similar to openssl_client 
```cpp

// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include <string.h>

#include "openssl/ssl.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "lwip/sockets.h"
#include "lwip/netdb.h"
#include "cJSON.h"

#define OPENSSL_EXAMPLE_TARGET_TCP_PORT    443

#define OPENSSL_EXAMPLE_TARGET_NAME        "ogweinkling-default-rtdb.firebaseio.com"
#define OPENSSL_EXAMPLE_TARGET_TCP_PORT    443

char *ssl_append = ".json HTTP/1.1\r\nHost: ogweinkling-default-rtdb.firebaseio.com\r\nAccept: text/event-stream\r\nConnection: close\r\n\r\n";
char OPENSSL_EXAMPLE_REQUEST[200] = "GET /hardware_light_fan/";

#define OPENSSL_EXAMPLE_TASK_NAME          "openssl_example"

#define OPENSSL_EXAMPLE_LOCAL_TCP_PORT     443

char firebase_url[300] = "https://ogweinkling-default-rtdb.firebaseio.com/hardware_light_fan/";
int url_size;

char *http_send_data;
int ret;
SSL_CTX *ctx;
SSL *ssl;
int sockfd;

xTaskHandle openssl_handle;
QueueHandle_t xQueue2;

void intial_update();
struct AMessage *pxMessage;
int cat_flag = 0 ;
void firebase_server_event_ssl(void *p)
{
	//printf ( "Flag : %d \n",cat_flag);
	const static char *TAG = "openssl_example";
	if ( cat_flag == 0 )
	{
	strcat (OPENSSL_EXAMPLE_REQUEST,e_user_id);
	strcat (OPENSSL_EXAMPLE_REQUEST,ssl_append);
	cat_flag = 1;
	}
    struct sockaddr_in sock_addr;
    struct hostent *hp;
    struct ip4_addr *ip4_addr;

    int recv_bytes = 0;

    char send_data[200];
    strcpy(send_data,OPENSSL_EXAMPLE_REQUEST);
    const int send_bytes = sizeof(send_data);

    ESP_LOGI(TAG, "OpenSSL demo thread start OK");

    ESP_LOGI(TAG, "get target IP address");
    hp = gethostbyname(OPENSSL_EXAMPLE_TARGET_NAME);
    if (!hp) {
        ESP_LOGI(TAG, "failed");
        goto failed1;
    }
    ESP_LOGI(TAG, "OK");

    ip4_addr = (struct ip4_addr *)hp->h_addr;
    ESP_LOGI(TAG, IPSTR, IP2STR(ip4_addr));

    ESP_LOGI(TAG, "create SSL context ......");
    ctx = SSL_CTX_new(TLSv1_1_client_method());
    if (!ctx) {
        ESP_LOGI(TAG, "failed");
        goto failed1;
    }
    ESP_LOGI(TAG, "OK");

    ESP_LOGI(TAG, "create socket ......");
    sockfd = socket(AF_INET, SOCK_STREAM, 0);
    if (sockfd < 0) {
        ESP_LOGI(TAG, "failed");
        goto failed2;
    }
    ESP_LOGI(TAG, "OK");

    ESP_LOGI(TAG, "bind socket ......");
    memset(&sock_addr, 0, sizeof(sock_addr));
    sock_addr.sin_family = AF_INET;
    sock_addr.sin_addr.s_addr = 0;
    sock_addr.sin_port = htons(OPENSSL_EXAMPLE_LOCAL_TCP_PORT);
    ret = bind(sockfd, (struct sockaddr*)&sock_addr, sizeof(sock_addr));
    if (ret) {
        ESP_LOGI(TAG, "failed");
        goto failed3;
    }
    ESP_LOGI(TAG, "OK");

    ESP_LOGI(TAG, "socket connect to remote %s ......", OPENSSL_EXAMPLE_TARGET_NAME);
    memset(&sock_addr, 0, sizeof(sock_addr));
    sock_addr.sin_family = AF_INET;
    sock_addr.sin_addr.s_addr = ip4_addr->addr;
    sock_addr.sin_port = htons(OPENSSL_EXAMPLE_TARGET_TCP_PORT);
    ret = connect(sockfd, (struct sockaddr*)&sock_addr, sizeof(sock_addr));
    if (ret) {
        ESP_LOGI(TAG, "failed");
        goto failed3;
    }
    ESP_LOGI(TAG, "OK");

    ESP_LOGI(TAG, "create SSL ......");
    ssl = SSL_new(ctx);
    //SSL_CTX_set_timeout(ctx,300);
    if (!ssl) {
        ESP_LOGI(TAG, "failed");
        goto failed3;
    }
    ESP_LOGI(TAG, "OK");

    SSL_set_fd(ssl, sockfd);

    ESP_LOGI(TAG, "SSL connected to %s port %d ......",
        OPENSSL_EXAMPLE_TARGET_NAME, OPENSSL_EXAMPLE_TARGET_TCP_PORT);
    ret = SSL_connect(ssl);
    if (!ret) {
        ESP_LOGI(TAG, "failed " );
        goto failed4;
    }
    ESP_LOGI(TAG, "OK");

    ESP_LOGI(TAG, "send https request to %s port %d ......",
        OPENSSL_EXAMPLE_TARGET_NAME, OPENSSL_EXAMPLE_TARGET_TCP_PORT);
    ret = SSL_write(ssl, send_data, send_bytes);
    if (ret <= 0) {
        ESP_LOGI(TAG, "failed");
        goto failed5;
    }
    ESP_LOGI(TAG, "OK");
    //printf(" SSSL _Flag : %d && mesh_flag : %d ",ssl_flag, mesh_flag);
    if( xQueue2 == 0 )
    {
	xQueue2 = xQueueCreate( 72, sizeof( struct AMessage * ) );
    }
	//printf("Connected \n");
    do {
        ret = SSL_read(ssl, recv_buf, OPENSSL_EXAMPLE_RECV_BUF_LEN - 1);
        //printf(" SSL Read : %d \n",ret);
        if (ret <= 0) {
            break;
        }
        recv_buf[ret] = '\0';
        recv_bytes += ret;
        ESP_LOGI(TAG, "%s", recv_buf);
        ssl_flag = 1 ;
        if ( strlen(recv_buf) > 1000 )
        {
        	intial_update();
        }
        printf (" Data : %s , Heap_memory : %d , count : %d \n",(char*)recv_buf,esp_get_free_heap_size(),ssl_idle);
        printf(" Data size : %d \n",strlen(recv_buf));
        ssl_idle = 0 ;
        if ( strlen(recv_buf) < 200 )
        {
        if ( strncmp( recv_buf, "event: put", 10) == 0 )
        {
        	printf(" Put Request : %s \n", recv_buf);
        	if( xQueue2 != 0 )
        	{
        		strcpy(xMessage.ucData,recv_buf);
        	    pxMessage = & xMessage;
        	    xQueueGenericSend( xQueue2, ( void * ) &pxMessage, ( TickType_t ) 0, queueSEND_TO_BACK );
        	}
         }
        }
    } while (1);

    ESP_LOGI(TAG, "totally read %d bytes data from %s ......", recv_bytes, OPENSSL_EXAMPLE_TARGET_NAME);
    vTaskDelete(NULL);
    //printf("Deleted \n");

failed5:
    //SSL_shutdown(ssl);
    //printf("Deleted \n");
    ssl_idle = 40;
    vTaskDelete(NULL);
failed4:
    //SSL_free(ssl);
    //ssl = NULL;
    //printf("Deleted \n");
    ssl_idle = 40;
    vTaskDelete(NULL);
failed3:
    //close(sockfd);
    //sockfd = -1;
    //printf("Deleted \n");
    ssl_idle = 40;
    vTaskDelete(NULL);
failed2:
    //SSL_CTX_free(ctx);
    //ctx = NULL;
    //printf("Deleted \n");
    ssl_idle = 40;
    vTaskDelete(NULL);
failed1:
    //printf("Deleted \n");
    ssl_idle = 40;
    vTaskDelete(NULL);
    //return ;
}

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] coredump (This provides stacks of tasks.)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8347766/sdkconfig.txt)
![Capture1](https://user-images.githubusercontent.com/42844966/160061220-040229b2-caf1-48d2-b7bd-1d79cbc24bbb.PNG)
[r4.log](https://github.com/espressif/esp-idf/files/8347771/r4.log)
[r2.log](https://github.com/espressif/esp-idf/files/8347772/r2.log)
[r3.log](https://github.com/espressif/esp-idf/files/8347774/r3.log)

david-cermak, 2022-04-01:
Hi @supradeeplohith 

First of all, I'd like to discourage any complex use of IDF's openssl component. Although it's still supported in `v4.3`, it's been deprecated in `v4.4` and already removed on the latest.
I'd suggest using `mbedtls` directly, or `esp-tls` instead. Seeing that you're connecting over `https`, please check also the `https_request` and `esp_https_client` components in IDF.
Also, IDF examples in general are not meant to provide a complete solution, just the initial idea. Users have to figure out all potential error handling based on the application needs.

If you really want to use openssl client, please try to:
* look at the socket itself, check it's readability (+writability) using `select()` to evaluate error and timeout conditions
* check for disconnection events, maybe adjust TCP parameters to restart the connection quickly.
* use some keepalive functionality of the server itself

supradeeplohith, 2022-04-11:
Hi @david-cermak 

Thanks for the response, I've tried https_request and esp_https_client i can able to get the data but similar to openssl_client updated data in json i am able to get, could you please assist me on this.

david-cermak, 2022-04-11:
@supradeeplohith could you please share the code then? How did you implement the recovery.
I don't see anything like that in the code above, you just exit the task (on success and on failure too), correct? (note that your gist is no longer accessible, so just checked the code pasted to the comment)

david-cermak, 2022-04-12:
@supradeeplohith Thanks for sharing, but what I see is the default IDF example with updated URI, no recovery, correct?

> exactly i need to get the data ie., updated node data in realtimeData base (firebase) insted of steaming and getting the data every time

I think you have to define your own strategy, how you want to handle the reported **internet restoration** issues. The first step is to check for connection/read errors. None of your examples actually did that. Please try to return errors and evaluate them:
```patch
-static void http_rest_with_url(void)
+static esp_err_t http_rest_with_url(void)
```
The second step is to decide when you want to update your data, i.e. periodically, on some event (reconnection)? But this is very application specific and I cannot help with that.

supradeeplohith, 2022-05-19:
@david-cermak @Alvin1Zhang Hi, regarding to openssl_client() template socket connection I'm facing new issue that no data is receiving  in the template [https://github.com/espressif/esp-idf/tree/v4.4/examples/protocols/esp_http_client](url)  which im using  integrated to google firebase recently im facing the issue, tell me whether openssl_client() has been deprecated ?

Please respond ASAP 
Regards,
SUPRADEEP M A

supradeeplohith, 2022-05-19:
@david-cermak please i expect suggestion from you, that give me suggestion that is similar to the openssl_client() and does the same functionality for same.
 

> HI @david-cermak
> 
> Please find gist link : [https://gist.github.com/supradeeplohith/a59c5e0fb64d40e1b349f6168d95e8c3](url) [https://gist.github.com/supradeeplohith/a59c5e0fb64d40e1b349f6168d95e8c3](url) [<script src="https://gist.github.com/supradeeplohith/a59c5e0fb64d40e1b349f6168d95e8c3.js"></script>](url) above link is template from espressif to get the data from cloud ie.., esp_http_client exactly i need to get the data ie., updated node data in realtimeData base (firebase) insted of steaming and getting the data every time Thanks
please re attach the reference code
My requirement is to get the data from firebase and i need persistent connection no loss of connection should not happen.

Basically we are integrating mobile app to firebase, if any update is occurred  with same some connection and with same same database i would like to receive the data 

please help me for the same

david-cermak, 2022-05-20:
Hi @supradeeplohith 

1) you can still use openssl client on IDF `v4.3` and `v4.4`.

2) if I understand correctly, you're looking for some kind of event, so you don't have to periodically check the state of you data, but just wait for a notification, correct? Unfortunately I'm not familiar with google firebase, but such notifications are usually implemented using some subscription mechanism. Could you please provide more details about these notifications, how it works?

Thanks!

supradeeplohith, 2022-05-20:
Hi @david-cermak
1. Openssl client is not working can please tell me why? previously i was able to get at least data from last 3 days im not able get the data itself. please help me
2.  as per the template openssl client example template, i have mentioned above , i will get the data as stream from as a form 
[image]
[image]
of json ,after that  it only gives only data when update in firebase for particular node it will gives..i want same as above explanation application @david-cermak 
3. overall - if any update in firebase for particular node openssl was giving updated data for particular node in json,
4. i request you to please check the code - tell me the reason why this openssl client is not working as it was functioning before, now its completely not working

closed 2022-05-20 by supradeeplohith