espressif/esp-idf #8657: ESP-IDF FOTA via the webserver fails when increasing chunk size

krupis, 2022-03-25:
I have picked up file_serving example from the esp-idf examples and modified it slightly.
This is the original upload handler that handles the file upload from the web to the SPIFFS. 

```
// Handler to upload a file onto the server 
static esp_err_t upload_post_handler(httpd_req_t *req)
{
    char filepath[FILE_PATH_MAX];
    FILE *fd = NULL;
    struct stat file_stat;

    // Skip leading "/upload" from URI to get filename 
    // Note sizeof() counts NULL termination hence the -1 
    const char *filename = get_path_from_uri(filepath, ((struct file_server_data *)req->user_ctx)->base_path,
                                             req->uri + sizeof("/upload") - 1, sizeof(filepath));
    if (!filename) {
        // Respond with 500 Internal Server Error 
        httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Filename too long");
        return ESP_FAIL;
    }

    // Filename cannot have a trailing '/' 
    if (filename[strlen(filename) - 1] == '/') {
        ESP_LOGE("UPLOAD", "Invalid filename : %s", filename);
        httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Invalid filename");
        return ESP_FAIL;
    }

    if (stat(filepath, &file_stat) == 0) {
        ESP_LOGE("UPLOAD", "File already exists : %s", filepath);
        // Respond with 400 Bad Request 
        httpd_resp_send_err(req, HTTPD_400_BAD_REQUEST, "File already exists");
        return ESP_FAIL;
    }

    // File cannot be larger than a limit 
    if (req->content_len > MAX_FILE_SIZE) {
        ESP_LOGE("UPLOAD", "File too large : %d bytes", req->content_len);
        // Respond with 400 Bad Request 
        httpd_resp_send_err(req, HTTPD_400_BAD_REQUEST,
                            "File size must be less than "
                            MAX_FILE_SIZE_STR "!");
        //Return failure to close underlying connection else the
        // incoming file content will keep the socket busy 
        return ESP_FAIL;
    }

    fd = fopen(filepath, "w");
    if (!fd) {
        ESP_LOGE("UPLOAD", "Failed to create file : %s", filepath);
        // Respond with 500 Internal Server Error //
        httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Failed to create file");
        return ESP_FAIL;
    }

    ESP_LOGI("UPLOAD", "Receiving file : %s...", filename);

    //Retrieve the pointer to scratch buffer for temporary storage 
    char *buf = ((struct file_server_data *)req->user_ctx)->scratch;
    int received;
    // Content length of the request gives
    // the size of the file being uploaded 
    int remaining = req->content_len;
    while (remaining > 0) {
        ESP_LOGI("UPLOAD", "Remaining size : %d", remaining);
        // Receive the file part by part into a buffer 
        if ((received = httpd_req_recv(req, buf, MIN(remaining, SCRATCH_BUFSIZE))) <= 0) {
            if (received == HTTPD_SOCK_ERR_TIMEOUT) {
                // Retry if timeout occurred 
                continue;
[... 1333 lines cut ...]
[1B][0;32mI (123909) UPLOAD: Remaining size : 12384[1B][0m
[1B][0;32mI (124009) UPLOAD: Remaining size : 10336[1B][0m
[1B][0;32mI (124109) UPLOAD: Remaining size : 8288[1B][0m
[1B][0;32mI (124209) UPLOAD: Remaining size : 6240[1B][0m
[1B][0;32mI (124309) UPLOAD: Remaining size : 4192[1B][0m
[1B][0;32mI (124409) UPLOAD: Remaining size : 2144[1B][0m
[1B][0;32mI (124509) UPLOAD: Remaining size : 96[1B][0m
[1B][0;32mI (124509) UPLOAD: File reception complete[1B][0m
[1B][0;32mI (124509) esp_image: segment 0: paddr=001b0020 vaddr=3f400020 size=35908h (219400) map[1B][0m
[1B][0;32mI (124569) esp_image: segment 1: paddr=001e5930 vaddr=3ffbdb60 size=05860h ( 22624) [1B][0m
[1B][0;32mI (124589) esp_image: segment 2: paddr=001eb198 vaddr=40080000 size=04e80h ( 20096) [1B][0m
[1B][0;32mI (124609) esp_image: segment 3: paddr=001f0020 vaddr=400d0020 size=ea284h (959108) map[1B][0m
[1B][0;32mI (124859) esp_image: segment 4: paddr=002da2ac vaddr=40084e80 size=16d70h ( 93552) [1B][0m
[1B][0;32mI (124879) esp_image: segment 5: paddr=002f1024 vaddr=50000000 size=00010h (    16) [1B][0m
[1B][0;32mI (124919) esp_image: segment 0: paddr=001b0020 vaddr=3f400020 size=35908h (219400) map[1B][0m
[1B][0;32mI (124989) esp_image: segment 1: paddr=001e5930 vaddr=3ffbdb60 size=05860h ( 22624) [1B][0m
[1B][0;32mI (125009) esp_image: segment 2: paddr=001eb198 vaddr=40080000 size=04e80h ( 20096) [1B][0m
[1B][0;32mI (125019) esp_image: segment 3: paddr=001f0020 vaddr=400d0020 size=ea284h (959108) map[1B][0m
[1B][0;32mI (125259) esp_image: segment 4: paddr=002da2ac vaddr=40084e80 size=16d70h ( 93552) [1B][0m
[1B][0;32mI (125299) esp_image: segment 5: paddr=002f1024 vaddr=50000000 size=00010h (    16) [1B][0m
[1B][0;32mI (125389) UPDATE: Restarting...[1B][0m

```
</details>
I would like to understand what is happening and how to make it more efficient. Is the wifi not able to keep up with reading 2MB chunks very fast from the webserver? How to make this as efficient as possible so that the OTA writes only when the ESP32 sucesfully read all the data from the webserver?

mahavirj, 2022-03-28:
@krupis 

Can you please check stack utilization of `httpd` task? Alternatively if you change `SEND_DATA` buffer to be `static` does it fix the problem?

Note: If this does not help, please share your example project along with `sdkconfig`

krupis, 2022-03-28:
Hello. I think I have found a problem. It was a misunderstanding of how httpd_req_recv works from my side.

The issue is that I call 
```
esp_ota_write(well_done_handle, buf, read_chunk_length);
```
passing the read_chunk_length parameter which is not correct.  It should be as following;
```
esp_ota_write(well_done_handle, buf, received );
```

That is because httpd_req_recv  may not be able to read the amount of data that is passed as a length parameter. The actual data that has been received will be returned and stored in parameter received.

mahavirj, 2022-03-29:
Thanks for confirming. I will go ahead and close the issue then.

closed 2022-03-29 by mahavirj