espressif/esp-idf #8658: Since latest esp-idf update we are missing 20kb of heap and get a lot of heap corruptions

0xFEEDC0DE64, 2022-03-25:
Our firmware project has been running fine before our latest idf upgrade, which was an upgrade from an 3 months old if to latest master.

With the idf update we faced a loss of around 20kb of free heap, even with our app unchanged. Also our ESP32s keep crashing every few hours now because of heap corruptions.

I suspect the wifi blob as the stack trace always shows the detected heap corruptions caused by ppTask()

Can we downgrade to a 3 months old wifi blob with newest idf? Have the interfaces been changed?

Dazza0, 2022-03-25:
@0xFEEDC0DE64 Could you provide the following:

- IDF tag/commit ID before and after upgrade
- Crash log with backtrace

>  Have the interfaces been changed?

Depends on whether you upgraded switched to a different [major release](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/versions.html) of not. Within the same major version, the API should be the same.

> Can we downgrade to a 3 months old wifi blob with newest idf?

I don't think we support this.

0xFEEDC0DE64, 2022-03-25:
I upgraded esp-idf from df1f9ec26e81138494462451889e7cb05dfc4ad4 to master

```
CORRUPT HEAP: Invalid data at 0x3ffe631c. Expected 0xfefefefe got 0x00000000                                                                                                                                                                                                                                                  
                                                                                                                                                                                                                                                                                                                              
assert failed: multi_heap_malloc multi_heap_poisoning.c:237 (ret)                                                                                                                                                                                                                                                             
                                                                                                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                                                                                              
Backtrace:0x40081f12:0x3ffd1a900x40089375:0x3ffd1ab0 0x40091206:0x3ffd1ad0 0x4008eba2:0x3ffd1bf0 0x40082062:0x3ffd1c10 0x40082085:0x3ffd1c40 0x400820b9:0x3ffd1c60 0x40091295:0x3ffd1c80 0x400912b8:0x3ffd1ca0 0x402578c7:0x3ffd1cc0 0x40258d1f:0x3ffd1cf0 0x4024a614:0x3ffd1d20 0x40235f89:0x3ffd1d90 0x40236771:0x3ffd1ea0 0
x40234c7b:0x3ffd1f00 0x40094a99:0x3ffd1f20 0x40092ae8:0x3ffd1f40 0x4008cd11:0x3ffd1f70                                                                                                                                                                                                                                        
0x40081f12: panic_abort at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/esp_system/panic.c:403                                                                                                                                                                                                                 
                                                                                                                                                                                                                                                                                                                              
0x40089375: esp_system_abort at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/esp_system/esp_system.c:135                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                                                                              
0x40091206: __assert_func at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/newlib/assert.c:85                                                                                                                                                                                                                   
                                                                                                                                                                                                                                                                                                                              
0x4008eba2: multi_heap_malloc at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/heap/multi_heap_poisoning.c:237                                                                                                                                                                                                  
 (inlined by) multi_heap_malloc at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/heap/multi_heap_poisoning.c:219                                                                                                                                                                                                
                                                                                                                                                                                                                                                                                                                              
0x40082062: heap_caps_malloc_base at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/heap/heap_caps.c:139                                                                                                                                                                                                        [... cut ...]                                  
                                                                                                                                                                                                                                                                                                                              
0x4008cd11: vPortTaskWrapper at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141
```

AxelLin, 2022-03-28:
> I upgraded esp-idf from [df1f9ec](https://github.com/espressif/esp-idf/commit/df1f9ec26e81138494462451889e7cb05dfc4ad4) to master
> 

Can you run git bisect to find the bad commit?

MaxwellAlan, 2022-03-28:
hi @0xFEEDC0DE64 
Can you help find the bad commit like @AxelLin said or can you provide a demo code for us to reproduce this issue?
From your description, there may be two issue, one is why the heap size decrease 20kb, and the other is why will crash(maybe also the cause by the issue one). Can you share more details, such as how your app using WiFi(STA+SoftAP or only SoftAP, or with BT etc?)?

MaxwellAlan, 2022-03-28:
hi @0xFEEDC0DE64 
I using example/wifi/iperf test with commit df1f9ec2 and  e3c0229d, using **heap** cmd to show the heap usage before WiFi init and WiFi SoftAP start, the result is below:

IDF commit | Before WiFi Init | SoftAP Start
-- | -- | --
df1f9ec2 | 216448 B | 214016 B
e3c0229d | 215220 B | 212572 B

Can you also upload your app sdkconfig?

jack0c, 2022-05-20:
@0xFEEDC0DE64 any updates?

closed 2022-07-13 by Alvin1Zhang