espressif/esp-idf #8659: Http Webserver doesn't work after 2nd time softap init

danhpham1510, 2022-03-25:
## Environment

- Development Kit:      ESP-WROOM-32
- Kit version: V1
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.4
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: Yes (VSCode)
- Power Supply:         USB

## Problem Description
I have combined softAP and webserver examples so that a webserver runs over the softAP (192.168.4.1;80). It works fine the first time I start the AP, but after I deinit/terminate the AP and start a new one, I can't connect to the webserver at the old IP address anymore.

### Expected Behavior
After starting a new softap. the webserver still runs and work on 192.168.4.1:80
### Actual Behavior
After deinit and start a new softap, can't connect to webserver.

### Code to reproduce this issue

```
   static httpd_handle_t server = NULL;
  //Initialize NVS
  esp_err_t ret = nvs_flash_init();
  if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
    ESP_ERROR_CHECK(nvs_flash_erase());
    ret = nvs_flash_init();
  }
  ESP_ERROR_CHECK(ret);

  ESP_LOGI(TAG, "ESP_WIFI_MODE_AP");
  esp_netif_t * ap =wifi_init_softap();

  ESP_ERROR_CHECK(esp_netif_init());

  ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_AP_STAIPASSIGNED, &connect_handler, &server));
  ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &disconnect_handler, &server));

  vTaskDelay(10000/portTICK_PERIOD_MS);
  //deinit
  esp_wifi_stop();
  esp_wifi_deinit();
  esp_netif_destroy(ap);
  esp_wifi_set_mode(WIFI_MODE_NULL);
  esp_event_loop_delete_default();
  vTaskDelay(5000/portTICK_PERIOD_MS);   
  //start new ap
  ap = wifi_init_softap();
```

## Debug Logs

```
I (0) cpu_start: App cpu up.
I (457) cpu_start: Pro cpu start user code
I (457) cpu_start: cpu freq: 160000000
I (457) cpu_start: Application information:
I (461) cpu_start: Project name:     wifi
I (466) cpu_start: App version:      8d87a30-dirty
I (472) cpu_start: Compile time:     Mar 25 2022 11:05:29
I (478) cpu_start: ELF file SHA256:  39c4ce1e78a3ce29...
I (484) cpu_start: ESP-IDF:          v4.4-dirty
I (489) heap_init: Initializing. RAM available for dynamic allocation:
I (496) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (502) heap_init: At 3FFB74C8 len 00028B38 (162 KiB): DRAM
I (508) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (515) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (521) heap_init: At 400942E4 len 0000BD1C (47 KiB): IRAM
I (529) spi_flash: detected chip: generic
I (532) spi_flash: flash io: dio
I (537) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (623) webserver: ESP_WIFI_MODE_AP
I (633) wifi:wifi driver task: 3ffc05c8, prio:23, stack:6656, core=0
I (633) system_api: Base MAC address is not set
I (633) system_api: read default base MAC address from EFUSE
I (653) wifi:wifi firmware version: 7679c42
I (653) wifi:wifi certification version: v7.0
I (653) wifi:config NVS flash: enabled
I (653) wifi:config nano formating: disabled
I (663) wifi:Init data frame dynamic rx buffer num: 32
I (663) wifi:Init management frame dynamic rx buffer num: 32
I (673) wifi:Init management short buffer num: 32
I (673) wifi:Init dynamic tx buffer num: 32
I (673) wifi:Init static rx buffer size: 1600
I (683) wifi:Init static rx buffer num: 10
I (683) wifi:Init dynamic rx buffer num: 32
I (693) wifi_init: rx ba win: 6
I (693) wifi_init: tcpip mbox: 32
I (693) wifi_init: udp mbox: 6
I (703) wifi_init: tcp mbox: 6
I (703) wifi_init: tcp tx win: 5744
I (713) wifi_init: tcp rx win: 5744
I (713) wifi_init: tcp mss: 1440
I (713) wifi_init: WiFi IRAM OP enabled
I (723) wifi_init: WiFi RX IRAM OP enabled
I (733) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (833) wifi:mode : softAP (7c:9e:bd:48:0b:69)
I (833) wifi:Total power save buffer number: 16
I (843) wifi:Init max length of beacon: 752/752
I (843) wifi:Init max length of beacon: 752/752
I (843) webserver: wifi_init_softap finished. SSID:DANH-ESP32 password:password channel:1
I (1803) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (1803) wifi:station: b8:8a:60:cd:6a:fe join, AID=1, bgn, 40U
I (1823) webserver: station b8:8a:60:cd:6a:fe join, AID=1
I (1873) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (1883) webserver: Starting webserver
I (1883) webserver: Starting server on port: '80'
I (1883) webserver: Registering URI handlers
W (3043) wifi:<ba-add>idx:2 (ifx:1, b8:8a:60:cd:6a:fe), tid:0, ssn:61, winSize:64
I (12073) wifi:station: b8:8a:60:cd:6a:fe leave, AID = 1, bss_flags is 658531, bss:0x3ffc6924
I (12073) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
W (12073) wifi:<ba-del>idx
I (12083) webserver: station b8:8a:60:cd:6a:fe leave, AID=1
I (15853) wifi:flush txq
I (15853) wifi:stop sw txq
I (15853) wifi:lmac stop hw txq
I (15853) wifi:Deinit lldesc rx mblock:10
I (20873) wifi:wifi driver task: 3ffc7ad4, prio:23, stack:6656, core=0
I (20883) wifi:wifi firmware version: 7679c42
I (20893) wifi:wifi certification version: v7.0
I (20893) wifi:config NVS flash: enabled
I (20893) wifi:config nano formating: disabled
I (20893) wifi:Init data frame dynamic rx buffer num: 32
I (20903) wifi:Init management frame dynamic rx buffer num: 32
I (20903) wifi:Init management short buffer num: 32
I (20913) wifi:Init dynamic tx buffer num: 32
I (20913) wifi:Init static rx buffer size: 1600
I (20913) wifi:Init static rx buffer num: 10
I (20923) wifi:Init dynamic rx buffer num: 32
I (20923) wifi_init: rx ba win: 6
I (20933) wifi_init: tcpip mbox: 32
I (20933) wifi_init: udp mbox: 6
I (20933) wifi_init: tcp mbox: 6
I (20943) wifi_init: tcp tx win: 5744
I (20943) wifi_init: tcp rx win: 5744
I (20953) wifi_init: tcp mss: 1440
I (20953) wifi_init: WiFi IRAM OP enabled
I (20963) wifi_init: WiFi RX IRAM OP enabled
I (20973) wifi:mode : softAP (7c:9e:bd:48:0b:69)
I (20973) wifi:Total power save buffer number: 16
I (20973) webserver: wifi_init_softap finished. SSID:DANH-ESP32 password:password channel:1
I (23763) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (23763) wifi:station: b8:8a:60:cd:6a:fe join, AID=1, bgn, 40U
I (23783) webserver: station b8:8a:60:cd:6a:fe join, AID=1
I (28843) wifi:station: b8:8a:60:cd:6a:fe leave, AID = 1, bss_flags is 658531, bss:0x3ffc49a4
I (28843) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (28853) webserver: station b8:8a:60:cd:6a:fe leave, AID=1

```

Am I missing something when deinitializing the old AP?

xueyunfei998, 2022-04-21:
hi @danhpham1510 

Did you implement this function `wifi_init_softap` yourself?

This function has no return value in idf. It is a void type.

danhpham1510, 2022-04-21:
> hi @danhpham1510
> 
> Did you implement this function `wifi_init_softap` yourself?
> 
> This function has no return value in idf. It is a void type.

Hi, I wrote this myself, the only difference in my version compared to IDF's example is that my function returns  an instance of esp_netif_t * so i can run esp_netif_destroy()

xueyunfei998, 2022-04-23:
hi @danhpham1510 
I (20973) wifi:mode : softAP (7c:9e:bd:48:0b:69)
I (20973) wifi:Total power save buffer number: 16
I (20973) webserver: wifi_init_softap finished. SSID:DANH-ESP32 password:password channel:1
I (23763) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (23763) wifi:station: b8:8a:60:cd:6a:fe join, AID=1, bgn, 40U
I (23783) webserver: station b8:8a:60:cd:6a:fe join, AID=1
I (28843) wifi:station: b8:8a:60:cd:6a:fe leave, AID = 1, bss_flags is 658531, bss:0x3ffc49a4
I (28843) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (28853) webserver: station b8:8a:60:cd:6a:fe leave, AID=1

From here, it seems that four handshakes failed.

I need your sdkconfig file, and put your wifi_init_softap function code

MaxwellAlan, 2024-01-11:
Hi @danhpham1510 

We will close this issue. Feel free to reopen if you still have the issue.

closed 2024-01-11 by MaxwellAlan