espressif/esp-idf #8662: pthread allow Stack on Spiram

f-hoepfinger-hr-agrartechnik, 2022-03-25:
when you have a lot of Tasks, 
internal Memory might be used up fast. 

now, 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-allow-stack-external-memory
allow use of Stack on SPIRAM. 
but not via C++ Pthreads. 
https://github.com/espressif/esp-idf/tree/master/examples/cxx/pthread

would be a nice Extension right ? 

Thanks, 
Franz

0xjakob, 2022-03-28:
@f-hoepfinger-hr-agrartechnik

> would be a nice Extension right ?

Certainly! But I'm afraid that the [constructor of std::thread](https://en.cppreference.com/w/cpp/thread/thread/thread) does not allow any additional parameters than the actual function to be executed in the thread.

Maybe I missed something, so if you have another idea, please let me know!

franz-ms-muc, 2022-03-29:
esp_pthread_set_cfg(&cfg) 
The cfg could be extended.

0xjakob, 2022-03-30:
@f-hoepfinger-hr-agrartechnik @franz-ms-muc I see now (Should have looked at the example more thoroughly). Just to clarify: you would like a possibility to configure `std::thread` via the `esp_pthread` API? If that's correct, then I think it's possible. We would add this to our backlog but we currently have some other high-priority tasks. So it might take some (> 1 month). If you eventually implement something like this yourself, feel free to open a pull request and then we can discuss there.

f-hoepfinger-hr-agrartechnik, 2022-03-30:
Hi,

we did one implementation here: 
https://github.com/hr-agrartechnik/esp-idf/blob/HR_4.4_ram/components/pthread/pthread.c

but i would not want to open a Pull Request as i feel the Quality is not good enough for a Pull request. 
maybe later when we have done a bunch more test. 

it is "generally" running OK, so we use it in a Prototype.

f-hoepfinger-hr-agrartechnik, 2022-09-23:
Ported same to 4.4.2
https://github.com/hr-agrartechnik/esp-idf/tree/HR_4.4.2_ram

any outlook that we can get such in ESP-IDF ?

f-hoepfinger-hr-agrartechnik, 2023-01-26:
will be here shortly.
https://github.com/espressif/esp-idf/pull/10623

closed 2024-01-30 by espressif-bot