espressif/esp-idf #8663: Authmode above WPA2_PSK for AP mode fails

mmaciola, 2022-03-25:
Can't increase authmode above WIFI_AUTH_WPA2_PSK for AP mode.

### Steps to reproduce
Copy example https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c
Use configuration as below (please note the example sets pmf_cfg that is not available for IDF-4.4):
```
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .ssid_len = strlen(EXAMPLE_ESP_WIFI_SSID),
            .channel = EXAMPLE_ESP_WIFI_CHANNEL,
            .password = EXAMPLE_ESP_WIFI_PASS,
            .max_connection = EXAMPLE_MAX_STA_CONN,
            .authmode = WIFI_AUTH_WPA2_WPA3_PSK,
        },
    };
```

### Behavior
```
(637) wifi_init: rx ba win: 6
(637) wifi_init: tcpip mbox: 32
(647) wifi_init: udp mbox: 6
(647) wifi_init: tcp mbox: 6
(657) wifi_init: tcp tx win: 5744
(657) wifi_init: tcp rx win: 5744
(657) wifi_init: tcp mss: 1440
(667) wifi_init: WiFi IRAM OP enabled
(667) wifi_init: WiFi RX IRAM OP enabled
W (677) wifi:authmode is WEP or WPA2_ENTERPRISE or wrong, 7

ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40088334
file: „../main/main.c” line 74
func: wifi_init_softap
expression: esp_wifi_set_config(WIFI_IF_AP, &wifi_config)

abort() was called at PC 0x40088337 on core 0
```

## Environment
```
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)

Operating System: mac os x
Java Runtime Version: 17.0.2+8
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.4.1.202202101602
ESP-IDF v4.4-dirty
Python set for IDF_PYTHON_ENV: Python 3.9.5
Python declared in PATH variable: Python 2.7.18
```

nishanth-radja, 2022-03-28:
@mmaciola The support of WPA3 is not yet supported of AP.It will be available soon.

AxelLin, 2022-03-29:
https://github.com/espressif/esp-idf/issues/6574

igrr, 2022-03-29:
Thanks for linking to the older issue @AxelLin. Let's close this issue and use the other one to track this.
@mmaciola You can follow the link to https://github.com/espressif/esp-idf/issues/6574 and click "Subscribe" button there to get notified when this feature is implemented.

shreyassheth1, 2023-04-26:
Hey @mmaciola  wpa3 softAP is not supported on v4.4. It is only supported on release/v5.1