espressif/esp-idf #8666: pull_down_en takes no effect for some gpios in ESP32C3

lz00qs, 2022-03-26:
## Environment

- Development Kit: https://item.taobao.com/item.htm?spm=a230r.1.14.7.72bd57deHXCzZp&id=645460125031&ns=1&abbucket=8#detail
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used:  ESP32C3
- IDF version (run ``git describe --tags`` to find it): v4.4-rc1
- Build System: Visual studio code
- Operating System:     Windows
- Using an IDE?:  Visual studio code
- Power Supply:  LDO 3.3V

## Problem Description

The  `pull_down_en`  option in  `gpio_config_t`  takes no effect for  `GPIO_NUM_8`  and  `GPIO_NUM_9` .

### Expected Behavior

When I let gpio8 or gpio9 in float state, I get low level.

### Actual Behavior

When I let gpio8 or gpio9 in float state, I got gpio level, it returned high level.

### Steps to reproduce

Just run the code and let the gpio that mentioned upper in float state.

### Code to reproduce this issue

```cpp
void app_main(void)
{
    esp_err_t err;

    ESP_LOGI(TAG, "Initializing...");
    
    err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);

    err = bluetooth_init();
    if (err)
    {
        ESP_LOGE(TAG, "esp32_bluetooth_init failed (err %d)", err);
        return;
    }

    /* Open nvs namespace for storing/restoring mesh example info */
    err = ble_mesh_nvs_open(&NVS_HANDLE);
    if (err)
    {
        return;
    }

    ble_mesh_get_dev_uuid(dev_uuid);

    /* Initialize the Bluetooth Mesh Subsystem */
    err = ble_mesh_init();
    if (err)
    {
        ESP_LOGE(TAG, "Bluetooth mesh init failed (err %d)", err);
    }
    static gpio_config_t gpioConfig = {};

    gpioConfig.intr_type = GPIO_INTR_DISABLE;
    gpioConfig.mode = GPIO_MODE_INPUT;
    gpioConfig.pull_down_en = 1;
    gpioConfig.pull_up_en = 0;
    gpioConfig.pin_bit_mask = 1ULL << GPIO_NUM_8;
    gpio_config(&gpioConfig);

    while (1)
    {
        ESP_LOGI(TAG,"GPIO 8 level: %d",gpio_get_level(GPIO_NUM_8));
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

lz00qs, 2022-03-26:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/8355318/sdkconfig.zip)

0xjakob, 2022-03-28:
@lz00qs Could you measure the voltage level on that pin while running the code above?

lz00qs, 2022-03-28:
> 

2.5V

lz00qs, 2022-03-28:
#include <stdio.h>
#include <freertos/portmacro.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>
#include <esp_log.h>
#include <driver/gpio.h>
#define TAG "test"
void app_main(void)
{
    static gpio_config_t gpioConfig = {};

    gpioConfig.intr_type = GPIO_INTR_DISABLE;
    gpioConfig.mode = GPIO_MODE_INPUT;
    gpioConfig.pull_down_en = 1;
    gpioConfig.pull_up_en = 0;
    gpioConfig.pin_bit_mask = 1ULL << GPIO_NUM_8;
    gpio_config(&gpioConfig);

    while (1)
    {
        ESP_LOGI(TAG,"GPIO 8 level: %d",gpio_get_level(GPIO_NUM_8));
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

lz00qs, 2022-03-28:
@0xjakob Ok, I got it, thanks for your reply!

closed 2022-03-28 by lz00qs

0xjakob, 2022-03-28:
@lz00qs No problem!