espressif/esp-idf #8669: BLE Mesh server model could not update reprovisoned client model's message sequence

lz00qs, 2022-03-28:
## Environment

- Development Kit: https://item.taobao.com/item.htm?spm=a230r.1.14.7.72bd57deHXCzZp&id=645460125031&ns=1&abbucket=8#detail
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used: ESP32C3
- IDF version (run `git describe --tags` to find it): v4.4-rc1
- Build System: Visual studio code
- Operating System: Windows
- Using an IDE?: Visual studio code
- Power Supply: LDO 3.3V

## Problem Description

I have two ble mesh devices, one has a client model , the other has a server model, and I use ESP-BLE-MESH android app to provision these devices. Both of these two devices enabled  `Store BLE Mesh configuration persistently`  option and I have stored client transaction identifier (tid) to NVS manually. Firstly, I provision both devices, and both client and server model work well. Secondly, I use `esp_ble_mesh_node_local_reset();`  to reset client node's ble mesh and then reboot client node. Thirdly, I reprovision the client node use the phone app, and the tid of the client node will be restore from nvs, and I load it to the `esp_ble_mesh_generic_client_set_state` to send generic onoff set message, but the server can not  parse( or do action ) to these message due to the sequence error. How can I refresh, update or storage the message sequence to solve this problem?

### Expected Behavior

Server model can do action to the message.

### Actual Behavior

server model's monitor will display:

```shell
W (299595) BLE_MESH: Replay: src 0x0008 dst 0xffff seq 0x00015e
W (299955) BLE_MESH: Replay: src 0x0008 dst 0xffff seq 0x00015f
W (300195) BLE_MESH: Replay: src 0x0008 dst 0xffff seq 0x000160
```

and `ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET`  event will not happen.

lz00qs, 2022-03-30:
In other words, can I get the sequence of the current device, store it, and then restore the sequence after reset?

JiaLWang, 2022-04-12:
Hi,

Sorry for the late reply.

For the client node, when you using `esp_ble_mesh_node_local_reset()` to reset it, the information in NVS will be erase(like sequence number). For the server node, it still store the information of address and sequence of the client node.

You reprovision the client node, it has the same unicast address and sequence start from `0` . And when you call `esp_ble_mesh_generic_client_set_state` function to send message to server node. The sever node will check the message with replay protect list. Because the `SRC` address already stored in the RPL list, and the sequence number of the message is less than the sequence number in the list. So the the server node will regard the message as attack message, and ignore the message.

Hope it can help you.

lz00qs, 2022-04-12:
> Hi,
>
> Sorry for the late reply.
>
> For the client node, when you using `esp_ble_mesh_node_local_reset()` to reset it, the information in NVS will be erase(like sequence number). For the server node, it [...]

Thanks for your reply. It seems both server and client node will remove sequence number after  `esp_ble_mesh_node_local_reset()` , I used generic onoff client and server examples in ESP-IDF to test it. The problem is if I reprovision the client node, I need to reprovision the server node to solve the replay problem, the coupling of the two nodes is too high. I want when client node call  `esp_ble_mesh_node_local_reset()`  , the sequence number of client node will not be removed from nvs. Are there any APIs in ESP-IDF can do that? When I use the chips of TI or Nordic, they give the APIs to do that.

JiaLWang, 2022-04-12:
Hi, 

 Both server and client node will erase NVS information after `esp_ble_mesh_node_local_reset() `. 
The `esp_ble_mesh_node_local_reset() ` function is what we use for internal testing. For the mesh specification, if you want to reset a node. Need configuration client send `Config Node Reset` message to reset the node. In addition, the NVS will also erase in this way.

One more question. Why you using `esp_ble_mesh_node_local_reset() ` to reset the node, then reprovision it. What is your purpose? 
Did you call `esp_ble_mesh_node_local_reset() ` using the `重置设备` option of the EspBLEMesh App?  If you just want to delete the information of this node on the APP, you can use option `删除设备`.

lz00qs, 2022-04-12:
> Hi,
>
> Both server and client node will erase NVS information after `esp_ble_mesh_node_local_reset() `. The `esp_ble_mesh_node_local_reset() ` function is what we use for internal testing. For the [...]

Yes, I use 重置设备 to reset the device, because when I use 删除设备, the mesh info in this node will not be removed, I enabled  `Store BLE Mesh configuration persistently`  option in menuconfig, it seems 
 `esp_ble_mesh_node_local_reset()`  is the only api can remove the mesh info, I call this api in provisioning callback  `ESP_BLE_MESH_NODE_PROV_RESET_EVT` , that means when I use 重置设备, the mesh info will be removed. If I don't removed the mesh info , how can I reprovision a node ( put another way, if I use 删除设备, I can not add this node into any mesh network again)? This situation is same as the first question in https://www.esp32.com/viewtopic.php?t=15713 . For your more question, this scenario will occur in after deleting this node, the user wants to add this node back. Both Taobao and Tuya and other commercially available smart platforms require that information not be deleted when resetting the node. See 3.10.2 in https://developer.alibaba.com/docs/doc.htm?treeId=578&articleId=109585&docType=1  . To better communicate this issue, I can provide you with my WeChat or cell phone number!

JiaLWang, 2022-04-12:
Hi,
If you remove the node local mesh infomation and don't erase the NVS mesh information, when you reboot the node, the mesh information will be restore form the NVS. It is a already provisioned node,  and you can not reprovision the node.
In a word, If you want to reprovision the node, the NVS information must be erased. 

Q: ` (put another way, if I use 删除设备, I can not add this node into any mesh network again)` 
A: When you use `删除设备`, just delete the node info on the local APP.  The node still in the network.

And the RPL attacker appear because the EspBLEMESH APP reassigned the previous address to the node. you can using `nrf_mesh` to have a try. Maybe it will not occur this problem.

And another complex scheme is define a vendor model to send a vendor message to server node. It is used to delete the server's RPL list.

Hope it can help you.

lz00qs, 2022-04-12:
> Hi, If you remove the node local mesh infomation and don't erase the NVS mesh information, when you reboot the node, the mesh information will be restore form the NVS. It is a already provisioned [...]

I think the best solution for me is I can get the seq number of client, and set this number to client ( this method has worked for me on many other ble chips ( such as TI, Nordic, Telink, etc...) ). I have tested vendor message and nrf_mesh , the replay problem also appear. I have solved this problem by reprovisioning a random address to the client from app a weeks ago. But I still want espressif use practices that are currently common in the industry.

JiaLWang, 2022-04-12:
Thank you for your suggestion.

But for the mesh stack, the sequence number is internal data, we haven't considered making it available at the moment. As for the other solutions you mentioned above, it's more of a matter for the application layer to consider.

If you still have doubts, you can leave your contact information and I will communicate with you further.

Thanks.

lz00qs, 2022-04-12:
Thanks for your help. It seems espressif will not open more api for mesh stack!

closed 2022-04-12 by lz00qs

Shane555, 2022-11-07:
@lz00qs I am also in the same situation as you. But from what i see, i think we can run a netkey refresh as shown [here](https://docs.espressif.com/projects/esp-idf/en/v4.1.3/api-guides/esp-ble-mesh/ble-mesh-faq.html#:~:text=1.21%20In%20the%20Key%20Refresh%20procedure%2C%20how%20does%20Provisioner%20update%20the%20Netkey%20owned%20by%20nodes%3F)

xuanchibkd14, 2023-06-08:
hi, is there any solution to this problem? i have been searching for weeks

lz00qs, 2023-06-09:
> 

thanks for your solution!

lz00qs, 2023-06-09:
> @lz00qs I am also in the same situation as you. But from what i see, i think we can run a netkey refresh as shown [...]

Thanks for your solution. My final solution is use a nvs flash space to store the seq.

lz00qs, 2023-06-09:
> hi, is there any solution to this problem? i have been searching for weeks
@Shane555 @xuanchibkd14 
Solution: Expose the interface for Sequence number acquisition, then manually divide an nvs region to store it and manually set the seq when bind app key.

modify: ~/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/settings.c
add two functions
```c
void bt_mesh_set_seq(uint32_t iSeq)
{
    bt_mesh.seq = iSeq;
    bt_mesh_store_seq();
}

uint32_t bt_mesh_get_seq(void)
{
    return bt_mesh.seq;
}
```

then modify: ~/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/settings.h
add:
```c
void bt_mesh_set_seq(uint32_t iSeq);
uint32_t bt_mesh_get_seq(void);
```

call those funcion in your node reset function, for example:
```c
void MeshNodeReset() {
    ESP_LOGI(TAG, "ESP_BLE_MESH_NODE_PROV_RESET_EVT");
    if (meshDb.seq < bt_mesh_get_seq())
        meshDb.seq = bt_mesh_get_seq();
    esp_ble_mesh_node_local_reset();
    ESP_LOGI(TAG, "seq:%d", meshDb.seq);
    meshDb.appIdx = 0;
    meshDb.chAddr[0] = 0xFFFF;
    ble_mesh_nvs_store(NVS_HANDLE, MESH_DB_KEY, &meshDb, sizeof(meshDb)); // must store tid
    esp_restart();
}

case ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND:
                ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND");
                ESP_LOGI(TAG, "elem_addr 0x%04x, app_idx 0x%04x, cid 0x%04x, mod_id 0x%04x",
                         param->value.state_change.mod_app_bind.element_addr,
                         param->value.state_change.mod_app_bind.app_idx,
                         param->value.state_change.mod_app_bind.company_id,
                         param->value.state_change.mod_app_bind.model_id);
                if (param->value.state_change.mod_app_bind.company_id == 0xFFFF &&
                    param->value.state_change.mod_app_bind.model_id == ESP_BLE_MESH_MODEL_ID_GEN_ONOFF_CLI)
                {
                    updateAppIdx(param->value.state_change.mod_app_bind.app_idx);
                    // 获取分组地址
//                    store.app_idx = param->value.state_change.mod_app_bind.app_idx;
//                    mesh_example_info_store(); /* Store proper mesh example info */
                }
                if (param->value.state_change.mod_app_bind.model_id == 0x02e5)
                {
                    updateAppIdx(param->value.state_change.mod_app_bind.app_idx);
                }
                if(bt_mesh_get_seq() < getSeq())
                {
                    bt_mesh_set_seq(getSeq() + 100);
                }
                break;
```