espressif/esp-idf #8670: 【ESP32-C3/USB-Serial】serial.serialutil.SerialTimeoutException: Write timeout

xiaolongba, 2022-03-28:
Hello Espressif Experts,

Recently, I tried to flash the FW with `/dev/ttyACM0`, but it failed, the error info as below:
```c
 Executing task: /opt/esp/python_env/idf5.0_py3.8_env/bin/python /opt/esp/idf/components/esptool_py/esptool/esptool.py -p /dev/ttyACM0 -b 2000000 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 hello_world.bin 0x0 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin <

esptool.py v3.3-dev
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 5347, in <module>
    _main()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 5340, in _main
    main()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 4647, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 115, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 664, in connect
    last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 629, in _connect_attempt
    self.sync()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 519, in sync
    val, _ = self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 458, in command
    self.write(pkt)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 421, in write
    self._port.write(buf)
  File "/opt/esp/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 663, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
The terminal process "/opt/esp/python_env/idf5.0_py3.8_env/bin/python '/opt/esp/idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/ttyACM0', '-b', '2000000', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32c3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', 'detect', '0x10000', 'hello_world.bin', '0x0', 'bootloader/bootloader.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 1.
```

however, I tried to switch serial from USB-Serial to CP210X, it works well, as below:
```c
> Executing task: /opt/esp/python_env/idf5.0_py3.8_env/bin/python /opt/esp/idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 2000000 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 hello_world.bin 0x0 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin <

esptool.py v3.3-dev
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:bb:e6:60
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x00034fff...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 151184 bytes to 81602...
Wrote 151184 bytes (81602 compressed) at 0x00010000 in 2.2 seconds (effective 546.4 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 20832 bytes to 12672...
Wrote 20832 bytes (12672 compressed) at 0x00000000 in 0.5 seconds (effective 348.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 439.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

I can confirm the HW connection for USB-SERIAL is OK as it can output the log info with the serial tool, as below:
[image]

could you give me a hand, whether it is linked to the usb-serial-jtag driver?

dobairoland, 2022-03-28:
Could you please try if putting the chip manually to bootloader mode before running esptool helps?

Here is how to do it:
https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html#manual-bootloader

xiaolongba, 2022-03-28:
> Could you please try if putting the chip manually to bootloader mode before running esptool helps?

@dobairoland bro, it is the same as above, write timeout, as below:
```c
The terminal process "/opt/esp/python_env/idf5.0_py3.8_env/bin/python '/opt/esp/idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/ttyACM0', '-b', '2000000', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32c3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', 'detect', '0x10000', 'hello_world.bin', '0x0', 'bootloader/bootloader.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 1.

> Executing task: /opt/esp/python_env/idf5.0_py3.8_env/bin/python /opt/esp/idf/components/esptool_py/esptool/esptool.py -p /dev/ttyACM0 -b 2000000 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 hello_world.bin 0x0 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin <

esptool.py v3.3-dev
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 5347, in <module>
    _main()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 5340, in _main
    main()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 4647, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 115, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 664, in connect
    last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 629, in _connect_attempt
    self.sync()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 519, in sync
    val, _ = self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 458, in command
    self.write(pkt)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 421, in write
    self._port.write(buf)
  File "/opt/esp/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 636, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
The terminal process "/opt/esp/python_env/idf5.0_py3.8_env/bin/python '/opt/esp/idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/ttyACM0', '-b', '2000000', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32c3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', 'detect', '0x10000', 'hello_world.bin', '0x0', 'bootloader/bootloader.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 1.
```

radimkarnis, 2022-03-29:
Hello @xiaolongba,
this is weird behavior, it seems like the serial communication works reliably only in one direction.
Could you please try:
1) lower the baud rate
2) pull the latest esptool from https://github.com/espressif/esptool and try flashing the compiled binaries separately (guide [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html))

If this doesn't resolve the issue, the USB-Serial-JTAG driver is probably to blame.

xiaolongba, 2022-03-29:
Hello @radimkarnis 

> 1. lower the baud rate

I tried to flash with 115200, but it still failed, the details as below:
```c
sptool.py v3.3-dev
Creating esp32c3 image...
Merged 1 ELF section
Successfully created esp32c3 image.
Generated /workspaces/hello_world/build/hello_world.bin
[787/788] cd /workspaces/hello_world/build/esp-idf/esptool_py ...tition-table.bin /workspaces/hello_world/build/hello_world.bin
hello_world.bin binary size 0x226f0 bytes. Smallest app partition is 0x100000 bytes. 0xdd910 bytes (87%) free.
[787/788] cd /opt/esp/idf/components/esptool_py && /opt/esp/to...ld -P /opt/esp/idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32c3 -p /dev/ttyACM0 -b 115200 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3-dev
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 5347, in <module>
    _main()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 5340, in _main
    main()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 4647, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 115, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 664, in connect
    last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 629, in _connect_attempt
    self.sync()
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 519, in sync
    val, _ = self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 458, in command
    self.write(pkt)
  File "/opt/esp/idf/components/esptool_py/esptool/esptool.py", line 421, in write
    self._port.write(buf)
  File "/opt/esp/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 663, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
```

> 2\. pull the latest esptool from https://github.com/espressif/esptool and try flashing the compiled binaries separately (guide [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html))

as you can see above, the esptool version is the latest 3.3

radimkarnis, 2022-03-29:
> as you can see above, the esptool version is the latest 3.3

We already have `v4.0-dev` in esptool master. On the other hand, the version probably doesn't matter, I am more interested in running esptool independently from ESP-IDF.

xiaolongba, 2022-03-29:
ok. because I am using the docker image for Flashing the ESP32-C3, so I am not sure if it is linked to the usb-serial-jtag driver, it can works well on windows 10.

xiaolongba, 2022-03-30:
thanks to all guys who help me here, this issue has been resolved as the driver on docker side.

closed 2022-03-30 by xiaolongba

radimkarnis, 2022-03-30:
Thanks for letting us know, I am glad this got resolved!