espressif/esp-idf #8672: How to read processor temperature with riscv ulp on esp32s3

dreamcmi, 2022-03-28:
I have used FSM ULP on ESP32 to read the processor temperature, using tsens instruction assembly, but now I want to read the processor temperature through riscv ulp on esp32s3, but I can't find relevant information, what should be done to get the data right?

ESP-Marius, 2022-03-30:
There is no example or API for this yet (I've added an internal feature request ticket to track this).

The relevant information would be in the `On-Chip Sensor and Analog Signal Processing` chapter of the TRM, but this chapter hasn't unfortunately not been published yet for ESP32-S3. For now I would recommend you to look at `32.4 Temperature Sensor`-section in the  [ESP32-S2 TRM](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf)

As far as I know the operations to read the temperature is the same on S2 as on S3.

KaeLL, 2022-03-30:
> but this chapter hasn't unfortunately not been published yet for ESP32-S3.

Why?

ESP-Marius, 2022-03-31:
> > but this chapter hasn't unfortunately not been published yet for ESP32-S3.
> 
> Why?

It's simply not finished yet. Every chapter we publish has to go through multiple review and translation rounds, so finishing the whole TRM is not something that can be done in a short amount of time. 

To remedy this we try to prioritize finishing the chapters which we think our users requires the most (e.g. for peripherals with major changes, core stuff like "system and memory").

I think it might be useful for our users to know which chapters are still missing though, e.g. by listing them in the ToC, so they at least know that this content will be there in the future. I'll talk to our doc team about this.

dreamcmi, 2022-06-09:
> There is no example or API for this yet (I've added an internal feature request ticket to track this).
>
> The relevant information would be in the `On-Chip Sensor and Analog Signal Processing` chapter [...]

Is there any update progress now?

ESP-Marius, 2022-06-13:
For the docs: as a result of this issue we've added an overview of the missing chapters in the TRM as well as their progress status (see page 2 https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf). The chapter in question is now 89% completed according to this.

As for the SW side of things: we pushed a ab69df3ea762b74f0747517018cfc631c55a9dbc for a power related issue that had been blocking the ULP support, so as of last week it's possible to use the ULP on S3 (enabled again in 6e79cc69f9dbabc95e7c5adaf0228aaac6ceb809).

We are prioritizing adding more ULP support in the coming weeks, but the priority will probably be ADC/I2C, not the temp sensor, so unfortunately I can't give you a date for when we'll have an example available for this.

NoNullptr, 2022-11-11:
While it's not functional, the code in my bug report #10164 follows the Technical Reference Manual on how to read out the temperature sensor.